emsesp/EMS-ESP #594: Warm water problem - Junkers  CERAPURMODUL ZBS 14/100 S-3 MA 23 + FW200 + IPM2

km4lin, 2020-11-04:
Hello,
I've just install EMS-GATEWAY v 1.6 via cables which goes to FW200
MQTT and Home Assistant works
I have been bought EMS-GATEWAY for monitoring boiler and thermostat and control of the warm water.
Monitoring works but controlling the warm water not. I can set the temperature of warm water but I can't force the boiler to heat warm water.
Please help, without controlling the warm water 

**Bug description**
Some of the commands didn't work (via telnet):
```
call boiler wwtapactivated
call boiler wwonetime
call boiler wwactivated
```
but some commands working, for example:
```
call boiler wwtemp 44
call boiler wwtemp 15
call thermostat temp 22 1
call thermostat temp 22 2
```

**Steps to reproduce**
Login by telnet via putty, enter the command

**Expected behavior**
Change status of warm water sensors and warming the water by boiler

**Device information**
```
{
  "System": {
    "version": "2.1.1b1",
    "uptime": "000+13:58:05.144",
    "freemem": 33,
    "fragmem": 2
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 48607,
    "#read requests sent": 6557,
    "#write requests sent": 110,
    "#incomplete telegrams": 5663,
    "#tx fails": 3,
    "rx line quality": 89,
    "tx line quality": 91,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:20.05)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "Junkers FW200 (DeviceID:0x10 ProductID:106, Version:12.12)",
      "handlers": "0xA3 0x06 0xA2 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x21 ProductID:102, Version:20.07)",
      "handlers": "0x10C"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x20 ProductID:102, Version:20.07)",
      "handlers": "0x10C"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:20.05)",
      "handlers": ""
    }
  ]
}
```

I can send the logs etc.
Regards

MichaelDvP, 2020-11-04:
> Some of the commands didn't work (via telnet):
> call boiler wwtapactivated

this is a special configuration, assosiated with shower timer, it sets the boiler is test-mode and changes sone values. I think it only works in ww-flow systems to gernerate the cold shot.

> call boiler wwonetime

This is a endless story, please discuss this in #589.

> call boiler wwactivated

depends on system. In some sytems it works, in other not. My controller MC10 has a rotary knob to set ww off/fixed temp/auto. If i try to change wwactivated via ems-esp, the boiler set the value, but a second later the MC10 overwrites the value to the knob setting. 
you can check with `watch on 33` if there is a rest of the value after changing it. 

Imho the best way to control WW is to set the temperature.

km4lin, 2020-11-05:
Thanks for your answer @MichaelDvP 
In my setup after command like `call boiler wwtemp 44` temperature backs to previous after aprox. 3 minutes.
So in case like this we need a workaround.
In Home Assistant we need a script where the temperature is set multiple times. For example:
```
repeat:
  count: '60'
  sequence:
    - service: mqtt.publish
      data:
        topic: ems-esp/boiler
        payload: '{"cmd":"wwtemp","data":48}'
    - delay: '00:01:00'
```
It works good in my setup.
Hopes it will help someone with similar setup like my

MichaelDvP, 2020-11-06:
Check in terminal by `watch on 33`  who sets the temperature back. If it is the FW200 than we have to set the DHW-temperature in the thermostat.

MichaelDvP, 2020-11-06:
@km4lin : Read #523 there we add the ww-commands to CW400 thermostat. Maybe it's similar with FW200.

km4lin, 2021-04-22:
> 
> 
> Check in terminal by `watch on 33` who sets the temperature back. If it is the FW200 than we have to set the DHW-temperature in the thermostat.

Sorry for delay, here is log:
After 4 minutes it changed back lower temperature

```
ems-esp:/# watch on 33
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x33
ems-esp:/# call boiler wwsettemp 50
000+12:24:20.347 I 3: [boiler] Setting boiler warm water temperature to 50 C
000+12:24:23.328 N 4: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 32 00 00 23 00 02 48 00 FF FF
000+12:24:23.727 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 32 00 00 23 00 02 48 00 FF FF
000+12:24:42.776 E 6: [telegram] Rx: 90 A0 FF 00 20 (CRC 20 != B1)
000+12:24:45.290 E 7: [telegram] Rx: 90 A1 FF 00 01 (CRC 01 != B5)
000+12:24:50.596 E 8: [telegram] Rx: A0 10 02 00 90 12 (CRC 12 != E2)
000+12:25:03.982 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 32 00 00 23 00 02 48 00 FF FF
000+12:25:04.866 E 10: [telegram] Rx: 90 0B FF 00 00 70 02 01 00 B9 00 D5 F3 34 00 FE 10 (CRC 10 != DE)
000+12:25:07.297 E 11: [telegram] Rx: 88 00 18 00 00 01 A0 59 00 00 02 40 C0 80 00 80 00 01 87 FF FF FF 00 00 00 00 00 00 00 1C (CRC 1C != B7)
000+12:25:08.855 E 12: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+12:25:10.668 E 13: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+12:26:02.807 N 14: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 32 00 00 23 00 02 48 00 FF FF
000+12:26:37.216 E 15: [telegram] Rx: 88 00 18 00 00 01 9F 59 00 00 02 40 C0 80 00 80 00 01 86 FF FF FF 00 00 00 00 00 00 00 34 (CRC 34 != 9D)
000+12:26:37.840 E 16: [telegram] Rx: 88 00 19 00 00 7F 80 00 80 00 FF FF 00 00 00 12 43 09 22 C9 00 00 00 06 15 60 00 0B 54 10 (CRC 10 != 95)
000+12:26:40.504 E 17: [mqtt] Failed to publish to ems-esp/boiler_data_info after 3 attempts
000+12:26:40.808 E 18: [mqtt] Failed to publish to ems-esp/thermostat_data after 3 attempts
000+12:26:41.112 E 19: [mqtt] Failed to publish to ems-esp/mixer_data after 3 attempts
000+12:26:45.286 E 20: [telegram] Rx: A1 10 FF 00 90 20 (CRC 20 != 2D)
000+12:26:57.530 E 21: [telegram] Rx: 90 00 FF 00 00 5C (CRC 5C != 06)
000+12:27:00.303 E 22: [telegram] Rx: A0 08 1E 00 21 (CRC 21 != 61)
000+12:27:02.971 N 23: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 32 00 00 23 00 02 48 00 FF FF
000+12:27:05.485 E 24: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+12:27:07.814 E 25: [telegram] Rx: A0 08 1A 02 00 08 FE A0 (CRC A0 != 9D)
000+12:27:08.305 E 26: [telegram] Rx: 90 0B FF 00 00 66 03 00 04 19 2D 37 05 (CRC 05 != 1D)
000+12:27:36.844 E 27: [telegram] Rx: E0 88 00 34 00 32 01 69 01 A9 81 00 00 03 00 03 0D 68 00 06 EF 00 85 (CRC 85 != 47)
000+12:27:46.603 E 28: [telegram] Rx: 88 00 18 00 00 01 9E 59 00 00 02 40 C0 80 00 80 00 01 85 FF FF FF 00 00 00 00 00 00 15 (CRC 15 != 77)
000+12:27:47.881 E 29: [telegram] Rx: 90 A0 02 20 (CRC 20 != 29)
000+12:27:50.492 E 30: [telegram] Rx: A1 10 02 00 66 14 07 00 00 00 (CRC 00 != FC)
000+12:27:50.779 E 31: [telegram] Rx: FE FF 04 11 (CRC 11 != 30)
000+12:28:01.897 E 32: [telegram] Rx: FC 80 FC 80 FC 00 F8 (CRC F8 != 3A)
000+12:28:02.512 N 33: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 32 00 00 23 00 02 48 00 FF FF
000+12:28:16.701 E 34: [telegram] Rx: F8 E0 FC FE 0A (CRC 0A != 3B)
000+12:28:58.515 E 35: [telegram] Rx: FE F8 FC FE 15 (CRC 15 != 6B)
000+12:29:02.272 N 36: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 32 00 00 46 00 02 48 00 FF FF

```

MichaelDvP, 2021-04-22:
In this telegram the value of `wwseltemp` is not changed. You set to 50 and 3.rd byte is 0x32 (50) also in the last telegram it is 50. 
No other device overwrites it. Maybe the `wwsettemp` changes, this is the monitored value, influenced by the switch-program and wwmode of the thermostat. Check your warm water settings. (post output from command `show`).

km4lin, 2021-04-22:
Here is the output of show command:
```
EMS-ESP version 2.2.1

Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:20.05)
  Heating active: off
  Warm water/DHW active: off
  Service code:
  Service code number: 0
  Last error: EA(49) 01.03.2018 14:38
  Selected flow temperature: 0 °C
  Burner selected max power: 89 %
  Burner current power: 0 %
  Heating pump modulation: 0 %
  Outside temperature: 9.3 °C
  Current flow temperature: 37.7 °C
  Return temperature: 35.5 °C
  Mixer switch temperature: 24.7 °C
  Gas: off
  Heating Pump: off
  Fan: off
  Ignition: off
  Heating activated: on
  Heating temperature setting: 50 °C
  Circuit pump modulation max power: 100 %
  Circuit pump modulation min power: 10 %
  Circuit pump delay time: 3 min
  Burner min period: 3 min
  Burner min power: 0 %
  Burner max power: 89 %
  Temperature hysteresis on: -10 °C
  Temperature hysteresis off: 0 °C
  Burner starts: 4677
  Burner active time: 415 days 18 hours 59 minutes
  Heating active time: 276 days 20 hours 48 minutes
  Boiler total uptime: 2735 days 17 hours 44 minutes
  Warm water selected temperature: 50 °C
  Warm water set temperature: 15 °C
  Warm water disinfection temperature: 72 °C
  Warm water type: buffer
  Warm water charging type: 3-way valve
  Warm water circulation pump available: off
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: off
  Warm water current temperature (intern): 35.5 °C
  Warm water current temperature (extern): 41.7 °C
  Warm water current tap water flow: 0.0 l/min
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: on
  Warm water active: off
  Warm water heating: on
  Warm water starts: 1777
  Warm water active time: 138 days 22 hours 10 minutes
  Maintenance message: -
  Scheduled maintenance: off
  Next maintenance in: 2300 hours

Thermostat: Junkers FW200 (DeviceID:0x10 ProductID:106, Version:12.12)
  Time: 16:34:02 22/04/2021
  (hc 1) Setpoint room temperature: 18.5 °C
  (hc 1) Current room temperature: 21.3 °C
  (hc 1) Heat temperature: 22.0 °C
  (hc 1) Eco temperature: 18.5 °C
  (hc 1) Nofrost temperature: 2.0 °C
  (hc 1) Mode: heat
  (hc 1) Mode type: eco
  (hc 2) Setpoint room temperature: 18.5 °C
  (hc 2) Current room temperature: 21.3 °C
  (hc 2) Heat temperature: 23.0 °C
  (hc 2) Eco temperature: 18.5 °C
  (hc 2) Nofrost temperature: 1.0 °C
  (hc 2) Mode: heat
  (hc 2) Mode type: eco

Mixer: Junkers IPM (DeviceID:0x21 ProductID:102, Version:20.07)
  (hc 2) Setpoint flow temperature in assigned hc: 0 °C
  (hc 2) Flow temperature in assigned hc (TC1): 21.8 °C
  (hc 2) Pump status in assigned hc (PC1): off
  (hc 2) Valve status: 0 %

Mixer: Junkers IPM (DeviceID:0x20 ProductID:102, Version:20.07)
  (hc 1) Setpoint flow temperature in assigned hc: 0 °C
  (hc 1) Pump status in assigned hc (PC1): off

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:20.05)

```

MichaelDvP, 2021-04-22:
Ok, selected warm water is 50°C and unchanged, but setpoint is 15°C. I think this is coupled to `Mode type eco` from the thermostat, which sets also mixer-pumps off and setpoint of flow temp to zero.

pipacsba, 2021-04-22:
I think Michael is right. FW200 can control the warm water times, too. It can introduce separate schedules, or have the same schedule as the heating.

closed 2021-05-18 by proddy