emsesp/EMS-ESP #596: Strange temperature values on Dallas sensor

sanschaussure, 2020-11-05:
Dear all, 

I installed the version 2.1 on a ESP-12 (ESP8266). The connected ds18b20 in non-parasitic mode shows a value of ~70 °C in normal room condition (21-23 °C). Is it somehow possible, that the value is in ° Fahrenheit and not in Celsius? 70 °F = 21 °C

[image]

proddy, 2020-11-05:
Hi, did it work before with other earlier versions of EMS-ESP? Did you try the sensor with other software, it could be malfunctioning?

sanschaussure, 2020-11-05:
I just start with version 2.1. So I have no information of former systems. I also tested the sensor on an Arduino Nano system with the "tester" example of the library for these sensors. Value is more or less correct.
I also tried another sensor on the EMS-ESP. This sensor didn't work at all while it worked fine on the Arduino system.

proddy, 2020-11-05:
Ok, that is odd. I first thought it may be different DS18 from the same family but the first number of the ID is 28 which is indeed a DS18B20. @MichaelDvP any ideas here?

sanschaussure, 2020-11-05:
BTW. I use a own developed board based on your circuit for the TX-path and the RX-path based on this circuit [(link)](http://www.kabza.de/MyHome/EMSBus/EMSbus.html). The System runs completely on 3.3 V from a buck converter feed with 12V. Between the positive and the data pin of the sensor is a 4.7k resistor (no parasitic mode).

proddy, 2020-11-05:
It should work then.

MichaelDvP, 2020-11-06:
@sanschaussure If the sensor works in other systems, it should also work in ems-esp. For me it looks like a power issue. Check the 3.3v supply on the sensor. Is it connected? Is is stable? Try with an additional capacitor.

jvm1000, 2020-12-21:
I have bought many ds18b20, all off them conected in Qubino Flush 1 Relay, a zwave device, and sometime, one of them gives me measurements of 80 degrees, I change it and throw the one that gives me the bad measure and that's it. There is no apparent reason.

closed 2021-10-04 by proddy