emsesp/EMS-ESP #597: MQTT tapwater_active is reporting unreliably

giddyhup, 2020-11-06:
v2.1.0b6 on GBx72/Trendline/Cerapur/Greenstar Si/27i

Messages on the subtopic tapwater_active do appear not working properly. When I use warm water an 'off' value is published before 'on' (a couple of seconds later) and when I stop it takes longer than it should be and 'off' is published two times. This happens even when I subscribe with QOS 2.

proddy, 2020-11-06:
so the problem is with publishing the MQTT topic. Are you seeing any MQTT Publish errors (in WebUI or Console)?
Is the warm water active correct in the web and console too? 
Just trying to narrow down the issue.

giddyhup, 2020-11-06:
There are 0 publish errors. In the web dashboard "Warm water/DHW active" shows "off" and "on" consistently what is being published. The console activity trace also seems consistent. Should there be consecutive similar messages?

proddy, 2020-11-06:
correct, so something wrong with the MQTT publish logic. I'll take a look.

giddyhup, 2020-11-06:
An additional observation: seemingly, 'off' is not published when no more water is tapped but when the warm water tap circuit shuts off (when the boiler's LED gets turns off - only noticeable when the heating circuit which drives the same LED is also off).

MichaelDvP, 2020-11-06:
We have `tapwater_active` published as own topic, this is always published on change, and in `boiler_data`, this is published as set in mqtt-config and can be delayed.
@giddyhup the `tapwater_active` is read from the boiler and is the same value as the internal boiler LED.

giddyhup, 2020-11-06:
@MichaelDvP thanks for the comments. If it is published on change then I wouldn't expect to get the same payload multiple times (that's less of an issue and can still be filtered out). 

The LED I am talking about, the master LED, is the same for the heating circuit and also on when no warm water is being tapped. If the topic is tied to that LED then it would not be of use. Perhaps we have a misunderstanding here.

MichaelDvP, 2020-11-06:
Sorry it's a misunderstanding, i dont know your boiler, i thought it is a LED that indicates DHW preparation. 

The function of the tapwater_active-topic depends on warm water system. If it is a flow-system, the boiler have a flow sensor and starts the burner if water flows. In a buffer-system there is no flow sensor and the boiler starts to charge the buffer if the buffertemperature falls below setpoint-5°C, then the tapwater_active-topic shows the charging, not the water-flow.

giddyhup, 2020-11-06:
Thanks, @MichaelDvP, for elaborating. I would assume my boiler is a flow system or at least acts like one. I have to wait for the water to become warm when I open the tap. Although, in non-eco setting this happens faster.  

Regardless, this only may explain delays but not why an 'off' payload is published first when I open the tap and why multiple similar messages are being published.

glitter-ball, 2020-11-06:
Many Bosch boilers have 'Eco' and 'Preheat' functions for hot water. What this does depends on the hot water system.

For an instantaneous hot water system (what we'd call a 'combi' boiler in the UK / flow boiler), the 'Eco' setting provides no heating of the water until you turn the tap on. If you select 'Preheat', the boiler will fire periodically to keep the DHW plate heat exchanger hot, so less water has to be run before it gets hot. You use more gas this way but waste less water... Sometimes this can be selected manually from the boiler, sometimes it's controlled by a time programme.

For a storage cylinder / buffer system, with 'Eco' selected, the boiler will heat the water when its temperature falls 10ºC below the setpoint. With 'Preheat' selected, the boiler will heat the water when it falls 5ºC below the setpoint.

glitter-ball, 2020-11-06:
I have a large blue LED on my boiler which lights in two circumstances : either when heating is active  or when it's charging the hot water cylinder.

Because the pump is in energy saving mode, the blue light can be on for heating with apparently nothing happening - i.e. no fan, no burner, no pump. Boiler just sitting there. This can be disabled in the menu or set just to light if there is an error.

proddy, 2020-11-07:
the boiler state is captured from the 0x18 telegram. What would be useful is to watch this telegram while the hot water tap is turned on->off->on while noting down the times. It'll also show us the service code numbers.

giddyhup, 2020-11-07:
I am not sure if there are too many telegrams but capturing and turning the warm water on and off simultaneously is a bit challenging in my household but here is my trace:

```
008+21:29:39.673 N 98: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 43 02 83 10 00 01 03 2D C0 01 7F 80 00 80 00 00 C2 FF 00 00 01 1C 00 02 18
008+21:29:40.318 N 99: [emsesp] Thermostat(0x18) -> (0x00), (0x2A5), data: 21 (offset 2)
008+21:29:40.773 N 100: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 43 02 84 10 00 09 03 2D C0 01 7F 80 00 80 00 00 F2 FF 00 00 00 C8 00 02 18
008+21:29:40.995 N 101: [emsesp] Thermostat(0x18) -> Boiler(0x08), (0x1A), data: 4B 10 64
008+21:29:41.698 N 102: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 43 02 84 10 49 09 03 25 C0 01 7F 80 00 80 00 00 EE FF 00 00 00 C8 5F 02 18
008+21:29:42.673 N 103: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 86 7F 4A 0A 13 05 C0 01 C7 80 00 80 00 00 EA FF 00 00 00 C9 5E 02 18
008+21:29:43.697 N 104: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 8A 7F 7A 0A 13 05 C0 01 F1 80 00 80 00 01 2A FF 00 00 00 C9 5F 02 18
008+21:29:44.673 N 105: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 90 7F 7C 0A 13 45 C0 01 A5 80 00 80 00 01 2B FF 00 00 00 C9 63 02 18
008+21:29:45.060 N 106: [emsesp] Thermostat(0x18) -> (0x00), (0x06), data: 14 0B 0C 07 1C 17 05 00 18 FF 00
008+21:29:45.698 N 107: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 95 7F 7E 0A 13 45 C0 01 53 80 00 80 00 01 30 FF 00 00 00 C9 5F 02 18
008+21:29:47.697 N 108: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 A2 7F 7E 08 03 65 C0 00 FE 80 00 80 00 01 34 FF 00 00 00 C9 0A 02 18
008+21:29:48.673 N 109: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 AC 7F 00 08 03 64 C0 00 F1 80 00 80 00 00 B1 FF 00 00 01 31 00 00 00
008+21:29:49.640 N 110: [emsesp] Thermostat(0x18) -> (0x00), (0x2A5), data: 20 (offset 2)
008+21:29:49.829 N 111: [emsesp] Thermostat(0x18) -> Boiler(0x08), (0x1A), data: 4B 10 00
008+21:29:49.898 N 112: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 B8 7F 00 00 03 64 C0 00 EB 80 00 80 00 00 00 FF 00 00 01 31 00 00 00
```
At the same time, there is one 'off', one 'on' and (at least) two more 'off' messages being published.

MichaelDvP, 2020-11-07:
looks ok to me, if i pick only the changes in status and not temperatures (which also triggers boiler_data) there should only be two publishes to topic tapwater. 

`008+21:29:39.673 N 98: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 43 02 83 10 00 01 03 2D C0 01 7F 80 00 80 00 00 C2 FF 00 00 01 1C 00 02 18`
heating on, flame off, heating+ignition+fan+burner on , servicenumber 284 (gas magnet valve opening)

`008+21:29:40.773 N 100: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 43 02 84 10 00 09 03 2D C0 01 7F 80 00 80 00 00 F2 FF 00 00 00 C8 00 02 18`
heating on, flame on, heating+ignition+fan+burner on, servicenumber 200 (heating)

`008+21:29:41.698 N 102: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 43 02 84 10 49 09 03 25 C0 01 7F 80 00 80 00 00 EE FF 00 00 00 C8 5F 02 18`
heating on, flame on, heating+fan+burner on, ignition off, servicenumber 200 (heating)

`008+21:29:42.673 N 103: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 86 7F 4A 0A 13 05 C0 01 C7 80 00 80 00 00 EA FF 00 00 00 C9 5E 02 18`
water on, flame on, heating off, fan+burner on, servicenumber 201 (DHW)
publish tapwater on

`008+21:29:44.673 N 105: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 90 7F 7C 0A 13 45 C0 01 A5 80 00 80 00 01 2B FF 00 00 00 C9 63 02 18`
water on, flame on, 3way-valve+fan+burner on

`008+21:29:47.697 N 108: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 A2 7F 7E 08 03 65 C0 00 FE 80 00 80 00 01 34 FF 00 00 00 C9 0A 02 18`
water off, flame on, 3way-valve+heatingpump+fan+burner on, servicenumber 201 (DHW)
publish tapwater off

`008+21:29:48.673 N 109: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 AC 7F 00 08 03 64 C0 00 F1 80 00 80 00 00 B1 FF 00 00 01 31 00 00 00`
water off, flame on, 3way-valve+heatingpump+fan on, servicenumber 305 (blocking after dhw priority)

`008+21:29:49.898 N 112: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 B8 7F 00 00 03 64 C0 00 EB 80 00 80 00 00 00 FF 00 00 01 31 00 00 00`
heating off, water off, flame off, 3way-valve+heatingpump+fan on, servicenumber 305

MichaelDvP, 2020-11-07:
@giddyhup if you do the same `watch on 18` and additional a `log debug` we can see the mqtt-publishes in the timeline too.

giddyhup, 2020-11-08:
It is extremely hard to get an event log when the tap and the computer with the telnet session are several rooms apart. Following is an example, but, as the issues states, I also encountered situation where the state remained on and where several 'off' messages were being published.
```
010+00:04:29.352 D 407: [mqtt] Publishing topic ems-esp/boiler_data (#61697, attempt #1, pid 41506)
010+00:04:29.847 N 408: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 DE 7F 00 00 03 64 C0 01 65 80 00 80 00 00 00 FF 00 00 01 31 00 00 00
010+00:04:29.847 D 409: [emsesp] Received UBAMonitorFast
010+00:04:30.896 N 410: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 DC 7F 00 00 03 44 C0 01 65 80 00 80 00 00 00 FF 00 00 01 31 00 00 00
010+00:04:30.896 D 411: [emsesp] Received UBAMonitorFast
010+00:04:35.903 N 412: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 DB 7F 00 02 13 64 C0 01 70 80 00 80 00 00 00 FF 00 00 01 1B 63 02 18
010+00:04:35.903 D 413: [emsesp] Received UBAMonitorFast
010+00:04:35.931 D 414: [mqtt] Publishing topic ems-esp/tapwater_active (#61698, attempt #1, pid 41507)
010+00:04:36.155 D 415: [emsesp] Received UBAMonitorSlow
010+00:04:36.336 D 416: [mqtt] Publishing topic ems-esp/thermostat_data (#61699, attempt #1, pid 41508)
010+00:04:36.378 D 417: [emsesp] Received UBAMaintenanceStatus
010+00:04:36.648 D 418: [emsesp] Received MC10Status
010+00:04:36.978 N 419: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 01 DC 7F 00 02 13 64 C0 01 99 80 00 80 00 00 00 FF 00 00 00 C9 5D 02 18
010+00:04:36.978 D 420: [emsesp] Received UBAMonitorFast
010+00:04:37.197 D 421: [emsesp] Received UBAMonitorWW
010+00:04:37.698 N 422: [emsesp] Thermostat(0x18) -> Boiler(0x08), (0x1A), data: 4B 10 00
010+00:04:37.800 N 423: [emsesp] Thermostat(0x18) -> (0x00), (0x2A5), data: 00 D0 20 2A 4B 00 2A 24 01 A1 03 03 01 01 A1 01 A7 00 00 11 01 03
010+00:04:37.800 D 424: [emsesp] Received RC300Monitor
010+00:04:38.285 N 425: [emsesp] Thermostat(0x18) -> (0x00), (0x2A5), data: 08 23 00 00 04 00 00 00 00 00 10 4B 00 3C 01 FF 00 00 (offset 22)
010+00:04:38.285 D 426: [emsesp] Received RC300Monitor
010+00:04:38.476 N 427: [emsesp] Thermostat(0x18) -> (0x00), (0x31D), data: 00 00 0A 07
010+00:04:38.476 D 428: [emsesp] Received RC300WWmode2
010+00:04:38.828 N 429: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 01 DE 7F 00 02 13 6D C0 01 51 80 00 80 00 00 00 FF 00 00 00 C9 62 02 18
010+00:04:38.828 D 430: [emsesp] Received UBAMonitorFast
010+00:04:39.061 N 431: [emsesp] Thermostat(0x18) -> (0x00), (0x06), data: 14 0B 0F 08 03 14 06 00 18 FF 00
010+00:04:39.061 D 432: [emsesp] Received RCTime
010+00:04:39.272 D 433: [mqtt] Publishing topic ems-esp/boiler_data (#61700, attempt #1, pid 41509)
010+00:04:39.828 N 434: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 01 E1 7F 00 0A 13 6D C0 01 49 80 00 80 00 00 EB FF 00 00 00 C9 61 02 18
010+00:04:39.828 D 435: [emsesp] Received UBAMonitorFast
010+00:04:39.879 D 436: [mqtt] Publishing topic ems-esp/heating_active (#61701, attempt #1, pid 41510)
010+00:04:39.980 D 437: [mqtt] Publishing topic ems-esp/tapwater_active (#61702, attempt #1, pid 41511)
010+00:04:40.828 N 438: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 01 E3 7F 00 0A 13 65 C0 01 54 80 00 80 00 00 F3 FF 00 00 00 C9 5C 02 18
010+00:04:40.828 D 439: [emsesp] Received UBAMonitorFast
010+00:04:45.828 N 440: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 E5 0F 2F 0A 13 65 C0 01 7B 80 00 80 00 00 A5 FF 00 00 00 C9 00 02 18
010+00:04:45.828 D 441: [emsesp] Received UBAMonitorFast
010+00:04:46.098 D 442: [emsesp] Received MC10Status
010+00:04:46.347 D 443: [emsesp] Received UBAMonitorWW
010+00:04:46.347 D 444: [mqtt] Publishing topic ems-esp/thermostat_data (#61703, attempt #1, pid 41512)
010+00:04:47.828 N 445: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 DF 0F 16 0A 13 65 C0 01 7A 80 00 80 00 00 73 FF 00 00 00 C9 00 02 18
010+00:04:47.828 D 446: [emsesp] Received UBAMonitorFast
010+00:04:49.278 D 447: [mqtt] Publishing topic ems-esp/boiler_data (#61704, attempt #1, pid 41513)
010+00:04:53.928 N 448: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 D5 0F 10 0A 13 65 C0 01 79 80 00 80 00 00 5D FF 00 00 00 C9 00 02 18
010+00:04:53.928 D 449: [emsesp] Received UBAMonitorFast
010+00:04:55.878 N 450: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 D5 0F 0F 0A 13 65 C0 01 79 80 00 80 00 00 5C FF 00 00 00 C9 00 02 18
010+00:04:55.878 D 451: [emsesp] Received UBAMonitorFast
010+00:04:56.199 D 452: [emsesp] Received MC10Status
010+00:04:56.357 D 453: [mqtt] Publishing topic ems-esp/thermostat_data (#61705, attempt #1, pid 41514)
010+00:04:56.421 D 454: [emsesp] Received UBAMonitorWW
010+00:04:57.902 N 455: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 D6 0F 0F 0A 13 65 C0 01 79 80 00 80 00 00 5B FF 00 00 00 C9 00 02 18
010+00:04:57.902 D 456: [emsesp] Received UBAMonitorFast
010+00:04:59.288 D 457: [mqtt] Publishing topic ems-esp/boiler_data (#61706, attempt #1, pid 41515)
```

Is there a way to get timestamps in epoch or actual time and not relative to uptime?

MichaelDvP, 2020-11-08:
I see the publishes only two times
> 010+00:04:35.931 D 414: [mqtt] Publishing topic ems-esp/tapwater_active (#61698, attempt #1, pid 41507)
> 010+00:04:39.980 D 437: [mqtt] Publishing topic ems-esp/tapwater_active (#61702, attempt #1, pid 41511)

The first one is a bit unclear, a change from 00 to 02 in byte 5 should not trigger the publish, maybe the trigger is one message before the log starts.  The second is the switch to on, caused by byte 5 of 0x18 going from 02 to 0A. 

some terminals has a option to log to file, than you have enough time to go to another room. Or use the syslog option in ems-esp, than you have also the epoch time from the syslog server. (you have to set the logging in syslog-settings and start the watch from a telnet-session).

giddyhup, 2020-11-09:
Ok, here is a merged log. Received MQTT messages are marked with MQT_ and I tried to merge them with the timestamped log entries:

```
LOG_17:50:48 _11+02:51:33.319 N 5: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 36 02 12 64 1A 09 03 25 C0 01 51 80 00 80 00 00 A3 FF 00 00 00 C8 00 02 18
LOG_17:50:48 _11+02:51:33.319 D 6: [emsesp] Received UBAMonitorFast
LOG_17:50:48 _11+02:51:33.539 D 7: [emsesp] Received MC10Status
LOG_17:50:49 _11+02:51:33.787 D 8: [emsesp] Received UBAMonitorWW
LOG_17:50:49 _11+02:51:35.368 N 9: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 36 02 11 64 1A 09 03 25 C0 01 51 80 00 80 00 00 A3 FF 00 00 00 C8 00 02 18
LOG_17:50:51 _11+02:51:35.368 D 10: [emsesp] Received UBAMonitorFast
LOG_17:50:51 _11+02:51:37.127 D 11: [mqtt] Publishing topic ems-esp/thermostat_data (#17235, attempt #1, pid 62580)
LOG_17:50:52 _11+02:51:39.013 N 12: [emsesp] Thermostat(0x18) -> Boiler(0x08), (0x1A), data: 36 64 64
LOG_17:50:54 _11+02:51:39.115 N 13: [emsesp] Thermostat(0x18) -> (0x00), (0x2A5), data: 00 D9 21 2B 36 00 2B 24 00 EB 07 03 01 00 EB 02 B7 00 00 11 01 03
LOG_17:50:54 _11+02:51:39.115 D 14: [emsesp] Received RC300Monitor
LOG_17:50:54 _11+02:51:39.648 N 15: [emsesp] Thermostat(0x18) -> (0x00), (0x2A5), data: 08 7B 00 00 04 00 00 00 00 00 64 4B 00 3C 01 FF 00 02 (offset 22)
LOG_17:50:55 _11+02:51:39.648 D 16: [emsesp] Received RC300Monitor
LOG_17:50:55 _11+02:51:39.840 N 17: [emsesp] Thermostat(0x18) -> (0x00), (0x31D), data: 00 00 0A 07
LOG_17:50:55 _11+02:51:39.840 D 18: [emsesp] Received RC300WWmode2
LOG_17:50:55 _11+02:51:40.262 D 19: [mqtt] Publishing topic ems-esp/boiler_data (#17236, attempt #1, pid 62581)
LOG_17:50:55 _11+02:51:43.268 N 20: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 36 02 12 64 1B 09 03 25 C0 01 51 80 00 80 00 00 A2 FF 00 00 00 C8 00 02 18
LOG_17:50:58 _11+02:51:43.268 D 21: [emsesp] Received UBAMonitorFast
LOG_17:50:58 _11+02:51:43.539 D 22: [emsesp] Received MC10Status
LOG_17:50:59 _11+02:51:43.861 D 23: [emsesp] Received UBAMonitorWW
LOG_17:50:59 _11+02:51:47.136 D 24: [mqtt] Publishing topic ems-esp/thermostat_data (#17237, attempt #1, pid 62582)
LOG_17:51:02 _11+02:51:49.318 N 25: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 12 7F 32 0A 13 05 C0 01 CF 80 00 80 00 00 D8 FF 00 00 00 C9 5E 02 18
LOG_17:51:04 _11+02:51:49.318 D 26: [emsesp] Received UBAMonitorFast
LOG_17:51:04 _11+02:51:49.359 D 27: [mqtt] Publishing topic ems-esp/heating_active (#17238, attempt #1, pid 62583)
LOG_17:51:05 _11+02:51:49.460 D 28: [mqtt] Publishing topic ems-esp/tapwater_active (#17239, attempt #1, pid 62584)
MQT_17:51:05.094725] on
LOG_17:51:05 _11+02:51:50.268 D 29: [mqtt] Publishing topic ems-esp/boiler_data (#17240, attempt #1, pid 62585)
LOG_17:51:05 _11+02:51:50.368 N 30: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 12 7F 7B 0A 13 45 C0 01 E7 80 00 80 00 01 58 FF 00 00 00 C9 61 02 18
LOG_17:51:06 _11+02:51:50.368 D 31: [emsesp] Received UBAMonitorFast
LOG_17:51:06 _11+02:51:53.368 N 32: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 1C 7F 7E 0A 13 45 C0 01 5D 80 00 80 00 01 56 FF 00 00 00 C9 5D 02 18
LOG_17:51:09 _11+02:51:53.368 D 33: [emsesp] Received UBAMonitorFast
LOG_17:51:09 _11+02:51:53.589 D 34: [emsesp] Received MC10Status
LOG_17:51:09 _11+02:51:53.837 D 35: [emsesp] Received UBAMonitorWW
LOG_17:51:09 _11+02:51:54.343 N 36: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 25 7F 7E 0A 13 65 C0 01 40 80 00 80 00 01 55 FF 00 00 00 C9 61 02 18
LOG_17:51:10 _11+02:51:54.343 D 37: [emsesp] Received UBAMonitorFast
LOG_17:51:10 _11+02:51:54.987 N 38: [emsesp] Thermostat(0x18) -> Boiler(0x08), (0x1A), data: 36 64 00
LOG_17:51:10 _11+02:51:55.267 N 39: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 00 02 2E 7F 7E 0A 13 65 C0 01 29 80 00 80 00 01 56 FF 00 00 00 C9 59 02 18
LOG_17:51:10 _11+02:51:55.267 D 40: [emsesp] Received UBAMonitorFast
LOG_17:51:10 _11+02:51:55.487 N 41: [emsesp] Thermostat(0x18) -> (0x00), (0x2A5), data: 20 (offset 2)
LOG_17:51:11 _11+02:51:55.487 D 42: [emsesp] Received RC300Monitor
LOG_17:51:11 _11+02:51:56.317 N 43: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 3B 7F 00 08 03 64 C0 01 24 80 00 80 00 01 37 FF 00 00 01 31 13 02 18
LOG_17:51:11 _11+02:51:56.317 D 44: [emsesp] Received UBAMonitorFast
LOG_17:51:11 _11+02:51:56.337 D 45: [mqtt] Publishing topic ems-esp/tapwater_active (#17241, attempt #1, pid 62586)
MQT_17:51:11.967928 off
LOG_17:51:12 _11+02:51:57.145 D 46: [mqtt] Publishing topic ems-esp/thermostat_data (#17242, attempt #1, pid 62587)
LOG_17:51:12 _11+02:51:57.267 N 47: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 47 7F 00 08 03 64 C0 01 26 80 00 80 00 00 0D FF 00 00 01 31 00 00 00
LOG_17:51:12 _11+02:51:57.267 D 48: [emsesp] Received UBAMonitorFast
LOG_17:51:12 _11+02:51:58.317 N 49: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 52 7F 00 00 03 64 C0 01 27 80 00 80 00 00 00 FF 00 00 01 31 00 00 00
LOG_17:51:13 _11+02:51:58.317 D 50: [emsesp] Received UBAMonitorFast
LOG_17:51:13 _11+02:51:58.358 D 51: [mqtt] Publishing topic ems-esp/heating_active (#17243, attempt #1, pid 62588)
LOG_17:51:14 _11+02:51:58.459 D 52: [mqtt] Publishing topic ems-esp/tapwater_active (#17244, attempt #1, pid 62589)
MQT_17:51:14.089772 off
LOG_17:51:14 _11+02:51:59.267 N 53: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 02 5B 7F 00 00 03 64 C0 01 42 80 00 80 00 00 00 FF 00 00 01 31 5B 00 00
LOG_17:51:14 _11+02:51:59.267 D 54: [emsesp] Received UBAMonitorFast
[... cut ...]
LOG_17:51:36 _11+02:52:20.730 D 153: [mqtt] Publishing topic ems-esp/heartbeat (#17262, attempt #1, pid 62607)
LOG_17:51:36 _11+02:52:21.331 N 154: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 F3 7F 00 00 03 64 C0 01 A1 80 00 80 00 00 00 FF 00 00 01 31 00 00 00
LOG_17:51:36 _11+02:52:21.331 D 155: [emsesp] Received UBAMonitorFast
LOG_17:51:36 _11+02:52:23.331 N 156: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 32 01 EB 7F 00 00 03 64 C0 01 A0 80 00 80 00 00 00 FF 00 00 01 31 00 00 00
```

MichaelDvP, 2020-11-10:
Yes i see there are some publishes to much, but i can not reproduce and can not find something in the code that causes it. 
Just to be sure, you are using the actual software?

giddyhup, 2020-11-10:
> Just to be sure, you are using the actual software?

I do not understand what else I should be using. The MQTT messages are from the command line on a raspberry pi (`mosquitto_sub -h broker.lan -t ems-esp/tapwater_active | ts 'MQT_%H:%M:%.S'`).

MichaelDvP, 2020-11-10:
In the first post you have mentioned v2.1.0b6. I checked this very old software and indeed the logic for tapwater in this build gives some extra messages, but this is changed a long time ago.
Next time please check if it is already fixed __before__ reporting bugs.  especially if there is newer __release__ with changelog: __Accurate detection of warm water and heating (#515)__

giddyhup, 2020-11-10:
Sorry about the confusion. I was asked by @proddy to open an issue. From the conversation it appeared that #515 was fixed with 2.1.0b2.

giddyhup, 2020-11-10:
I updated to 2.1.0 and duplicate 'off' messages are gone. Sometimes the 'off' seems to still be delayed but I have to assume that this is due to inertia in the whole system.  

Thanks.

proddy, 2021-01-05:
re-opening. With 2.20 @giddyhup is still not getting the active flags immediately sent in HA

giddyhup, 2021-01-05:
Thanks @proddy for re-opening the issue. What are the conditions when 'off' is published under the topic 'ems-esp/tapwater_active'? What can I trace to help troubleshooting?

proddy, 2021-01-05:
There's a bit of info in #515 . Are you seeing this discrepancy with both the heating and warm-water (tapwater), or both?

The logic is pulled from the 0x18 telegram at position 6. If warm water is active the bits 2 and 4 are set (b1010). Heating uses a different bit mask (b1001).

What you could do is open Console, `watch raw 18`, grab a few lines (they are sent every few seconds), turn on hot water and check, then turn off hot tap and check again. Look for the this byte:

```
[telegram] Rx: 08 00 18 00 4B 02 D2 41 41 09 01 25 40 02 3C 80 00 02 33 01 86 14 2D 48 00 C8 00 02 00 6F
                                          ^
                                          | 
```

In my example above the value 0x09 = b00001001 = only heating is active

MichaelDvP, 2021-01-05:
@giddyhup for now the `tapwater_active` checks only the status of boiler (ww-mode/heating-mode) and burner (on/off) given in the byte proddy mentioned. If the boiler heats it's buffer a bit longer as water flows, this is not detected. But this detection is in sync with the boiler internal detection and counts to `wwStarts` and `wwWorkM`.
For real flow detection we need to add the information of  flow sensor if there is one. Can you check `read 8 34`  one time when water is flowing, one time if water is stopped (databyte 9). The problem with flow sensor is, that it's value is 0 if there are no sensor, we can not decide if sensor is missing or there is no flow. Maybe it's a way to ignore the sensor if the value is zero and read it always after the first time it reports a positive value. If there is no flow sensor the boiler makes DHW only on temperature regardless of flow.
Also 0x34 reports the ww-system, if it is `buffered flow` heating the buffer will also show `tapwater_active`, you can check this `wwType` also in web-boiler-values or api/call boiler info.

giddyhup, 2021-01-05:
off/on/off:

[image]

During this test state changes were reported properly.  

The web interface shows 'flow' under 'Warm water type' for the boiler.

giddyhup, 2021-01-05:
These are the entries I see when I turn ww on and off (beginning in line 8):

[image]

proddy, 2021-01-05:
when did you turn it off, after # 20? Also is your central heating on too?

MichaelDvP, 2021-01-05:
Yes there is a flowsensor in your system. I've added a check to the dev. Please try. The telegram 0x34 is send every 10 sec,, i think this will be the reaction time.

giddyhup, 2021-01-05:
>when did you turn it off, after # 20? Also is your central heating on too?

Sorry, I was too busy fiddling with tablet's buttons to create the screenshot to look at the timestamp. Heating was running as well. I will try to create better conditions and turn the heating off the next time.

MichaelDvP, 2021-01-05:
> So, may I wait for the next compiled dev binary? (To me it is really not that urgent.)

It's already there [https://github.com/proddy/EMS-ESP/releases/tag/dev](https://github.com/proddy/EMS-ESP/releases/tag/dev)
> For my understanding, once this works it may take up to 10 seconds for the last state to be reported?

I think so, but maybe the boiler publishes the change out of order. You can check with `watch on 8` the boiler telegrams, open/close the water and check the telegrams 0x34 (flow) and 0x18 (status). 
The above telegrams shows only status, in N8 its heating, N9 starts ww, N26 water stop, but burner still active, N28 burner off. Tapwater ative shoud be shown from N9 to N25. The flow is not in these telegrams.

giddyhup, 2021-01-13:
Sorry for the delay. Things got in the way.

It looks much better now. I think I saw some duplicate Off messages but overall the reported state is consistent with what I expect.

closed 2021-01-13 by giddyhup