emsesp/EMS-ESP #598: Units for Solar Energy values

PhillyGilly, 2020-11-06:
I've just been tidying up my HA solar tab and I can't re-create the History Graph for my daily solar generation. This is because the Solar Energy Values don't have units i.e. that I previously added in my yaml config files.
[image]
The units appear to be added to temp values so their history graphs are fine.

PhillyGilly, 2020-11-06:
I also noticed in the above screen shot that Solar Pump working time is missing. It is in the auto-discovered entity but not in the MQTT message {"collectorTemp":39.2,"tankBottomTemp":47.4,"tankBottomTemp2":33.2,"solarPumpModulation":7,"cylinderPumpModulation":0,"solarPump":"on","valveStatus":"on","tankHeated":"off","collectorShutdown":"off","energyLastHour":733.6,"energyToday":884,"energyTotal":4111.6}

proddy, 2020-11-06:
The pump working time is not collected for the SM100, only the SM10. Not sure why. Was it there before?

proddy, 2020-11-06:
The unit of measure was missing for the HA MQTT "Energy" configs. Thanks for finding that. Just to be sure the Last Hour is `wh` and the Today and Total `kwh`? You can also test yourself by re-publishing the same topic using MQTTExplorer and adding 
```
  "unit_of_meas": "kw",
``` 
to the JSON.

MichaelDvP, 2020-11-06:
Pump working time for SM100 is on Norberts list as ID 913 (0291), but a bit confusing: The time is noted as 4 bytes, but @Norberts1 describe it as calculation-type 2, i.e. a 3-byte value. We have to check which bytes to read. 
@PhillyGilly can you do a `read 30 391` and note the pump working time from the thermostat display?
@proddy energy today is also Wh, but without decimal (capacity for 10 hours).

PhillyGilly, 2020-11-06:
The units from v1.9 are: 
  name: 'Solar Energy Last Hour'
  unit_of_measurement: 'Wh'
  name: 'Solar Energy Today'
  unit_of_measurement: 'Wh'
   name: 'Solar Energy Total'
   unit_of_measurement: 'kWh'
 TBH I can't remember if it was working but it is displayed on my Sense II
[image]

MichaelDvP, 2020-11-06:
@PhillyGilly 4868h pumptime, now we have to look in `read 30 391` for the position of `04 74 F0`.

PhillyGilly, 2020-11-06:
> @PhillyGilly 4868h pumptime, now we have to look in `read 30 391` for the position of `04 74 F0`.
I haven't used the console yet and my Windows Telnet isn't happy:
Microsoft Telnet> o 192.168.2.101 22
Connecting To 192.168.2.101...Could not open connection to the host, on port 22: Connect failed
Microsoft Telnet>

PhillyGilly, 2020-11-06:
> @PhillyGilly 4868h pumptime, now we have to look in `read 30 391` for the position of `04 74 F0`.
> I haven't used the console yet and my Windows Telnet isn't happy:
> Microsoft Telnet> o 192.168.2.101 [...]

Error in the Wiki it's port 23. however I'm still lacking the password

MichaelDvP, 2020-11-06:
You can also use mqtt, just send to topic `system` `{"cmd":"send","data":"0B B0 FF 00 20 02 91"}`
and look to topic `response`
Default password for console su is `ems-esp-neo`

PhillyGilly, 2020-11-06:
ems-esp:/$ su
Password:
ems-esp:/# [K001+09:02:10.822 N 0: [shell] su session opened on console
ems-esp:/# read 30 391
ems-esp:/# [K001+09:02:32.212 N 1: [emsesp] Solar(0x30) -> Me(0x0B), ?(0x391), data: 00 04 75 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ems-esp:/# [K001+09:02:33.314 N 2: [emsesp] Solar(0x30) -> Me(0x0B), ?(0x391), data: 00 00 00 00 00 01 20 9C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 24)
ems-esp:/# [K001+09:02:34.258 N 3: [emsesp] Solar(0x30) -> Me(0x0B), ?(0x391), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 50)
ems-esp:/#

It has just past 12:00 here so the counter may have incremented if it's in hours.

MichaelDvP, 2020-11-06:
Counter is in minutes `04 75 1C` in pos 1, or as 4 bytes in pos 0, but i think we stick to pos1, times are mostly in 3 bytes and the capacity is large enough. I'll add it to solar.

PhillyGilly, 2020-11-06:
@MichaelDvP thank you

proddy, 2020-11-10:
@PhillyGilly close this issue?

PhillyGilly, 2020-11-11:
@MichaelDvP sorry I haven't seen any new code releases since you commented on this. Can you give me a link and I'll close once I've got the units and hours run.

PhillyGilly, 2020-11-11:
All good. Thankyou @proddy @MichaelDvP

closed 2020-11-11 by PhillyGilly