emsesp/EMS-ESP #599: Heat Curve options / CW400

lsw271, 2020-11-07:
**Is your feature request related to a problem? Please describe.**
No, it's not a problem, I would like change heat curve options by mqtt > HA

**Describe the solution you'd like**
possibility to change heat curve options by mqtt > HA

**Additional context**
curve flow temp set to 56 degrees
[image]

max curve flow temp set to 60 degrees
[image]

room influence set to 4 (min 1 / max 10)
[image]

room influence set to off
[image]

temperature correction set to 1 (min -10 / max 10)
[image]

minimum locally outside temperature set to -20 (min -35 / max 10)
[image]

MichaelDvP, 2020-11-08:
> room influence set to 4 (min 1 / max 10)

0x2AF, data 0x06

> room influence set to off

0x2AF, data 0x00

This is a strange stepping of 0.75 degree. According to Norbert there is a factor in pos 1, Can you verify some some other settings and do a `read 10 2AF` to see the whole telegram.

MichaelDvP, 2020-11-08:
I've added the parameters to dev, roominfluence for now as scale 1 (as it is in RC35). 
@lsw271 please check, i'm sure we need a few iterations to make all work.

glitter-ball, 2020-11-09:
0x2AF from my thermostat, in case that's useful...

```
ems-esp:/# read 10 2AF
003+00:04:57.271 N 1: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x2AF), data: 09 28 00 00 41 2D 11 01 19 1E 00
ems-esp:/#
```

lsw271, 2020-11-09:
more telegrams

[image]

lsw271, 2020-11-09:
> 
> 
> I've added the parameters to dev, roominfluence for now as scale 1 (as it is in RC35).
> lsw271 please check, i'm sure we need a few iterations to make all work.

@MichaelDvP after update to 2.1.1b1 I cannot log on with putty
[image]

I also had a problem with web interface. I tried to log on few times and I get some network error (_NetworkError when attempting to fetch resource._)

I saw this mqtt messages - its wird because I've updated firmware earlier and I shoudn't have message with 2.1.0
[image]

Now, when I backed to stable 2.1.0 version I had the same problem like in this issue https://github.com/proddy/EMS-ESP/issues/593 - the gateway was offline in mqtt explorer, I can log on to web interface but I couldn't saw sensors in devices. I needed to disconnect the jack cable from the boiler, wait few minutes and then everything works ok

lsw271, 2020-11-09:
I also have one more option "min curve flow temp" exactly curve starting point. I think we should call it _curve start point_ for min temp and _curve end point_ for higher temp (56 degrees above). I tried to catch the telegram
[image]

I have also got few more options in thermostat for heating curve, I need to consider which options would be great to change by ems-esp

MichaelDvP, 2020-11-10:
In the telegrams i can only see which value changes, but i also need to know what you set on the thermostat to have the factor.
Assuming you you set roominfluence in steps from 1 to 10 degree we have factor one, For the minflowtemp i see the value 14, assuming also factor one it's 14 degree, or with factor two 7 degree?

I get these "_NetworkError when attempting to fetch resource._"  sometimes if i try to load the webgage while the ems-esp isn't finished with booting. After update or reboot wait a minute before access web/telnet.

lsw271, 2020-11-10:
Something is wrong. I tried again to update firmware and it was the same behaviour despite the long waiting. Uptime_sec in mqtt is always 6s, its looks like the ems gateway is resetting. When I logged in to web interface I got this error "No EMS devices found. Check the connections and for possible Tx errors.". Then I disconnected the jack cable from ems gateway and wait about 15minutes. Unfortunatelly it was worse because I couldn't log on to web interface - I couldnt see ems website (problem with connection). so I couldn't back to stable version by web interface. After few times of disconnecting jack cable from ems-esp and after waiting few minutes managed to get into the interface and restore the previous version.

Any ideas? @MichaelDvP @proddy @bbqkees. My gateway is v.1.3. Maybe I need to update firmware by hand (esphome flasher or something). Can I use another wemos d1 mini, I don't want to screw up this one default from gateway

MichaelDvP, 2020-11-11:
Sorry, i have no idea what's going wrong. The dev-binary wasn't updated, you've flashed the same version as on your last try. 
I've triggered a new build a few minutes ago.

Sure you can use a second wemos, i've done that with esp32 and different 8266 on the start of v2 to switch fast to v1.9. 
If you set both wemos to the same hostname and mqtt settings and both are online (one in ems-esp, one standalone) with different ip, the mqtt broker sometimes get confused, a broker-restart solves this.

bbqkees, 2020-11-11:
@lsw271 you can just use a different Wemos or ESP on the Gateway board.

lsw271, 2020-11-12:
I tried one more time with OTA, it was the same :( at the weekend I will flash another wemos d1 mini and I hope will be ok.

MichaelDvP, 2020-11-13:
I think i have found the cause. With the extra parameters the HA-discovery takes to long and crashes after ~ 30sec. 
It's the same with RC35 which has also a lot of parameters published as HA-sensors. (I'm not using HA and proddys RC20 has less parameters)
In this 30 sec you have time to switch to another mqtt format to test the software or start OTA to go back. It's nothing wrong ith the wemos. I'm testing now with different timing an it seems to work.

proddy, 2020-11-13:
that's most probably true. I've been knee-deep these last days debugging the MQTT and TCP stack code to try and solve the memory fragmentation errors on the ESP8266. The same happens with `publish ha` and also running the tests on the ESP8266 with `test general`.

@lsw271 you could also temporarily shut down your MQTT broker and see if EMS-ESP starts up

MichaelDvP, 2020-11-13:
I have `publish ha` working, but a lot of small changes. In summary:
- sensor register 50 ms
- on ha register skip the large normal mqtt message 
- split boiler ha register in two parts (ww extra), register in two cycles
- register thermostat main and ha in different cycles.
- publish ha do not publish all devices at once, but one by one in 2 sec delay.

It takes now 1-2 minutes after startup for all mqtt done, but seems stable.
Should i make a PR or push direct?

MichaelDvP, 2020-11-13:
@lsw271 here is a dev-build that works with my RC35 and HA-discovery. Can you check with your thermostat? It includes the heat-curve.
[ems-esp_esp8266_v2.1.1b1.bin.zip](https://github.com/proddy/EMS-ESP/files/5537223/ems-esp_esp8266_v2.1.1b1.bin.zip)
(source is on my fork).

lsw271, 2020-11-13:
> 
> 
> lsw271 here is a dev-build that works with my RC35 and HA-discovery. Can you check with your thermostat? It includes the heat-curve.
> [ems-esp_esp8266_v2.1.1b1.bin.zip](https://github.com/proddy/EMS-ESP/files/5537223/ems-esp_esp8266_v2.1.1b1.bin.zip)
> (source is on my fork).

@MichaelDvP 
nothing helps, uptime seconds always is up to 6 s even with single or nested mqtt format. I tried to restart, plug and unplug jack cable. With nested or single format is more mqtt messages but uptime is really low and data show and disapper and over and over again

proddy, 2020-11-13:
I've moved the HA issue to a separate issue to track (#606) so we can focus on the CW400 heat curve

lsw271, 2020-11-13:
> I've moved the HA issue to a separate issue to track (#606) so we can focus on the CW400 heat curve

@proddy 
I would like to test dev version(options for heat curve) even without HA, only in telnet but my gateway always restaring, max uptime seconds its about 6 and telnet is not working

proddy, 2020-11-13:
yes, I know. but we ruled out an HA MQTT issue. Something else is causing the crash. Does it only crash when the telnet is active or all the time?

lsw271, 2020-11-13:
All the time

proddy, 2020-11-13:
so you had problems with the upload before, then it worked and now when you're trying again it's failing (with restarts). correct? What did you use to upload it?

lsw271, 2020-11-13:
Yes, correct. I'm doing update by webui

proddy, 2020-11-13:
hmm, ok. could you try using the command line as described in https://emsesp.github.io/docs/#/Uploading-firmware ?

MichaelDvP, 2020-11-14:
@lsw271 I saw that also the stable sometimes have this issue:
> Now, when I backed to stable 2.1.0 version I had the same problem like in this issue #593 - the gateway was offline in mqtt explorer, I can log on to web interface but I couldn't saw sensors in [...]

This looks like a power issue, have you tried to power the wemos from usb?

MichaelDvP, 2020-11-15:
@lsw271 maybe it is the same issue as in #610, please check with the new dev-build.

lsw271, 2020-11-17:
> 
> 
> lsw271 maybe it is the same issue as in #610, please check with the new dev-build.

Excellent! It is working now. Tomorrow I will do some tests with the heat curve options and give you the answer

lsw271, 2020-11-19:
@MichaelDvP Only offsettemp and minexttemp aren't working correctly. 

Offsettemp set to 1 by telnet, in cw400 changed to 2, offsettemp set to 2 by telnet, in cw400 changed to 4 so the factor is 2. 

The minexttemp doesn't work at all, I send the telegram again, changed it to -20
[image]

MichaelDvP, 2020-11-20:
Please try again. BTW: With `log info` you can see a command execution message in telnet or syslog.

lsw271, 2020-11-20:
@MichaelDvP 
Setting maxflowtemp by telnet is working now. But in mqtt explorer I saw values (maxflowtemp, nofrosttemp, heatingtype) only after call command. Earlier when I changed the maxflowtemp by CW400 I haven't received values in mqtt. It is possible that the values will be always displayed and not only after sending the command?

About offsettemp, there is something wrong with factor or rounding, belowe values which I setted and what I've got from CW and mqtt_explorer

offsettemp set by telnet | cw400 | mqtt_explorer
-- | -- | --
10 | 10 | 5
9 | 9 | 4
8 | 8 | 4
7 | 7 | 3
6 | 6 | 3
5 | 5 | 2
4 | 4 | 2
3 | 3 | 1
2 | 2 | 1
1 | 1 | 0
0 | 0 | 0

MichaelDvP, 2020-11-21:
Ok, the values are fetched now and offset factor is fixed.

lsw271, 2020-11-24:
> 
> 
> Ok, the values are fetched now and offset factor is fixed.

@MichaelDvP values are fetched now but commands called by telnet isn't working. Tested in 2.1.1b5
[image]

MichaelDvP, 2020-11-24:
All call-commands from console were broken, fixed now.

lsw271, 2020-11-26:
It's working now, thanks!

closed 2020-11-26 by lsw271