emsesp/EMS-ESP #600: Capacitor polarity on the schematic diagram

realthk, 2020-11-09:
The [second schematic](https://emsesp.github.io/docs/_media/schematics/Schematic_EMS-ESP.png) shows the 10 uF capacitor (C8) with reversed polarity than the [first diagram](https://emsesp.github.io/docs/_media/schematics/circuit.png).

Which should be the correct?

proddy, 2020-11-09:
never noticed that. The second one (mine) is correct. But you can also just use a ceramic cap and not worry about which way its placed.

proddy, 2020-11-09:
Also look at BBQKees' [circuit]( https://raw.githubusercontent.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/master/PCB-files/V0.9/EMS-V09_schema.png). It's old but works fine

realthk, 2020-11-09:
BBQKees probably designed that circuit with ceramic cap in mind, as no polarity is shown on his diagram - but at the local parts shop here, one can only buy SMD 10 uF ceramic caps which is not an option for assembling this on a test board, it has to be electrolytic type (or perhaps bipolar) 
Thanks, then the positive pin should be connected to the op-amp.

bbqkees, 2020-11-10:
Make sure its a 25V cap for robustness. In my old schematic I mislabeled all caps to 16V, which might fail as the bus is already 15,5V.

realthk, 2020-11-10:
And does the exact value of 10 uF has any significance, or is it just a decoupling cap, where twice the value would work just as good?
(I happen to have some 22 uF bipolar caps at home)

bbqkees, 2020-11-10:
22 uF might work fine as well, most components are not that strict.

closed 2020-12-28 by proddy