emsesp/EMS-ESP #601: circulation pump

Bartekn86, 2020-11-10:
Hi, 
is it possible to run hot water circulation pump by "the finger" ? i see that the rc310 have that option on the panel.
I readed somewhere that this is possible but cant fin it ;/

thank you

MichaelDvP, 2020-11-10:
I wanted to add this in #518, but nobody of the RC3x0 users seems to be interrested.
If you catch the telegrams and settings, we can add it.

Bartekn86, 2020-11-11:
could you give advice please how to do it ?

MichaelDvP, 2020-11-12:
It's simple: 
1. make sure you are using the latest [release](https://github.com/proddy/EMS-ESP/releases/tag/v2.1.0) or [dev](https://github.com/proddy/EMS-ESP/releases/tag/dev) software on the ems-esp
2. open a telnet session with logging. For Windows use [putty](https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html), go to `session->logging` and choose `printable output` and a filename. For linux you can use tee `telnet <ipaddress> | tee -a -i logfile.txt`  with the \<ipaddress\> of ems-esp. (for MAC i don't know). 
3. in telnet type `watch on`
4. go to your RC310 and change all the settings you want to have new in emsesp, i.e. circulation pump off/on/timer/own prog, circulation mode 1,2,3.. x3 min, circulation start
5. post a description exactly what you've changed (sequence and values) and add the logfle.
6. we search the changes in the log, identify the telegrams and positions of the values, and add it to ems-esp.

proddy, 2020-11-13:
@MichaelDvP nice write-up. We could add these steps to the documentation/trouble-shooting section

sq4bja, 2020-11-14:
Below I paste listing of telegrams when I set circulation pump on, off and set every timer 1 for 3 minutes / hour etc...
I think it helps. RC310 thermostat.
Tom

┌──────────────────────────────────────────┐
│ [1mEMS-ESP version 2.1.1b0   [22m               │
│ [0;92m[4mhttps://github.com/proddy/EMS-ESP[0m        │
│                                          │
│ type [4mhelp[0m to show available commands     │
└──────────────────────────────────────────┘

[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ wat[0G[K[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ watch on
Watching incoming telegrams, displayed in decoded format
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:16:50.105 N 0: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 00 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:16:51.828 N 1: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 34 01 F6 01 F6 21 00 05 03 00 01 07 E8 00 0C FC 00 80 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:16:52.068 N 2: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 0A 00 00 00 00 80 40 80 00 01 F6 80 00 00 00 FF 30 48 00 CB 00 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:16:52.352 N 3: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:16:55.883 N 4: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0B 15 0E 15 14 05 00 10 FF 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:16:59.627 N 5: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:16:59.692 N 6: [emsesp] Thermostat(0x10) -> All(0x00), RC300Floordry(0x267), data: 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:16:59.887 N 7: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 0A 04
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:00.413 N 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 16 2F 47
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:00.788 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 46 39 00 06 FA 14 01 0A 64 0A 02 00 00 00 00 00 00 00 00 00 00 00 00 1E 00 1E
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:01.652 N 10: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 1E 00 (offset 25)
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:01.714 N 11: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:02.061 N 12: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 34 01 F6 01 F6 21 00 05 03 00 01 07 E8 00 0C FC 00 80 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:02.286 N 13: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 0A 00 00 00 00 80 40 80 00 01 F6 80 00 00 00 FF 30 48 00 CB 00 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:02.698 N 14: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 57 80 00 80 00 00 00 00 00 00 16 90 05 5F 61 00 00 00 04 57 79 00 09 94 80 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:03.050 N 15: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 80 00 (offset 25)
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:03.411 N 16: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 34 FB 00 19 FF 07 46 D8 FF FF 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:03.889 N 17: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 DB 21 2A 00 00 2A 27 00 63 03 03 01 00 63 03 F3 00 00 11 01 03 08 8C 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:04.262 N 18: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 06 07 FF DC 00 00 FF 00 36 00 3C 01 FF 01 02 (offset 25)
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:05.171 N 19: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2A 26 27 02 40 FF FF 00 2C 02 F1 20 01 0F 05 00 00 01 22
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:05.710 N 20: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode(0x2F5), data: 01 FF 04 01 00 00 08 01 00 08 02 00 00 00 00 00 00 00 00 00 01
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:06.076 N 21: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: 3C 2D
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:06.452 N 22: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300OutdoorTemp(0x23A), data: 00 56
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:06.958 N 23: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Settings(0x240), data: 32 00 02 00 00 00 00 00 FF 02 EC 06 FF 00 00 00 00 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:11.758 N 24: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 0A 00 00 00 00 80 40 80 00 01 F6 80 00 00 00 FF 30 48 00 CB 00 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:11.977 N 25: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:17:12.241 N 26: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 34 01 F6 01 F6 21 00 05 03 00 01 07 E8 00 0C FC 00 80 00
[... cut ...]
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:19:31.357 N 163: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:19:31.654 N 164: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 34 01 F6 01 F6 21 00 05 03 00 01 07 E8 00 0C FC 00 80 00
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ [0G[K000+00:19:31.877 N 165: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 DD
[0;92m[1mems-esp[0m:[0;94m[1m/[0m$ exit
`

MichaelDvP, 2020-11-15:
>> post a description exactly what you've changed (sequence and values) 

> I set circulation pump on, off and set every timer 1 for 3 minutes / hour etc...

I think we have different opinions what exactly means....
Clear so far is, the settings 1x3min, ... are only send to the boiler (0x33, offset 7, already implemented) and not stored in thermostat. 
The other settings are in 0x2F5 (offset 1 and 3) and monitored by 0x31D, offset 3. But i can not figure out what parameter is set.

Please do this: 
- Note the setting, type `read 10 2F5`, 
- change the setting, not it, type `read 10 2F5`
- repeat for other settings. I need the description of the setting and the corresponding telegram.

tp1de, 2020-12-09:
@MichaelDvP @sq4bja  I may help for ww-circulation on RC310 thermostat

operating types: (all with circulation pump frequency set to 1x3 min / h) 
(1) off / (2) on / (3) like ww-program / (4) own time program 

**read 10 2F5**
(1) Thermostat(0x10) -> Me(0x0D), RC300WWmode(0x2F5), data: 01 FF 04 00 00 00 08 05 00 08 04 00 00 00 00 00 00 00 00 00 01
(2) Thermostat(0x10) -> Me(0x0D), RC300WWmode(0x2F5), data: 01 FF 04 01 00 00 08 05 00 08 04 00 00 00 00 00 00 00 00 00 01
(3) Thermostat(0x10) -> Me(0x0D), RC300WWmode(0x2F5), data: 01 FF 04 02 00 00 08 05 00 08 04 00 00 00 00 00 00 00 00 00 01
(4) Thermostat(0x10) -> Me(0x0D), RC300WWmode(0x2F5), data: 01 FF 04 03 00 00 08 05 00 08 04 00 00 00 00 00 00 00 00 00 01

The switch-on frquency is not stored in 2F5: (5) 2x3 min /h  and (6) 3x3 min / h
(5) Thermostat(0x10) -> Me(0x0D), RC300WWmode(0x2F5), data: 01 FF 04 03 00 00 08 05 00 08 04 00 00 00 00 00 00 00 00 00 01
(6) Thermostat(0x10) -> Me(0x0D), RC300WWmode(0x2F5), data: 01 FF 04 03 00 00 08 05 00 08 04 00 00 00 00 00 00 00 00 00 01

tp1de, 2020-12-09:
@MichaelDvP   What would be the equivalent mqtt command for read 10 2F5 on console?

MichaelDvP, 2020-12-09:
Thanks, i've already figured it and add to b7, `call thermostat wwcircmode x`  the frequncy is not sored in thermostat, it is sended to the boiler and the boiler command should work `call boiler wwcircmode x`.

> What would be the equivalent mqtt command for read 10 2F5 on console?

This can be done by raw command to system: `{"cmd":"send","data":"0D 90 FF 00 20 01 F5"}` the result is published in topic `response`.  With api you can send a raw command, but do not get a response, because a later response is not possible.

tp1de, 2020-12-09:
It is quite difficult to understand what is still open and what is already solved ....

But back on mqqt. Mqqt for reading data is working fine. But sending does not seam to work. I do not get any response. Yesterday I got a respnse some hours later ... What is wrong?

[image]

MichaelDvP, 2020-12-09:
> It is quite difficult to understand what is still open and what is already solved ....

It wasn't realy solved, it was a unconfirmed guess comparing some telegrams from people with and without circulationpump. Now it's confirmed. 

Can you check the response in terminal on `call system send "0D 90 FF 20 01 F5"`. If you get a response this is also published to mqtt. Also if you send raw from mqtt you will get the response also in terminal. You have set the own address to 0x0D (modem) i think. First byte has to be own address!

tp1de, 2020-12-09:
>Can you check the response in terminal on call system send "0D 90 FF 20 01 F5"

Within mqtt I see in the moment that once I send the command (onle one time) I get a response from ems-esp (same content) send every 60 seconds. Should this be?

MichaelDvP, 2020-12-09:
Sorry command was wrong : `call system send "0D 90 FF 00 20 01 F5"` This is like `read 10 2F5` and you get the complete telegram back. If you like to see a difference read only the first byte: `call system send "0D 90 FF 00 01 01 F5".

tp1de, 2020-12-09:
Same 60 secs loop in terminal and "command not found" displayed:

[image]

MichaelDvP, 2020-12-09:
For commands first set `su` password `esp-neo` (if you havn't changed it).

tp1de, 2020-12-09:
>For commands first set su password esp-neo (if you havn't changed it).
How do I change the su password (ems-esp-neo)? su with password works but then passwd is unknown.

..... and the 60 secs loop continues !!!!

MichaelDvP, 2020-12-09:
> How do I change the su password 

Web Security settings => Super User Password

tp1de, 2020-12-09:
I had changed the Super User Password already some time ago. Now I can't change it again. It has to be esp.neo?

.... and I have to go now with the dog for 30 minutes.

**By the way the loop sending responses still goes on: In the terminal and by mqtt (every 60 seconds a new message)**

tp1de, 2020-12-09:
Does any send command initiates repeating responses every 60 second? (terminal / console)
How can I stop it then?

MichaelDvP, 2020-12-10:
The 60 sec period is the normal fetch, but the repeated publish was a bug which is fixed now.  But you can also stop it by using a read command to a not fetched, unknown, notexisting  or long telegram.  i.g. `read 8 0`

tp1de, 2020-12-10:
The commands for setting wwcircmode work fine with web-api as well with mqtt - SUPER

tp1de, 2020-12-10:
@MichaelDvP 
My try to change the super user password failed. After changing the password a new login into admin was requested. I reloaded the browser and logged in. The password kept to be the same as before. This was the second change, since I changed it before.

As a result I can not login with su on the terminal anymore. Neither old nor new password is working.
The mqtt system comand is not working anymore. No response.
I tried a system restart, but still the same. **What shall I do?**

MichaelDvP, 2020-12-10:
I can not reproduce. Sure, if i change the su password i have to login to webpage again, that's what is written in the large blue box. 
Than the password for su is changed and in terminal it works. Other passwords for users/admin/guest/wifi/mqtt/AP are not effected, they have to be / can be changed seperatly.

tp1de, 2020-12-10:
I have a new finding: Changing the su pw in clear letters does not work. If I change in mode where entries are not shown, then the su password is changed for terminal su command. In web ui old password keeps unchanged .

proddy, 2020-12-17:
@tp1de I tested changing the secret/su password (With marked * and clear text) and couldn't reproduce what you're seeing

MichaelDvP, 2020-12-17:
The only thing i'm seeing is: with focus on the input field \<enter\> saves the password, klicking on the eye sets the focus to this and \<enter\> only switches clear text but not save the setting. It's always a good idea to use the `SAVE` button, than the password is always changed as expected.

tp1de, 2020-12-17:
@MichaelDvP I can not confirm.

When I change password in secret mode, then pw is changed for super user but still the old password is shown when I click on the eye. A paswword change with the eye open does not change anything anymore.

A pw change in secret mode works but I am not logged out.
A pw change with eye open does not work but I am logged out.

**Since the mqqt send command does not work anymore as well: Doe I have to set a special su pw?**

MichaelDvP, 2020-12-17:
Sorry, i can not reproduce an issue. Su-password change works as expected as clear text and hidden and mqtt send also.

tp1de, 2020-12-17:
So what shall I do?

tp1de, 2020-12-17:
... and any special su pw needed?

proddy, 2020-12-17:
@tp1de it would help if you record a video showing the error as both myself and Michael can't reproduce it. Also create a new GitHub issue or discussion for this, as this issue is about the circulation pump we added in 2.1.1!

tp1de, 2020-12-17:
@proddy Fine to close the issue. I hesitate with the video because it will show my password in clear text .... I will try to factory reset the whole system ....

closed 2020-12-17 by proddy