emsesp/EMS-ESP #602: No warmwater temp selection

og9gjz567, 2020-11-12:
Hi,
I've updated from the latest 1.9.x Version to the very nice looking  v2.1.1b1 ;)
I like the new UI and the mqtt autodiscover function. At the moment there is no warmwater entity:
[image]

Is there a easy way to get this back by yaml-code? This is my old code, but it doesn't work:

```
- platform: mqtt
  name: Warmwasser
  modes:
    - "auto"
    - "off"
  min_temp: 40
  max_temp: 60
  temp_step: 1
  
  current_temperature_topic: "ems-esp/boiler_data"
  temperature_state_topic: "ems-esp/boiler_data"
  mode_state_topic: "ems-esp/boiler_data"

  current_temperature_template: "{{ value_json.wWCurTmp }}"
  temperature_state_template: "{{ value_json.wWSelTemp }}"
  mode_state_template: "{% if value_json.wWActivated == 'off' %} off {% else %} auto {% endif %}"

  temperature_command_topic: "ems-esp/boiler_cmd"
  mode_command_topic: "ems-esp/boiler_cmd"
```

proddy, 2020-11-12:
It should be there. If you use MQTTExplorer you should see an topic `homeassistant/sensor/ems-esp/boiler_wWCurTemp`. Can you check that?

The `Mixer hc2 Current pump status` is strange. It shouldn't be showing as an MQTT Device. It's a bug we need to fix.

og9gjz567, 2020-11-12:
Hi,

thanks a lot. MQTT Explorer shows the topic:
[image]

og9gjz567, 2020-11-12:
Hi,
with the following yaml code it's working for now:

```
- platform: mqtt
  name: Warmwasser
  modes:
    - "auto"
    - "off"
  min_temp: 40
  max_temp: 60
  temp_step: 1
  
  current_temperature_topic: "ems-esp/boiler_data_ww"
  temperature_state_topic: "ems-esp/boiler_data_ww"
  mode_state_topic: "ems-esp/boiler_data_ww"

  current_temperature_template: "{{ value_json.wWCurTemp }}"
  temperature_state_template: "{{ value_json.wWSelTemp }}"
  mode_state_template: "{% if value_json.wWActivated == 'off' %} off {% else %} auto {% endif %}"

  temperature_command_topic: "ems-esp/boiler"
  mode_command_topic: "ems-esp/boiler"
```

proddy, 2020-11-12:
great. I'll add the yaml to the doc for others that need a thermostat like component to change the boiler temperatures.

glitter-ball, 2020-11-12:
This looks like the old DHW climate entity - yes? If so, I'll add it to my setup too. Thanks.

og9gjz567, 2020-11-13:
Sorry, I'm wrong. with my yaml I'm able to get the values, but I can't set them.
If I change the value in ha:
[image]
and in mqtt explorer:
[image]

If I use mqtt explorer to change the temp I use the topic **ems-esp/boiler** and the data: **{"cmd":"wwtemp", "data":"48"}**
Is there a way (template?) to get ha to use the working json?

proddy, 2020-11-13:
yes, i think there's an example in the doc?

og9gjz567, 2020-11-13:
Hi, I think this one is the only working way at the moment: [](https://community.home-assistant.io/t/enhanced-version-of-mqtt-hvac-climate-platform-with-proper-history-chart/75304/31)

HA doesn’t have a mode_command_template to allow you to convert from Home Assistant’s states to ems-esp states. So the only way is to publish to a fake topic, and then do anautomation which will change the "simple value" to an json, which can be understood by ems-esp.

proddy, 2020-11-13:
I mean there are some examples in the EMS-ESP documentation on how to send commands to EMS-ESP from HA (using json and value templates). There also some issues and threads in the gitter that show you to use the HA Climate component to change the boiler wwtemp. I don't have access now but let me know if you're unable to find them and I'll check later

proddy, 2020-11-14:
The Mixing should show correctly now in 2.1.1b2. Best wait for a new build before testing.

closed 2020-11-14 by proddy

NFis, 2020-12-30:
> Hi, I think this one is the only working way at the moment: https://community.home-assistant.io/t/enhanced-version-of-mqtt-hvac-climate-platform-with-proper-history-chart/75304/31
>
> HA doesn’t have a [...]

Hi,
my workaround for setting warmwater temperature in Home Assistant is to send the command "wwtemp" with the temperature value to ems-esp/boiler (my config is GB162/BC10/RC35): 

- use the climate "Warmwater" example (insert it in configuration.yaml) in this threat and replace the target of temperature_command_topic to a new unused mqtt-topic: 
temperature_command_topic: "temp/boiler_temp" (example)

- Restart HA and place climate.warmwater on a card in HA.
When you now change the setpoint temperature in the warmwater climate in HA the new value is published via the topic "temp/boiler_temp".

- create a new automation in HA GUI which is triggered by mqtt-topic "temp/boiler_temp" and then publishes command and value to the topic ems-esp/boiler:

**Name** 
Set_WW_Temperature (example, you name it)
**Triggers** 
Trigger type 
MQTT
Topic 
temp/boiler_temp
**Actions**
Action type 
Call service
Service 
mqtt.publish
Service data
topic: ems-esp/boiler
payload: '{"cmd":"wwtemp", "data":{{trigger.payload}},"id":1}'

or as yaml:

alias: Set_WW_Temperature
trigger:
  - platform: mqtt
    topic: temp/boiler_temp
action:
  - service: mqtt.publish
    data:
      topic: ems-esp/boiler
      payload: '{"cmd":"wwtemp", "data":{{trigger.payload}},"id":1}'
mode: single

Changing the the warmwater mode via HA has no effect (perhaps you have to address the thermostat).

proddy, 2020-12-30:
Thanks @NFis for sharing. Do you mind if I copy this and add to https://emsesp.github.io/docs/#/Howto ?

The Home Assistant Climate control is for HVAC (heating, ventilating, and air conditioning) devices such as thermostats. I did think of extending the EMS-ESP HA logic to treat MQTT messages from a Climate control purposely used to control the boiler, but it is a bit out of context and not for everyone. The solution using a trigger is the simplest. One day I may get around to building a custom HA component in Python that simplifies all of this.

NFis, 2020-12-30:
Hi Proddy,
first of all thank you for this great project, I use it since 1.6.0! Of course feel free to share the example. Your mqtt command interface is flexible enough to adapt it to everything, but an integration into a HA Climate Control would be the icing on the cake.

og9gjz567, 2020-12-30:
Yap. That works perfect!! Thanks ;)