emsesp/EMS-ESP #603: Pull all mqtt-topics and vars

og9gjz567, 2020-11-12:
Hi,
I've got a lot of shellies in my house. If I reboot my home assistant I publish a mqtt message with the topic shellies/announce, which will cause that every Shelly connected to the broker will respond with all available information.
Is there something similiar? Do you think it would be a nice feature?

proddy, 2020-11-12:
yes, that's a nice addition. There is a console command 'publish' which does the same so we can easily expose that as an API command, e.g. http://ems-esp/api?device=system&cmd=publish

og9gjz567, 2020-11-13:
Very cool. I'll add this to my ha start!

proddy, 2020-11-13:
you shouldn't need too since those MQTT topics have the retain flag. When HA reboots, it should just all work

proddy, 2020-11-14:
in 2.1.1b2.

closed 2020-11-14 by proddy