emsesp/EMS-ESP #604: flashing ESP32

misu1lica, 2020-11-13:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [documentation help section](https://emsesp.github.io/docs)*

*Completing this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information from http://ems-esp.local/api?device=system&cmd=report.*

**Additional context**
*Add any other context about the problem here.*

misu1lica, 2020-11-13:
Hi
I have ESP32 WROOM-32UE and I flash with "flash_download_tool_v3.8.5" version EMS-ESP-2_1_0-esp32.bin.
Programming is OK, but when I restart ESP32, the sistem don't have a Acces point esp-ems.
On the RS232 I see the butloader information.
What can I do?

proddy, 2020-11-13:
So no AP is seen in your wifi router? Also do you see any LEDs flashing?

I'm not familiar with the download tool. Do you have a link to it?

misu1lica, 2020-11-13:
The link: 

https://www.espressif.com/sites/default/files/tools/flash_download_tool_v3.8.5.zip
Yes, I don't seen AP.
This module don't have LED on board, I try to put one.

misu1lica, 2020-11-16:
Hi,
 the LED does not flashing. 
The program can't program ESP32 WROOM-32UE?
I try to program with uPyCraft_V1.1.exe  and the same, the ESP did not have AP

On serial I read this:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

proddy, 2020-11-16:
If you build the standard 'blink' app does it work?

If you're new to building and uploading, start with PlatformIO and pick the right flash mode for your board (dio/dout).

misu1lica, 2020-11-16:
I programm esp8266 and all is OK.
But with esp32 don't work.

proddy, 2020-11-16:
I can't help you without more information, sorry.

misu1lica, 2020-11-17:
Hi
use ESPHome-Flasher-1.2.0-windows-x64 , https://github.com/esphome/esphome-flasher/releases work for ESP32!
But i must put in flash mode my ESP32 but press RST+BOOT, then release RST and after 1 secund relase BOOT.
And another schematic
[image]
misu

closed 2020-11-17 by misu1lica

proddy, 2020-11-17:
@misu1lica impressive design. Can you share some photos and also how you intend to use it? specially those photocouplers

misu1lica, 2020-11-17:
I started with the motherboard and now I'm starting to program ESP.
For OK4 I want to monitor the boiler supply voltage (230Vac). We put power plants on the UPS - we still have voltage interruptions.
For OK3 I want to start a boiler.
For OK5 voltage drop alarm ...