emsesp/EMS-ESP #605: Thermostat Junkers FR50 not responding to manipulation commands on HT3 bus

Oderik, 2020-11-13:
**Bug description**
My Junkers FR50 thermostat cannot be controlled by EMS-ESP. Commands like temp and mode do not result in an actual change of the thermostat settings.

**Expected behavior**
Commands like temp and mode should change the corresponding thermostat settings.

**Terminal session**
```
┌──────────────────────────────────────────┐
│ EMS-ESP version 2.1.0                    │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

ems-esp:/$ su
Password:
000+00:23:11.672 N 0: [shell] su session opened on console
ems-esp:/# watch on
Watching incoming telegrams, displayed in decoded format
ems-esp:/# log debug
Log level = debug
000+00:23:19.909 N 1: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 38 01 30 4C 00 01 23 20 C0 01 71 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:23:19.909 D 2: [emsesp] Received UBAMonitorFast
000+00:23:23.309 N 3: [emsesp] Boiler(0x08) -> 13(0x13), ?(0x05), data: 00 (offset 34)
ems-esp:/# call thermostat info
{
  "time": "13:21:05 13/11/2020",
  "hc1": {
    "seltemp": 20.5,
    "currtemp": 17.9,
    "mode": "manual",
    "modetype": "heat"
  }
}
000+00:23:23.617 N 4: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 38 01 30 4C 00 01 23 20 C0 01 71 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:23:23.617 D 5: [emsesp] Received UBAMonitorFast
000+00:23:23.857 N 6: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 32 01 71 80 00 81 00 00 01 00 00 E5 03 00 A8 50 00
000+00:23:23.857 D 7: [emsesp] Received UBAMonitorWW
ems-esp:/# call thermostat temp 14
000+00:23:30.077 I 8: [thermostat] Setting thermostat temperature to 14.0 for heating circuit 1, mode auto
000+00:23:30.094 D 9: [telegram] Sending write Tx [#167], telegram: 8B 10 FF 11 00 65 1C E2
000+00:23:30.122 D 10: [emsesp] Last Tx write successful
000+00:23:30.122 D 11: [telegram] Sending post validate read, type ID 0x16F to dest 0x10
000+00:23:30.257 D 12: [telegram] Sending read Tx [#168], telegram: 8B 90 FF 00 20 00 6F DA
000+00:23:30.625 D 13: [telegram] Sending read Tx [#169], telegram: 8B 90 FF 00 20 00 6F DA
000+00:23:30.681 D 14: [emsesp] Last Tx read successful
000+00:23:30.685 N 15: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersMonitor(0x16F), data: 03 01 00 CD 00 B3
000+00:23:30.685 D 16: [emsesp] Received JunkersMonitor
000+00:23:30.756 N 17: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0B 0D 0D 16 02 05 00
000+00:23:30.756 D 18: [emsesp] Received RCTime
000+00:23:33.256 N 19: [emsesp] Boiler(0x08) -> 13(0x13), ?(0x05), data: 00 (offset 34)
000+00:23:33.558 N 20: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 38 01 2F 4C 00 01 23 20 C0 01 71 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:23:33.558 D 21: [emsesp] Received UBAMonitorFast
000+00:23:33.772 N 22: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 32 01 71 80 00 81 00 00 01 00 00 E5 03 00 A8 50 00
000+00:23:33.772 D 23: [emsesp] Received UBAMonitorWW
ems-esp:/# call thermostat info
{
  "time": "13:22:02 13/11/2020",
  "hc1": {
    "seltemp": 20.5,
    "currtemp": 17.9,
    "mode": "manual",
    "modetype": "heat"
  }
}
000+00:23:43.252 N 24: [emsesp] Boiler(0x08) -> 13(0x13), ?(0x05), data: 00 (offset 34)
000+00:23:43.560 N 25: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 38 01 2D 4C 00 01 23 20 C0 01 71 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:23:43.560 D 26: [emsesp] Received UBAMonitorFast
000+00:23:43.800 N 27: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 32 01 71 80 00 81 00 00 01 00 00 E5 03 00 A8 50 00
000+00:23:43.800 D 28: [emsesp] Received UBAMonitorWW
ems-esp:/#
```

**Device information**
``` json
{
  "System": {
    "version": "2.1.0",
    "uptime": "000+00:14:43.993",
    "freemem": 31,
    "fragmem": 2
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 0,
    "publish_time_thermostat": 0,
    "publish_time_solar": 0,
    "publish_time_mixer": 0,
    "publish_time_other": 0,
    "publish_time_sensor": 0,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 3,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "on",
    "api_enabled": "off",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 650,
    "#read requests sent": 107,
    "#write requests sent": 0,
    "#incomplete telegrams": 10,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:18.06)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "Junkers FR50 (DeviceID:0x10 ProductID:147, Version:18.02)",
      "handlers": "0xA3 0x06 0xA2 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:18.06)",
      "handlers": ""
    }
  ]
}
```

**Additional context**
This was also briefly discussed at Gitter: https://gitter.im/EMS-ESP/community?at=5fae339d8a236947baa1516f

MichaelDvP, 2020-11-13:
Can you please do a `read 10 165`  and `read 10 179` and post the output. 
There are two Junkers variants and maybe this is matched wrong.

Oderik, 2020-11-13:
```
ems-esp:/# read 10 165
000+01:08:36.557 N 3: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x165), data: <empty>
ems-esp:/# read 10 179
000+01:08:39.758 N 4: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x179), data: 01 04 10 38 03 0A 20 29
```

MichaelDvP, 2020-11-13:
Thanks, this thermostat uses telegram 179, i change it in the database.

Oderik, 2020-11-13:
> Thanks, this thermostat uses telegram 179, i change it in the database.

So is this probably the underlying issue or is it just something additional? 😅

MichaelDvP, 2020-11-13:
This is the issue. The setpoint temperature is in byte 7 of 179 (last byte 0x29 = 41 => 20.5°C) 
You can test with a write to 179: `send telegram "0B 10 FF 07 00 79 20"` set the setpoint to 16°C

Oderik, 2020-11-13:
You are right, it works.

Actually it did not work the two or three times I fired the command but now the temperature changed. Thanks! What do you think how long it will take until a fix is released?

MichaelDvP, 2020-11-13:
I add it with the next dev-update. I have some other changes and waiting for proddys feedback. But i posted a test-buid to #599 which also includes the database update.

Oderik, 2020-11-13:
Installed you test bould and BAM I can control the FR50 from Home Assistant. That's totally awesome, thank you very much!

Should the ticket be closed now then or do we wait for something else like release?

proddy, 2020-11-13:
thanks Michael! super quick as usual...

MichaelDvP, 2020-11-14:
> Should the ticket be closed now then or do we wait for something else like release?

It's now in the [dev build](https://github.com/proddy/EMS-ESP/releases/tag/dev) so this can be closed. If you miss other functions feel free to open a new issue.

commit c2536d8b (2020-11-13): Junkers FR50 as device-type Junker2, fix #605
closed 2020-11-14 by MichaelDvP