emsesp/EMS-ESP #606: refactor how MQTT messages are sent to HA during startup

proddy, 2020-11-13:
I have `publish ha` working, but a lot of small changes. In summary:
- sensor register 50 ms
- on ha register skip the large normal mqtt message 
- split boiler ha register in two parts (ww extra), register in two cycles
- register thermostat main and ha in different cycles.
- publish ha do not publish all devices at once, but one by one in 2 sec delay.

It takes now 1-2 minutes after startup for all mqtt done, but seems stable.
Should i make a PR or push direct?

_Originally posted by @MichaelDvP in https://github.com/proddy/EMS-ESP/issues/599#issuecomment-726725575_

proddy, 2020-11-13:
@MichaelDvP commented 
> I think i have found the cause. With the extra parameters the HA-discovery takes to long and crashes after ~ 30sec.
It's the same with RC35 which has also a lot of parameters published as HA-sensors. (I'm not using HA and proddys RC20 has less parameters)
In this 30 sec you have time to switch to another mqtt format to test the software or start OTA to go back. It's nothing wrong ith the wemos. I'm testing now with different timing an it seems to work."

and @proddy replied

> that's most probably true. I've been knee-deep these last days debugging the MQTT and TCP stack code to try and solve the memory fragmentation errors on the ESP8266. The same happens with publish ha and also running the tests on the ESP8266 with test general."

proddy, 2020-11-13:
I've been refactoring the MQTT codes (as mentioned above) to reduce the memory fragmentation and heap which is killing the ESP8266 when sending 70+ publish messages.

@MichaelDvP I've haven't looked at your PR yet but be careful if using the delay() as it can cause a watchdog reset. The correct way is to put the publish messages on the MQTT queue and let the service handle them, which goes each 200ms...but then it opens up other issues. Anyway, still experimenting....

MichaelDvP, 2020-11-13:
Queue deep was set to 70, but using this crashes due to memory, also it's not enough to hold all the boiler sensors. But a working queue deep of 20 is not usefull for the lot of ha-sensor-messages. 
The messages are small and should fit into a single tcp-package. This is why i reduced the delay and split the sensor publishes in logical blocks (boiler, boiler_ww, thermostat, thermostat_hc). The complete delay for a block is around 1 sec, more than factor 10 below the actual dev.  I've tested `publish ha` with telnet open and log debug. Some debug messages get lost (queue full), but mqtt explorer recieves all publishes and ems-esp does not crash. 

It's a small change based on the actual code to make it work with the additional publishes. A complete refactor of the publish requires a lot of changes. And i don't have a good idea how to manage this. Add only one sensor per program loop with indexing?

MichaelDvP, 2020-11-13:
Here is the log from the publish, the dallas are now also included.
[20201113_145749_EMS.log](https://github.com/proddy/EMS-ESP/files/5538216/20201113_145749_EMS.log)

proddy, 2020-11-14:
@MichaelDvP I noticed in your last changes you opted to use StaticJsonDocument, which is nice as it doesn't use the heap. Last time I was talking to Benoit (arduinojson author) he recommended only to use StaticJsonDocument for any documents < 1024 bytes as it sits on the stack which is precious memory. Anything larger may cause crashes, but obvioulsy at a cost of heap issues. If it works, let's leave it, but just wanted to let you know some of my previous design principles when choosing between dynamicjson and staticjson.

MichaelDvP, 2020-11-14:
Since the boiler is splited into 2 parts now, each publish should be less than 1024. 
Should we make a EMSESP_MAX_JSON_SIZE_STATIC 1024 and use this? Than larger json are cut, but not crashing.
I worry about the thermostat json, if someone uses up to 4 heatingcircuits the nested format grows with all the curve-parameters.
But he can use single. 

The register_mqtt_ha_sensor uses Dynamic, because shrinktofit seems to create smaller packages, but it may be better to make a static json with optimal size? But the ha-register is only on start every call with same size and never again, it should not make much fragmentation.

MichaelDvP, 2020-11-14:
I've checked the thermostat mqtt for my RC35, i think this is also the thermostat with most parameters. 
The main is 182 bytes long and 279 bytes per heating circuit. Maybe some bytes more depending on states. 
A static json with max 1024 will fit for max. 3 nested circuits.  

Wait for the complaint of the first user with RC3xx and four circuits? (And tell him to use single format). Or change to dynamic?

proddy, 2020-11-15:
I said wait, use static for the HA ones and all the reset (i.e. thermostat_data) can be dynamic since they're queued anyway

proddy, 2020-11-15:
making a few minor changes now

proddy, 2020-11-16:
> I've checked the thermostat mqtt for my RC35, i think this is also the thermostat with most parameters.
> The main is 182 bytes long and 279 bytes per heating circuit. Maybe some bytes more depending [...]

@MichaelDvP we have to build in some fail-safe checks to prevent buffer overruns and the EMS-ESP crashing. It's a nightmare when someone buys one of BBQKees's boards, plugs it in and then gets constant restarts.

proddy, 2020-11-16:
BTW I use this link a lot to calculate size of the payloads: https://arduinojson.org/v6/assistant/

closed 2020-11-17 by proddy