emsesp/EMS-ESP #608: Commands refresh and publish

deejaybeam, 2020-11-13:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [documentation help section](https://emsesp.github.io/docs)*

*Completing this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information from http://ems-esp.local/api?device=system&cmd=report.*

**Additional context**
*Add any other context about the problem here.*

I'm using v1.9.5 and like this project very much! - awesome!
Now I want to upgrade the existing version from 1.9.5 to 2.1 with pure mqtt integration to v2.1
One thing I'm missing is the commands to update/publish mqtt messages.
For now I use a homeassistant script to change e.g. thermostat_cmd_mode2 to auto like this:
```
'ems_hc2_auto':
  alias: HC2 auto
  sequence:
    - service: mqtt.publish
      data:
        topic: 'ems-esp/thermostat_cmd_mode2'
        payload: 'auto'
    - service: mqtt.publish
      data:
        topic: 'ems-esp/generic_cmd'
        payload: '{cmd:"refresh"}'
    - service: mqtt.publish
      data:
        topic: 'ems-esp/generic_cmd'
        payload: '{cmd:"publish"}'
```
...instantly after pushing the button in the gui.
in the documentation I don't find anything about that for the new (2.1) version.
I prefer to have messages about stats with fixed intervals so I force an update after change.

Is this possible with v2.x also?

My integration looks like this:
[image]

because of WAF (WomanAcceptanceFactor) it looks like the manual remote control...
Now - If you push a button you see an instant change.

thanks for building this awesome project!! - chris

proddy, 2020-11-13:
The publish command is not exposed to the API but this will be done soon (there's already an issue open for this at #381). To list the commands you can type `show commands` from the console.

When it's implemented the payload would look like `{"cmd":"publish"}` to the topic `ems-esp/system`. See https://emsesp.github.io/docs/#/MQTT?id=sending-commands

It's a small change so should be done soon. Watch out for new builds on https://github.com/proddy/EMS-ESP/releases/tag/dev

BTW did you know in v2 you can control the frequency of the MQTT publishes for each device/sensor. Or have it automatic so it publishes on every change.

deejaybeam, 2020-11-14:
Wow - thanks for the fast response!
No, I did´t know that it´s possible to set an indiviual publish frequency setting for each sensor/device.
That sounds very interesting. So it could be possible to set a frequency of e.g. 60 second for each device, but an immediatly response only for the thermostat (auto/day/night) to have an visual feedback in the gui? 
This would be great!
I will have a look...
Thanks again for this awesome project!

MichaelDvP, 2020-11-14:
The `publish` command #603 is already in the latest dev, the refresh is now called `fetch` and works only from console, but easy to implement.  
@deejaybeam:  But you don't need your script anymore. Since #477 ems-esp do this automatically:
After each write the value is fetched and published out of order. 

@proddy: there are now some commands  double, as console command and as call. i.e. `send telegram "0B 88 18 00 20"` and `call system send "0B 88 18 00 20"` works identical, also `publish` and `call system publish`. Remove the console only commands?

proddy, 2020-11-14:
yes, no need for the console commands 'send' and 'publish' so these can be removed. Also while at it extend the fetch and publish as system commands alongside info, report and pin.

Do you want to make the change? My environment is in a bit of a mess, using a JTAG debugger to understand lwip TCP buffers to improve the MQTT heap issues...

And remember to add to the documentation :)

MichaelDvP, 2020-11-14:
> Do you want to make the change? 

ok.
> And remember to add to the documentation :)

Also done.

proddy, 2020-11-14:
not removing publish and send telegram from console? I don't mind either way

proddy, 2020-11-14:
> not removing publish and send telegram from console? I don't mind either way

sorry, didn't pull latest. all good

closed 2020-11-14 by proddy

deejaybeam, 2020-11-15:
Works like a charm!