emsesp/EMS-ESP #609: Home Assistant lacking boiler sensors with development builds 2.1.1

Oderik, 2020-11-14:
**Bug description**
Using current development builds (v2.1.1b1 or v2.1.1b2) Home Assistant integration of EMS-ESP devices behaves strange. The boiler's name and other meta data is not recognized and some sensors are missing (including the current tap water flow).

Going back to firmware v2.1.0 everything works as expected.

**Steps to reproduce**
Install firmware v2.1.1b1 or v2.1.1b2, connect to my boiler model, observe Home Assistant.

**Expected behavior**
Boiler recognition in Home Assistant works like in version v2.1.0.

**Screenshots**
Please request if needed!

**Device information**
Copied from https://github.com/proddy/EMS-ESP/issues/605
```
{
  "System": {
    "version": "2.1.0",
    "uptime": "000+00:14:43.993",
    "freemem": 31,
    "fragmem": 2
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 0,
    "publish_time_thermostat": 0,
    "publish_time_solar": 0,
    "publish_time_mixer": 0,
    "publish_time_other": 0,
    "publish_time_sensor": 0,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 3,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "on",
    "api_enabled": "off",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 650,
    "#read requests sent": 107,
    "#write requests sent": 0,
    "#incomplete telegrams": 10,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:18.06)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "Junkers FR50 (DeviceID:0x10 ProductID:147, Version:18.02)",
      "handlers": "0xA3 0x06 0xA2 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:18.06)",
      "handlers": ""
    }
  ]
}
```

**Additional context**
I guess this is related to https://github.com/proddy/EMS-ESP/issues/606

MichaelDvP, 2020-11-14:
Then sensor-json SMALL is not enough for longer boiler-names and values. I'll change it.

MichaelDvP, 2020-11-14:
Please check if it is fixed now.

Oderik, 2020-11-14:
I don't have a build environment set up. Can you provide me with a precompiled firmware (esp8266)?

Also, thanks for the quick fixing again! 👍

proddy, 2020-11-14:
@Oderik have a look at the https://emsesp.github.io/docs/#. It explains where to get the dev builds.

MichaelDvP, 2020-11-14:
@proddy seems that the path has changed. 
The doc links to [https://github.com/proddy/EMS-ESP/releases/tag/dev](https://github.com/proddy/EMS-ESP/releases/tag/dev), but the bin is now placed to [https://github.com/proddy/EMS-ESP/releases](https://github.com/proddy/EMS-ESP/releases)

Oderik, 2020-11-14:
... and both of them don't provide any development binaries. 🤓

proddy, 2020-11-14:
yes, just noticed as well. something is wrong with the build scripts. Trying to fix...

MichaelDvP, 2020-11-14:
> ... and both of them don't provide any development binaries. 🤓

In [https://github.com/proddy/EMS-ESP/releases](https://github.com/proddy/EMS-ESP/releases) i can see a dev-build 2.1.1b2, the assets link to a b2 [https://github.com/proddy/EMS-ESP/releases/download/untagged-ea3f664060ea7a996694/EMS-ESP-2_1_1b2-esp8266.bin](https://github.com/proddy/EMS-ESP/releases/download/untagged-ea3f664060ea7a996694/EMS-ESP-2_1_1b2-esp8266.bin). But i can't see from which commit.

Oderik, 2020-11-14:
Ok, found the binary. Now all sensors are back (and also some previously missing values of the FR50 thermostat).
But now Home Assistant has trouble manipulating the thermostat again. This time it's about the heating circuit...
`[thermostat] Set temperature: Heating Circuit 207 not found or activated for device ID 0x10`

MichaelDvP, 2020-11-14:
What has been sent? I only get this if i send `call thermostat temp 20 207` (or other temperature as 20)
Is it from HA-discovery or do you set the commands yourself? 
Can you take a `log debug` and `watch 10` to see whats on the bus.

Oderik, 2020-11-14:
It's HA-discovery.

Mode setting seems to work. When I set temperatures via the HA UI:
```
0+00:02:38.033 D 88: [mqtt] Publishing topic ems-esp/boiler_data (#142, attempt #1, size 485, pid 1)
000+00:02:38.234 D 89: [mqtt] Publishing topic ems-esp/boiler_data_ww (#143, attempt #1, size 408, pid 1)
000+00:02:39.378 D 90: [emsesp] Received UBAMonitorFast
000+00:02:39.440 D 91: [mqtt] Publishing topic ems-esp/boiler_data (#144, attempt #1, size 485, pid 1)
000+00:02:39.619 D 92: [emsesp] Received UBAMonitorWW
000+00:02:39.641 D 93: [mqtt] Publishing topic ems-esp/boiler_data_ww (#145, attempt #1, size 408, pid 1)
000+00:02:40.313 W 94: [thermostat] Set temperature: Heating Circuit 168 not found or activated for device ID 0x10
000+00:02:40.313 W 95: [thermostat] Set temperature: Heating Circuit 168 not found or activated for device ID 0x10
000+00:02:44.053 D 96: [emsesp] Received UBAMonitorFast
000+00:02:44.067 D 97: [mqtt] Publishing topic ems-esp/boiler_data (#146, attempt #1, size 485, pid 1)
000+00:02:44.269 D 98: [mqtt] Publishing topic ems-esp/boiler_data_ww (#147, attempt #1, size 408, pid 1)
000+00:02:49.351 D 99: [emsesp] Received UBAMonitorFast
000+00:02:49.497 D 100: [mqtt] Publishing topic ems-esp/boiler_data (#148, attempt #1, size 485, pid 1)
000+00:02:49.590 D 101: [emsesp] Received UBAMonitorWW
000+00:02:49.698 D 102: [mqtt] Publishing topic ems-esp/boiler_data_ww (#149, attempt #1, size 408, pid 1)
000+00:02:50.343 W 103: [thermostat] Set temperature: Heating Circuit 221 not found or activated for device ID 0x10
000+00:02:50.343 W 104: [thermostat] Set temperature: Heating Circuit 221 not found or activated for device ID 0x10
000+00:02:59.323 D 105: [emsesp] Received UBAMonitorFast
000+00:02:59.354 D 106: [mqtt] Publishing topic ems-esp/boiler_data (#150, attempt #1, size 485, pid 1)
000+00:02:59.555 D 107: [mqtt] Publishing topic ems-esp/boiler_data_ww (#151, attempt #1, size 410, pid 1)
000+00:02:59.559 D 108: [emsesp] Received UBAMonitorWW
000+00:02:59.756 D 109: [mqtt] Publishing topic ems-esp/boiler_data (#152, attempt #1, size 485, pid 1)
000+00:02:59.957 D 110: [mqtt] Publishing topic ems-esp/boiler_data_ww (#153, attempt #1, size 412, pid 1)
000+00:03:00.003 D 111: [emsesp] Fetching values for device ID 0x08
000+00:03:00.003 D 112: [telegram] Tx read request to device 0x08 for type ID 0x14
000+00:03:00.003 D 113: [telegram] Tx read request to device 0x08 for type ID 0x16
000+00:03:00.003 D 114: [telegram] Tx read request to device 0x08 for type ID 0x19
000+00:03:00.003 D 115: [telegram] Tx read request to device 0x08 for type ID 0x33
000+00:03:00.003 D 116: [emsesp] Fetching values for device ID 0x09
000+00:03:00.003 D 117: [emsesp] Fetching values for device ID 0x10
000+00:03:00.003 D 118: [telegram] Tx read request to device 0x10 for type ID 0x16F
000+00:03:00.003 D 119: [telegram] Tx read request to device 0x10 for type ID 0x179
000+00:03:00.068 D 120: [telegram] Sending read Tx [#45], telegram: 8B 88 14 00 20 2C
000+00:03:00.107 D 121: [emsesp] Last Tx read successful
000+00:03:00.107 D 122: [emsesp] Received UBATotalUptime
000+00:03:00.158 D 123: [mqtt] Publishing topic ems-esp/boiler_data (#154, attempt #1, size 485, pid 1)
000+00:03:00.286 D 124: [telegram] Sending read Tx [#46], telegram: 8B 88 16 00 20 24
000+00:03:00.336 D 125: [emsesp] Last Tx read successful
000+00:03:00.336 D 126: [emsesp] Received UBAParameters
000+00:03:00.359 D 127: [mqtt] Publishing topic ems-esp/boiler_data_ww (#155, attempt #1, size 412, pid 1)
000+00:03:00.448 D 128: [telegram] Sending read Tx [#47], telegram: 8B 88 19 00 20 18
000+00:03:00.509 D 129: [emsesp] Last Tx read successful
000+00:03:00.509 D 130: [emsesp] Received UBAMonitorSlow
000+00:03:00.560 D 131: [mqtt] Publishing topic ems-esp/boiler_data (#156, attempt #1, size 485, pid 1)
000+00:03:00.629 D 132: [telegram] Sending read Tx [#51], telegram: 8B 88 19 19 20 2A
000+00:03:00.667 D 133: [emsesp] Last Tx read successful
000+00:03:00.667 D 134: [emsesp] Received UBAMonitorSlow
000+00:03:00.761 D 135: [mqtt] Publishing topic ems-esp/boiler_data_ww (#157, attempt #1, size 412, pid 1)
000+00:03:00.847 D 136: [telegram] Sending read Tx [#48], telegram: 8B 88 33 00 20 B0
000+00:03:00.897 D 137: [emsesp] Last Tx read successful
000+00:03:00.897 D 138: [emsesp] Received UBAParameterWW
000+00:03:01.033 D 139: [telegram] Sending read Tx [#49], telegram: 8B 90 FF 00 20 00 6F DA
000+00:03:01.084 D 140: [emsesp] Last Tx read successful
000+00:03:01.084 D 141: [emsesp] Received JunkersMonitor
000+00:03:01.163 D 142: [mqtt] Publishing topic ems-esp/heartbeat (#158, attempt #1, size 141, pid 1)
000+00:03:01.194 D 143: [telegram] Sending read Tx [#50], telegram: 8B 90 FF 00 20 00 79 CC
000+00:03:01.252 D 144: [emsesp] Last Tx read successful
000+00:03:01.252 D 145: [emsesp] Received JunkersSet
000+00:03:02.564 D 146: [emsesp] Received UBAMonitorFast
000+00:03:02.576 D 147: [mqtt] Publishing topic ems-esp/boiler_data (#159, attempt #1, size 486, pid 1)
```

Oderik, 2020-11-14:
It worked once, then again there was an error:
```
000+00:05:43.733 D 404: [mqtt] Publishing topic ems-esp/boiler_data_ww (#286, attempt #1, size 413, pid 1)
000+00:05:48.844 D 405: [emsesp] Received UBAMonitorFast
000+00:05:48.960 D 406: [mqtt] Publishing topic ems-esp/boiler_data (#287, attempt #1, size 486, pid 1)
000+00:05:49.084 D 407: [emsesp] Received UBAMonitorWW
000+00:05:49.162 D 408: [mqtt] Publishing topic ems-esp/boiler_data_ww (#288, attempt #1, size 413, pid 1)
000+00:05:49.363 D 409: [mqtt] Publishing topic ems-esp/boiler_data (#289, attempt #1, size 486, pid 1)
000+00:05:49.564 D 410: [mqtt] Publishing topic ems-esp/boiler_data_ww (#290, attempt #1, size 413, pid 1)
000+00:05:52.770 D 411: [emsesp] Received UBAMonitorFast
000+00:05:52.783 D 412: [mqtt] Publishing topic ems-esp/boiler_data (#291, attempt #1, size 486, pid 1)
000+00:05:52.984 D 413: [mqtt] Publishing topic ems-esp/boiler_data_ww (#292, attempt #1, size 413, pid 1)
000+00:05:55.996 I 414: [thermostat] Setting thermostat temperature to 19.0 for heating circuit 1, mode auto
000+00:05:55.996 I 415: [thermostat] Setting thermostat temperature to 19.0 for heating circuit 1, mode auto
000+00:05:56.040 D 416: [telegram] Sending write Tx [#66], telegram: 8B 10 FF 07 00 79 26 50
000+00:05:56.077 D 417: [emsesp] Last Tx write successful
000+00:05:56.077 D 418: [telegram] Sending post validate read, type ID 0x16F to dest 0x10
000+00:05:56.142 D 419: [emsesp] Received JunkersSet
000+00:05:56.202 D 420: [mqtt] Publishing topic ems-esp/thermostat_data (#293, attempt #1, size 142, pid 1)
000+00:05:56.414 D 421: [telegram] Sending read Tx [#68], telegram: 8B 90 FF 00 20 00 6F DA
000+00:05:56.464 D 422: [emsesp] Last Tx read successful
000+00:05:56.464 D 423: [emsesp] Received JunkersMonitor
000+00:05:56.514 D 424: [emsesp] Received JunkersMonitor
000+00:05:56.604 D 425: [mqtt] Publishing topic ems-esp/thermostat_data (#294, attempt #1, size 142, pid 1)
000+00:05:56.788 D 426: [telegram] Sending write Tx [#67], telegram: 8B 10 FF 07 00 79 26 50
000+00:05:57.162 D 427: [telegram] Sending write Tx [#69], telegram: 8B 10 FF 07 00 79 26 50
000+00:05:57.198 D 428: [emsesp] Last Tx write successful
000+00:05:57.296 D 429: [emsesp] Received JunkersSet
000+00:05:58.815 D 430: [emsesp] Received UBAMonitorFast
000+00:05:58.820 D 431: [mqtt] Publishing topic ems-esp/boiler_data (#295, attempt #1, size 486, pid 1)
000+00:05:59.021 D 432: [mqtt] Publishing topic ems-esp/boiler_data_ww (#296, attempt #1, size 413, pid 1)
000+00:05:59.056 D 433: [emsesp] Received UBAMonitorWW
000+00:06:00.006 D 434: [emsesp] Fetching values for device ID 0x08
000+00:06:00.006 D 435: [telegram] Tx read request to device 0x08 for type ID 0x14
000+00:06:00.006 D 436: [telegram] Tx read request to device 0x08 for type ID 0x16
000+00:06:00.006 D 437: [telegram] Tx read request to device 0x08 for type ID 0x19
000+00:06:00.006 D 438: [telegram] Tx read request to device 0x08 for type ID 0x33
000+00:06:00.006 D 439: [emsesp] Fetching values for device ID 0x09
000+00:06:00.006 D 440: [emsesp] Fetching values for device ID 0x10
000+00:06:00.006 D 441: [telegram] Tx read request to device 0x10 for type ID 0x16F
000+00:06:00.006 D 442: [telegram] Tx read request to device 0x10 for type ID 0x179
000+00:06:00.029 D 443: [telegram] Sending read Tx [#70], telegram: 8B 88 14 00 20 2C
000+00:06:00.068 D 444: [emsesp] Last Tx read successful
000+00:06:00.068 D 445: [emsesp] Received UBATotalUptime
000+00:06:00.189 D 446: [telegram] Sending read Tx [#71], telegram: 8B 88 16 00 20 24
000+00:06:00.227 D 447: [mqtt] Publishing topic ems-esp/boiler_data (#297, attempt #1, size 486, pid 1)
000+00:06:00.234 D 448: [emsesp] Last Tx read successful
000+00:06:00.234 D 449: [emsesp] Received UBAParameters
000+00:06:00.372 D 450: [telegram] Sending read Tx [#72], telegram: 8B 88 19 00 20 18
000+00:06:00.429 D 451: [mqtt] Publishing topic ems-esp/boiler_data_ww (#298, attempt #1, size 413, pid 1)
000+00:06:00.439 D 452: [emsesp] Last Tx read successful
000+00:06:00.439 D 453: [emsesp] Received UBAMonitorSlow
000+00:06:00.559 D 454: [telegram] Sending read Tx [#76], telegram: 8B 88 19 19 20 2A
000+00:06:00.597 D 455: [emsesp] Last Tx read successful
000+00:06:00.597 D 456: [emsesp] Received UBAMonitorSlow
000+00:06:00.746 D 457: [telegram] Sending read Tx [#73], telegram: 8B 88 33 00 20 B0
000+00:06:00.795 D 458: [emsesp] Last Tx read successful
000+00:06:00.795 D 459: [emsesp] Received UBAParameterWW
000+00:06:00.938 D 460: [telegram] Sending read Tx [#74], telegram: 8B 90 FF 00 20 00 6F DA
000+00:06:00.988 D 461: [emsesp] Last Tx read successful
000+00:06:00.988 D 462: [emsesp] Received JunkersMonitor
000+00:06:01.088 D 463: [telegram] Sending read Tx [#75], telegram: 8B 90 FF 00 20 00 79 CC
000+00:06:01.149 D 464: [emsesp] Last Tx read successful
000+00:06:01.149 D 465: [emsesp] Received JunkersSet
000+00:06:01.235 D 466: [mqtt] Publishing topic ems-esp/heartbeat (#299, attempt #1, size 141, pid 1)
000+00:06:04.985 D 467: [emsesp] Received UBAMonitorFast
000+00:06:05.059 D 468: [mqtt] Publishing topic ems-esp/boiler_data (#300, attempt #1, size 486, pid 1)
000+00:06:05.260 D 469: [mqtt] Publishing topic ems-esp/boiler_data_ww (#301, attempt #1, size 413, pid 1)
000+00:06:05.619 W 470: [thermostat] Set temperature: Heating Circuit 3 not found or activated for device ID 0x10
000+00:06:05.619 W 471: [thermostat] Set temperature: Heating Circuit 3 not found or activated for device ID 0x10
[... cut ...]
000+00:06:19.148 D 481: [mqtt] Publishing topic ems-esp/boiler_data_ww (#307, attempt #1, size 413, pid 1)
000+00:06:19.350 D 482: [mqtt] Publishing topic ems-esp/boiler_data (#308, attempt #1, size 486, pid 1)
000+00:06:19.551 D 483: [mqtt] Publishing topic ems-esp/boiler_data_ww (#309, attempt #1, size 413, pid 1)
000+00:06:19.975 D 484: [emsesp] Received UBAMonitorFast
000+00:06:20.156 D 485: [mqtt] Publishing topic ems-esp/boiler_data (#310, attempt #1, size 486, pid 1)
000+00:06:20.357 D 486: [mqtt] Publishing topic ems-esp/boiler_data_ww (#311, attempt #1, size 413, pid 1)
```

MichaelDvP, 2020-11-14:
What do you send via mqtt? Topic and payload? 
HA-discovery should make a subscirption to `thermostat_hc1` and temperatur settings and modes goes to this topic.
Can you check with mqtt-explorer or other tools?
I dont have HA, so i can only check with mqtt explorer that the topics and subscription are ok, and sending to the subscriptions work.

proddy, 2020-11-14:
I have HA so let me see if I can simulate this

norberts1, 2020-11-15:
@Oderik  @proddy @MichaelDvP same comments on using the FR50 thermostat for controlling the boiler.
It's not possible to control the boiler with the FR50 thermostat on the heatronic3-bus. Older documents from Junkers tell us that the FR50 can not be controlled by the MBLan (modem). The software inside could be the reason for this.
Also the factory-builddate (FD) is importent for all of the Fxyz Junkers thermostats.
See also that comments on: https://www.mikrocontroller.net/topic/317004?page=single#4149528

Table for factory-builddates (Junkers Fxzy thermostats):
[image]

FR50 / MBLan comment 
[image]

Oderik, 2020-11-15:
I am currently looking at MQTT Explorer but I'm still a bit confused. I don't know how to find any original messages, I only see stats. HA seems to talk to the correct topic. At least, "thermostat_hc1" is at HA and ems-esp.

[image]

Oderik, 2020-11-15:
@norberts1 thanks for the information but I am not sure what that tells us. Do you mean it's impossible to control the FR50 from another bus participant? Because we already proved that's not true. I can control it via the console and with different firmware version also via MQTT.

norberts1, 2020-11-15:
@Oderik OK, good to know that also the FR50 can controll the boiler via the heatronic-bus. 
Then that information above **isn't valid anymore**, so only that factory-builddate (FD) is importent.
By the way: any modul on the heater-bus able to send that required telegrams to the thermostat using valid device-ID's is able to control the system.

Oderik, 2020-11-15:
> By the way: any modul on the heater-bus able to send that required telegrams to the thermostat using valid device-ID's is able to control the system.

Sounds interesting! Maybe we can discuss that in another threat (like this one https://github.com/proddy/EMS-ESP/issues/524) so we don't lose focus here.

MichaelDvP, 2020-11-15:
> EDIT2: HA sends 19 instead of { "cmd": "temp", "data": 19 } to update the temperature. That's odd - I'm sure it worked with an earlier firmware.

If you send to topic `thermostat` the syntax is `{"cmd":"temp","data":19,"hc":1}`, but HA-climate control sends to a special topic subscription `thermostat_hc1` with only the mode or the temperature value. But which of these commands ends in ems-esp with the wrong hc?

Oderik, 2020-11-15:
HA sends `19` to `thermostat_hc1` and that results in ems-esp confusing the hc.
When I send `{ "cmd": "temp", "data": 19 }` to `thermostat_hc1` using MQTT Explorer, the thermostat responds as expected.

proddy, 2020-11-15:
@Oderik Michael is correct. The climate component in Home Assistant can only send single values to a topic, as opposed to json strings. Which is why when the MQTT format is set to HA EMS-ESP will subscribe to ems-esp/thermostat_hc1 or what ever the heating circuit is. 

What exactly isn't working?

Oderik, 2020-11-15:
HA sends temperature value (let's say `19`) to topic `ems-esp/thermostat_hc1`.

Expected result: ems-esp sets temp of hc1 thermostat to 19.

Observed result: ems-esp logs an error, that "heating circuit [random number except 1]" is not available

proddy, 2020-11-15:
ah ok, clear. let me try and reproduce.

proddy, 2020-11-15:
ok, found it. fixing...

Oderik, 2020-11-16:
With dev release v2.1.1b3 everything works fine now. Thank you very much!

closed 2020-11-16 by Oderik