emsesp/EMS-ESP #610: ESP32 reset loop

sq4bja, 2020-11-14:
Hi,
In my ESP32 from firmware 2.1.1b1 I`m getting reset loop. I tested b1 and b2. In b0 everything works good.  The reset loop stops when I disconnect the ems bus. 
I have Buderus GB072 and RC310 thermostat.
Tom

Bartekn86, 2020-11-14:
have same issiue gb072 rc310

proddy, 2020-11-14:
its working on my setup so looks like something related to the thermostat changes perhaps. @MichaelDvP what do you think?

sq4bja, 2020-11-14:
We both have RC310 models. Maybe this is the reason?

proddy, 2020-11-14:
does it happen when you're in console? There is a problem with the debug logging causing weird things which I'm looking into

sq4bja, 2020-11-14:
No. This happens immediately after starting. As soon as EMS-ESP receives the telegrams. After 5-20 seconds, a restart takes place. I can see a flashing LED in ESP32 and also uptime in EMS-ESP panel. After several minutes, ESP32 stops responding.

MichaelDvP, 2020-11-15:
Is it related to HA-discovery? Have you tried with single/nested mqtt format? 
I tested with esp32 and GB125/RC35 and it work. I also switched the different formats, no crash.
I thinks GB125 have some values less than GB072, but RC35 have some more than RC310 for now.

proddy, 2020-11-15:
that's a good point. Start EMS-ESP without the EMS bus connected, go to Web and disable MQTT and see if it still crashes. I will help narrow down the root cause.

sq4bja, 2020-11-15:
Did not help. I turned off MQTT and upgraded to B2. After restart, problems with reading EMS bus again and continuous restarts. I'm going back to B0.

proddy, 2020-11-15:
damn. Are you able to compile and build the firmware yourself? There's one more thing we can try

sq4bja, 2020-11-15:
Yes, I can try.

proddy, 2020-11-15:
I'll build a test case to simulate this. What is the device_id and product_id for your GB072?

sq4bja, 2020-11-15:
[image]

proddy, 2020-11-15:
I made a test to see if I reproduce the crash in offline mode (not connected to the EMS bus), using the URL http://ems-esp/api?device=system&cmd=test&data=general2. But doesn't crash, so not sure what is causing it. I need more data!

sq4bja, 2020-11-15:
OK, below listing of telegrams send before reset EMS-ESP:
```
┌───────────────────────────────────────────
│ EMS-ESP version 2.1.1b2                  │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

ems-esp:/$ watch on
Watching incoming telegrams, displayed in decoded format
000+00:00:09.549 N 0: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2BC), data: FF 2E 2A 26 1E 03 00 FF FF 00 2A 01 E1 20 01 0F 05 00 00 01 22
000+00:00:09.628 N 1: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 8C 00 00 00 00 80 40 80 00 01 9D 80 00 00 00 FF 30 48 00 CB 00 00 00
000+00:00:09.895 N 2: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
000+00:00:10.127 N 3: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 2F 01 9D 01 9D 01 00 05 03 00 01 08 45 00 0D 00 00 80 00
000+00:00:10.437 N 4: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x2AF), data: 04 28 00 00 30 2D 0A 01 18 1E 00
000+00:00:10.906 N 5: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x2B0), data: 03 28 00 00 4B 2D 11 01 19 1E 00
000+00:00:11.259 N 6: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x2B1), data: 03 28 00 00 4B 2D 11 01 19 1E 00
000+00:00:12.134 N 7: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x2B2), data: 03 28 00 00 4B 2D 11 01 19 1E 00
ems-esp:/$

```

sq4bja, 2020-11-15:
And here is log all
```
ems-esp:/$ log all
Log level = all
000+00:00:08.928 D 0: [telegram] Sending read Tx [#0], telegram: 0B 88 07 00 20 A8
000+00:00:12.584 D 1: [telegram] Sending read Tx [#1], telegram: 0B 88 07 00 20 A8
000+00:00:12.659 D 2: [emsesp] Last Tx read successful
000+00:00:12.659 D 3: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
000+00:00:12.659 D 4: [telegram] Tx read request to device 0x08 for type ID 0x02
000+00:00:12.659 D 5: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
000+00:00:12.659 D 6: [telegram] Tx read request to device 0x09 for type ID 0x02
000+00:00:12.659 D 7: [emsesp] New EMS device detected with ID 0x10. Requesting version information.
000+00:00:12.659 D 8: [telegram] Tx read request to device 0x10 for type ID 0x02
000+00:00:12.785 D 9: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
000+00:00:12.785 D 10: [telegram] Tx read request to device 0x08 for type ID 0x02
000+00:00:12.785 D 11: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
000+00:00:12.785 D 12: [telegram] Tx read request to device 0x09 for type ID 0x02
000+00:00:12.785 D 13: [emsesp] New EMS device detected with ID 0x10. Requesting version information.
000+00:00:12.785 D 14: [telegram] Tx read request to device 0x10 for type ID 0x02
000+00:00:12.982 D 15: [telegram] Sending read Tx [#2], telegram: 0B 88 02 00 20 BC
000+00:00:13.027 D 16: [emsesp] Last Tx read successful
000+00:00:13.027 D 17: [emsesp] Adding new device GBx72/Trendline/Cerapur/Greenstar Si/27i (device ID 0x08, product ID 123, version 06.08)
000+00:00:13.027 D 18: [boiler] Adding new Boiler with device ID 0x08
000+00:00:13.027 D 19: [emsesp] Fetching values for device ID 0x08
000+00:00:13.027 D 20: [telegram] Tx read request to device 0x08 for type ID 0x14
000+00:00:13.027 D 21: [telegram] Tx read request to device 0x08 for type ID 0x16
000+00:00:13.027 D 22: [telegram] Tx read request to device 0x08 for type ID 0x19
000+00:00:13.027 D 23: [telegram] Tx read request to device 0x08 for type ID 0x33
000+00:00:13.027 D 24: [telegram] Tx read request to device 0x08 for type ID 0xE6
000+00:00:13.027 D 25: [telegram] Tx read request to device 0x08 for type ID 0xEA
000+00:00:13.356 D 26: [telegram] Sending read Tx [#3], telegram: 0B 89 02 00 20 B4
000+00:00:13.392 D 27: [emsesp] Last Tx read successful
000+00:00:13.392 D 28: [emsesp] Adding new device BC25 (device ID 0x09, product ID 125, version 03.03)
000+00:00:13.392 D 29: [emsesp] Fetching values for device ID 0x09
000+00:00:13.654 D 30: [telegram] Sending read Tx [#4], telegram: 0B 90 02 00 20 7C
000+00:00:13.718 D 35: [telegram] Tx read request to device 0x10 for type ID 0x2A6
000+00:00:13.718 D 36: [telegram] Tx read request to device 0x10 for type ID 0x2A7
000+00:00:13.718 D 37: [telegram] Tx read request to device 0x10 for type ID 0x2A8
000+00:00:13.718 D 38: [telegram] Tx read request to device 0x10 for type ID 0x2B9
000+00:00:13.718 D 39: [telegram] Tx read request to device 0x10 for type ID 0x2BA
000+00:00:13.718 D 40: [telegram] Tx read request to device 0x10 for type ID 0x2BB
000+00:00:13.718 D 41: [telegram] Tx read request to device 0x10 for type ID 0x2BC
000+00:00:13.718 D 42: [telegram] Tx read request to device 0x10 for type ID 0x2AF
000+00:00:13.718 D 43: [telegram] Tx read request to device 0x10 for type ID 0x2B0
000+00:00:13.718 D 44: [telegram] Tx read request to device 0x10 for type ID 0x2B1
000+00:00:13.718 D 45: [telegram] Tx read request to device 0x10 for type ID 0x2B2
000+00:00:13.718 D 46: [telegram] Tx read request to device 0x10 for type ID 0x29B
000+00:00:13.718 D 47: [telegram] Tx read request to device 0x10 for type ID 0x29C
000+00:00:13.718 D 48: [telegram] Tx read request to device 0x10 for type ID 0x29D
000+00:00:13.718 D 49: [telegram] Tx read request to device 0x10 for type ID 0x29E
000+00:00:13.718 D 50: [emsesp] Fetching values for device ID 0x10
000+00:00:13.718 D 51: [telegram] Tx read request to device 0x10 for type ID 0x2F5
000+00:00:13.718 D 52: [telegram] Tx read request to device 0x10 for type ID 0x31B
000+00:00:13.718 D 53: [telegram] Tx read request to device 0x10 for type ID 0x23A
000+00:00:13.718 D 54: [telegram] Tx read request to device 0x10 for type ID 0x240
000+00:00:14.052 D 55: [telegram] Sending read Tx [#14], telegram: 0B 90 FF 00 20 01 A5 19
000+00:00:14.184 D 56: [emsesp] Last Tx read successful
000+00:00:14.184 D 57: [emsesp] Received RC300Monitor
000+00:00:14.184 D 58: [emsesp] Toggling fetch for device ID 0x10, telegram ID 0x2A5 to 1
000+00:00:14.184 D 59: [emsesp] Toggling fetch for device ID 0x10, telegram ID 0x2B9 to 1
000+00:00:14.301 D 60: [emsesp] Received RC300Floordry
000+00:00:14.559 D 61: [emsesp] Received RC300Monitor
000+00:00:14.948 D 62: [telegram] Sending read Tx [#34], telegram: 0B 90 FF 19 20 01 A5 D1
000+00:00:15.057 D 63: [emsesp] Last Tx read successful
000+00:00:15.057 D 64: [emsesp] Received RC300Monitor
000+00:00:15.142 D 65: [emsesp] Received RC300Monitor
000+00:00:15.349 D 66: [emsesp] Received RC300WWmode2
000+00:00:15.796 D 67: [telegram] Sending read Tx [#15], telegram: 0B 90 FF 00 20 01 A6 1A
000+00:00:16.164 D 68: [emsesp] Received UBAMonitorFast
000+00:00:16.433 D 69: [emsesp] Received MC10Status
[... 25 lines cut ...]
000+00:00:19.024 D 95: [emsesp] Last Tx read successful
000+00:00:19.024 D 96: [emsesp] Received RC300Set
[... cut ...]
000+00:00:20.621 D 110: [telegram] Sending read Tx [#24], telegram: 0B 90 FF 00 20 01 B1 0D
000+00:00:20.708 D 111: [emsesp] Last Tx read successful
000+00:00:20.708 D 112: [emsesp] Received RC300Summer
000+00:00:20.969 D 113: [telegram] Sending read Tx [#25], telegram: 0B 90 FF 00 20 01 B2 0E
000+00:00:21.052 D 114: [emsesp] Last Tx read successful
000+00:00:21.052 D 115: [emsesp] Received RC300Summer
000+00:00:21.392 D 116: [telegram] Sending read Tx [#26], telegram: 0B 90 FF 00 20 01 9B 27
ems-esp:/$

```

MichaelDvP, 2020-11-15:
Ah, i see the error, forgot to add heatingcurve_ids to heating_circuit serach. I'll fix it.

MichaelDvP, 2020-11-15:
Please check if it is fixed now or something other to search for.

sq4bja, 2020-11-15:
Ok. It was that... Now everyfing works perfectly.
Thank You!

Bartekn86, 2020-11-15:
all ok thank you

closed 2020-11-15 by sq4bja