emsesp/EMS-ESP #615: mqtt retain flag not set for all topics

deejaybeam, 2020-11-15:
**Bug description**
Retain option is set in mqtt-settings, but not all topics have the flag set

**Steps to reproduce**
Enable retain setting and check mqtt messages

**Expected behavior**
Retain flag shuld be set on all topics

**Screenshots**
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

**Device information**
{
  "System": {
    "version": "2.1.0",
    "uptime": "000+00:33:05.387",
    "freemem": 41,
    "fragmem": 4
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 1,
    "mqtt_qos": 0,
    "mqtt_retain": "on",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 2506,
    "#read requests sent": 454,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GB125/MC10 (DeviceID:0x08, ProductID:72, Version:02.04)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC30 (DeviceID:0x10, ProductID:67, Version:02.08)",
      "handlers": "0xA3 0x06 0xA2 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37"
    },
    {
      "type": "Mixer",
      "name": "MM10 (DeviceID:0x21, ProductID:69, Version:02.00)",
      "handlers": "0xAA 0xAB 0xAC"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.00)",
      "handlers": ""
    }
  ]
}

**Additional context**
Some topics have the retain flag set:
* ems-esp/status
* ems-esp/info

Most topic have the retain flag not set:
* ems-esp/heartbeat
* ems-esp/boiler_data
* ems-esp/heating_active
* ems-esp/tapwater_active
* ems-esp/boiler_data_ww
* ems-esp/thermostat_data
* ems-esp/thermostat_data_hc1
* ems-esp/thermostat_data_hc2
* ems-esp/mixer_data_hc2

regards chris

proddy, 2020-11-15:
well spotted. I'm not seeing the RETAIN any more either. I'll look into it...

proddy, 2020-11-15:
which version of MQTTExplorer are you using? 0.3.5 or 0.4 beta (which incorrectly reports retain flags)

deejaybeam, 2020-11-15:
I´m using 0.4.0-beta1.
Tried again with 0.3.5 but with same result.
Also tried mqttfx - same result:
[image]

proddy, 2020-11-15:
I'm seeing the same behaviour. Not sure why the broker is not picking this up. I'm debugging the mqtt libs to see if I can find out

deejaybeam, 2020-11-15:
Maybe I´m wrong. I have increased the publish interval on ems-esp, wait for an update in mqtt-explorer, closed it and restart mqtt-explorer. After connection I can see the topic, which should not be possible if they are not stored on broker (retain-flag). After first update on topic the retained label disapear...
So maybe this is by design and the retained label just means it´s a stored message, sent before the client subscribes and is cleard after a message is received while active subscription.
But I don´t know, maybe someone?

proddy, 2020-11-15:
yes, that was my understanding too. Try doing a publish in MQTTExplorer to some dummy topic with the retain check-box ticket. It'll show as retained and then vanish. I just did a quick test on my windows PC running `mosquitto.exe -v` and can see the publish messages coming in from EMS-ESP are indeed retained:

```
1605471244: Received PUBLISH from ems-esp32 (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/shower_active/config', ... (156 bytes))
1605471244: Received PUBLISH from ems-esp32 (d0, q0, r1, m0, 'ems-esp32/shower_active', ... (3 bytes))
1605471244: Received PUBLISH from ems-esp32 (d0, q0, r1, m0, 'ems-esp32/heartbeat', ... (129 bytes))
1605471244: Received PUBLISH from ems-esp32 (d0, q0, r1, m0, 'ems-esp32/info', ... (56 bytes))
1605471245: Received PUBLISH from ems-esp32 (d0, q0, r1, m0, 'homeassistant/sensor/ems-esp/status/config', ... (292 bytes))
1605471245: Received PUBLISH from ems-esp32 (d0, q0, r1, m0, 'ems-esp32/status', ... (6 bytes))
```

(notice the r1)

would be good to find out the reason though

deejaybeam, 2020-11-15:
So it looks like the issue is solved and can be closed, because ems-esp sends topic with the retain flag.
Do you agree?
Or do you want to leave it open to see if someone knows the exact behavior?

proddy, 2020-11-15:
I'll leave it open and search the web for an explanation.

proddy, 2020-11-15:
Noticed if you restart MQTTExplorer the magically reappear as RETAINED. Closing this for now

closed 2020-11-15 by proddy