emsesp/EMS-ESP #617: Create a table-based generic device handler for telegrams

hpanther, 2020-11-17:
So that we don't have to write new code for each added telegram, and don't have to repeat the same code over and over, write a generic implementation.

proddy, 2020-11-17:
This is very hard to do for a number of reasons:
1. memory. Keeping all telegrams and their respected data values will not fit into the 8-9K available heap on an ESP8266. Which is why it's done dynamically now. It has the shortest memory footprint.
2. reference variables. EMS-ESP typically needs to keep local copies of the data values. For example in the Thermostat the values of each heating circuit is loaded into classes (HC1...HCn) and the data referenced everywhere in the code to make decisions. If it was just as simple as reading telegrams and exporting data then you could use a hash map/lookup, but don't underestimate the amount of business logic going
3. readability. There is very little code duplication. When we find some we usually refactor it out.

I am open to ideas and suggestions though, and will support anyone that can improve or tidy up the code. 

Also I did take a look at https://github.com/danielwippermann/resol-vbus.rs but it just parses and throws out data.

hpanther, 2020-11-17:
I'm working on a proposal for this

proddy, 2020-11-18:
great. If you have code you want to share, let me create a new feature branch where you can push the PRs too.

closed 2021-07-25 by proddy