emsesp/EMS-ESP #618: gb072 + rc310

Bartekn86, 2020-11-17:
hi, 
could you tell me did you remove fan modulation ?  i can not see it in new build

proddy, 2020-11-17:
what's the name of the parameter and where is it missing (web, console, mqtt)?

Bartekn86, 2020-11-17:
missing in web,  mqtt to. i do not remember exacly what was the name but it shows in percent speed of burners fan

proddy, 2020-11-17:
ah ok. looks like it ran away. when did you last see it? which build?

Bartekn86, 2020-11-18:
hm... hard to say when exacly , max 4 builds ago it works

proddy, 2020-11-18:
do you remember the name of the parameter? It is from the boiler? There is no "fan modulation", there ar pump modulations and fan on/off in the Boiler for example.

Bartekn86, 2020-11-18:
i do not remember. in domoticz i had value  range 0-100 %. could you give me an older version fw ? i can downgrade and check.

proddy, 2020-11-18:
can you remember which device it was (boiler, solar, heatpump)? Also the name? Then you can look at the domocticz plugin to see what it maps too. I checked the code and can't see anything that has been removed.

bbqkees, 2020-11-19:
@Bartekn86 There is no fan percentage device in the Domoticz plugin. There is only one on/off fan device (fanWork).

Bartekn86, 2020-11-22:
heh, weird, I was absolutely sure there was something like that, then I misinterpreted the descriptions.

closed 2021-10-04 by proddy