emsesp/EMS-ESP #620: Boiler heating status in Home Assistant thermostat

realthk, 2020-11-18:
I've just replaced a simple on-off thermostat (that was also [integrated into HA through MQTT](https://github.com/fashberg/WThermostatBeca)) with a Bosch CW400 connected to my Bosch 2300 boiler.

It works fine with EMS-ESP, apart from a small issue: there is no heating status info in the HA MQTT thermostat where it used to be.

The first part on this diagram was with the old thermostat, that provided this info. The green columns on the left show when the boiler was heating - but not any more with the new bus thermostat on the right:
[image]
It was done with status="heating", status="idle" MQTT messages 
(though obviously it wasn't 100% correct all the time, because DHW has priority over heating, and the dumb old thermostat had no info about hot water production. Now, communicating with the boiler, even that could be avoided)

I guess it is not so straightforward now, as I cannot even see any sensor in HA (or a value in the console of EMS-ESP) that reflects the heating status of the boiler... 
The only way I could detect it is `sensor.boiler_gas==on &&  sensor.boiler_warm_water_charging==off`, then it must be heating right now.

Could the MQTT thermostat messages be extended with this info, or is it too much boiler/thermostat dependant?

proddy, 2020-11-18:
simple to add, would need to first test on other systems. What would the name this control? @MichaelDvP what's your opinion?

MichaelDvP, 2020-11-19:
This is what `heating_active` do, Published as single value (always on change) and in boiler_data (scheduled). The boiler value is also registered as ha-sensor.

realthk, 2020-11-19:
Not in the case of my Bosch 2300 boiler (recognised by EMS-ESP as Logamax Plus GB122), no.

It is now currently in idle mode, but heating_active is on regardless:
```
Boiler: Bosch Logamax Plus GB122 (DeviceID:0x08 ProductID:234, Version:05.03)
  Heating active: on
  Warm water/DHW active: on
  Warm water selected temperature: 0 °C
  Warm water set temperature: 54 °C
  Warm water disinfection temperature: 70 °C
  Selected flow temperature: 39 °C
  Burner selected max power: 100 %
  Burner current power: 0 %
  Pump modulation: 100 %
  Heat pump modulation: 0 %
  Warm water circulation pump available: off
  Warm water circulation pump freq: 0x3min
  Warm water circulation active: off
  Outside temperature: 4.5 °C
  Warm water current temperature (intern): 35.5 °C
  Warm water current temperature (extern): 0.0 °C
  Current flow temperature: 33.7 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: off
  Gas: off
  Flame current: 0.0 uA
  Pump: on
  Fan: off
  Ignition: off
  Warm water heating: off
  Heating activated: on
  Heating temperature setting: 70 °C
  Burner min period: 10 min
  Burner min power: 1 %
  Burner max power: 0 %
  Temperature hysteresis on: -6 °C
  Temperature hysteresis off: 6 °C
  Warm water starts: 12717
  Warm water active time: 6276
  Burner starts: 7855
  Warm Water active time: 4 days 8 hours 36 minutes
  Total burner operating time: 27 days 15 hours 57 minutes
  Total heat operating time: 18 days 20 hours 0 minutes
  Total UBA working time: 386 days 14 hours 40 minutes

Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05)
  Time: 09:00:34 19/11/2020
  Damped outdoor temperature: 3.0 °C
  Building: medium
  Warm water mode: own_prog
  Heating circuit 1:
    Setpoint room temperature: 20.5 °C
    Current room temperature: 23.2 °C
    Comfort temperature: 20.5 °C
    Eco temperature: 18.0 °C
    Manual temperature: 20.0 °C
    Target flow temperature: 39 °C
    Mode: auto
    Mode type: comfort
```

This is when it was in DHW mode (gas was off when copied the values, but probably because water was already hot enough).   `Warm water charging: on` is fine:
```
  Heating active: on
  Warm water/DHW active: on
  Warm water selected temperature: 0 °C
  Warm water set temperature: 50 °C
  Warm water disinfection temperature: 70 °C
  Selected flow temperature: 50 °C
  Burner selected max power: 100 %
  Burner current power: 0 %
  Pump modulation: 100 %
  Heat pump modulation: 0 %
  Warm water circulation pump available: off
  Warm water circulation pump freq: 0x3min
  Warm water circulation active: off
  Outside temperature: 6.6 °C
  Warm water current temperature (intern): 15.1 °C
  Warm water current temperature (extern): 0.0 °C
  Current flow temperature: 44.4 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: on
  Warm water recharging: off
  Warm water temperature ok: on
  Gas: off
  Flame current: 44.7 uA
  Pump: on
  Fan: on
  Ignition: off
  Warm water heating: on
  Heating activated: on
  Heating temperature setting: 51 °C
  Burner min period: 10 min
  Burner min power: 1 %
  Burner max power: 0 %
  Temperature hysteresis on: -6 °C
  Temperature hysteresis off: 6 °C
  Warm water starts: 12699
  Warm water active time: 6267
  Burner starts: 7826
  Warm Water active time: 4 days 8 hours 27 minutes
  Total burner operating time: 27 days 10 hours 0 minutes
  Total heat operating time: 18 days 14 hours 21 minutes
  Total UBA working time: 385 days 21 hours 47 minutes
```

And this is when the boiler was in heating mode for real. The only way I could assign this state to a HA binary_sensor was with a template` {{ is_state('sensor.boiler_gas', 'on') and is_state('sensor.boiler_warm_water_charging', 'off') }}`
```
  Heating active: on
  Warm water/DHW active: on
  Warm water selected temperature: 0 °C
  Warm water set temperature: 54 °C
  Warm water disinfection temperature: 70 °C
  Selected flow temperature: 39 °C
  Burner selected max power: 100 %
  Burner current power: 36 %
  Pump modulation: 100 %
  Heat pump modulation: 25 %
  Warm water circulation pump available: off
  Warm water circulation pump freq: 0x3min
  Warm water circulation active: off
  Outside temperature: 6.4 °C
  Warm water current temperature (intern): 37.0 °C
  Warm water current temperature (extern): 0.0 °C
  Current flow temperature: 32.9 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: off
  Gas: on
  Flame current: 64.1 uA
  Pump: on
  Fan: on
  Ignition: off
  Warm water heating: off
  Heating activated: on
  Heating temperature setting: 55 °C
  Burner min period: 10 min
  Burner min power: 1 %
  Burner max power: 0 %
  Temperature hysteresis on: -6 °C
  Temperature hysteresis off: 6 °C
  Warm water starts: 12700
  Warm water active time: 6268
  Burner starts: 7828
  Warm Water active time: 4 days 8 hours 28 minutes
  Total burner operating time: 27 days 10 hours 6 minutes
  Total heat operating time: 18 days 14 hours 26 minutes
  Total UBA working time: 385 days 22 hours 25 minutes

```

MichaelDvP, 2020-11-19:
What ems-esp-version do you use? 
Can you give me a `read 8 18` from a mismatch state?
`heating active` is read from the boiler as a state that burner is on and heating is active. (depending on system it's possible that heating and warm water is heated simultanious).

realthk, 2020-11-19:
EMS/ESP is the latest main, v2.1.0

The boiler currently doesn't do anything, but heating active is still on:

```
┌──────────────────────────────────────────┐
│ EMS-ESP version 2.1.0                    │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

ems-esp:/$ su
Password: 
000+00:06:53.567 N 0: [shell] su session opened on console
ems-esp:/# read 8 18
000+00:07:04.159 N 1: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFast(0x18), data: <empty>
ems-esp:/# show values
Boiler: Bosch Logamax Plus GB122 (DeviceID:0x08 ProductID:234, Version:05.03)
  Heating active: on
  Warm water/DHW active: on
  Warm water selected temperature: 0 °C
  Warm water set temperature: 52 °C
  Warm water disinfection temperature: 70 °C
  Selected flow temperature: 0 °C
  Burner selected max power: 0 %
  Burner current power: 0 %
  Pump modulation: 0 %
  Heat pump modulation: 0 %
  Warm water circulation pump available: off
  Warm water circulation pump freq: 0x3min
  Warm water circulation active: off
  Outside temperature: 6.1 °C
  Warm water current temperature (intern): 31.7 °C
  Warm water current temperature (extern): 0.0 °C
  Current flow temperature: 32.9 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: off
  Gas: off
  Flame current: 0.0 uA
  Pump: off
  Fan: off
  Ignition: off
  Warm water heating: off
  Heating activated: on
  Heating temperature setting: 70 °C
  Burner min period: 10 min
  Burner min power: 1 %
  Burner max power: 0 %
  Temperature hysteresis on: -6 °C
  Temperature hysteresis off: 6 °C
  Warm water starts: 12717
  Warm water active time: 6276
  Burner starts: 7857
  Warm Water active time: 4 days 8 hours 36 minutes
  Total burner operating time: 27 days 16 hours 13 minutes
  Total heat operating time: 18 days 20 hours 16 minutes
  Total UBA working time: 386 days 15 hours 34 minutes

Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05)
  Time: 09:55:07 19/11/2020
  Damped outdoor temperature: 3.5 °C
  Building: medium
  Warm water mode: own_prog
  Heating circuit 1:
    Setpoint room temperature: 19.0 °C
    Current room temperature: 23.7 °C
    Comfort temperature: 20.5 °C
    Eco temperature: 18.0 °C
    Manual temperature: 19.0 °C
    Target flow temperature: 0 °C
    Summer temperature: 17 °C
    Summer mode: auto °C
    Mode: manual
    Mode type: comfort

```

MichaelDvP, 2020-11-19:
Ok, your boiler doesn't have the 0x18 telegram, thats why servicecode and `heating_active` not working. 
Can you check with `read 8 E4` for the newer version of UBAMonitor. There should be the servicecode and the boiler state, but in this telegram only a few values are known.  Maybe we can work out some more.

realthk, 2020-11-19:
Thanks!
This is `read 8 E4` in idle state:
```
000+00:01:22.293 N 1: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 E3 00 00 00 02 01 F7 00 00 00 00 00 00 14 00 01 E3 00 00
000+00:01:22.552 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
```

This is from DHW mode:
```
000+00:04:01.084 N 3: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 F8 (offset 23)
000+00:04:01.264 N 4: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 FA (offset 23)
```

Idle again a few times (not every byte is the same):
```
000+00:05:03.993 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 02 29 00 00 00 02 02 0B 00 00 00 00 00 00 14 00 02 29 00 00
000+00:05:04.262 N 6: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
ems-esp:/# read 8 E4
000+00:06:03.727 N 7: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 02 25 00 00 00 02 02 2A 00 00 00 00 00 00 14 00 02 25 00 00
000+00:06:03.987 N 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
ems-esp:/# read 8 E4
000+00:06:13.744 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 02 23 00 00 00 02 02 2D 00 00 00 00 00 00 14 00 02 23 00 00
000+00:06:14.003 N 10: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
```

During heating (a few times because it seems to change quite a lot):
```
000+00:10:34.818 N 11: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 C8 46 01 67 0B 11 03 02 01 A5 00 00 00 00 00 B2 14 09 01 67 00 00
000+00:10:34.969 N 12: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 66 (offset 23)
ems-esp:/# read 8 E4
000+00:11:02.044 N 13: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 C8 46 01 3B 0B 0F 03 02 01 76 00 00 00 00 00 8F 14 09 01 3B 00 00
000+00:11:02.232 N 14: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 0A 0A (offset 35)
ems-esp:/# read 8 E4
000+00:11:32.886 N 15: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 C8 46 01 36 0B 0F 03 02 01 66 00 00 00 00 00 91 14 08 01 36 00 00
000+00:11:33.037 N 16: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 33 (offset 23)
ems-esp:/# read 8 E4
000+00:11:37.822 N 17: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 C8 46 01 34 0B 0F 03 02 01 66 00 00 00 00 00 96 14 08 01 34 00 00
000+00:11:38.271 N 18: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0A 0A (offset 25)
ems-esp:/# read 8 E4
000+00:11:48.558 N 19: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 C8 46 01 33 0B 0F 03 02 01 62 00 00 00 00 00 99 14 08 01 33 00 00
000+00:11:48.818 N 20: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0A 0A (offset 25)
```

And finally idle mode after heating:
```
000+00:05:53.044 N 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 4F 00 00 00 02 01 45 00 00 00 00 00 00 14 00 01 4F 00 00
000+00:05:53.304 N 9: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
ems-esp:/# read 8 e4
000+00:06:19.820 N 10: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 75 00 00 00 02 01 4A 00 00 00 00 00 00 14 00 01 75 00 00
000+00:06:20.079 N 11: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
ems-esp:/# read 8 e4
000+00:06:24.287 N 12: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 79 00 00 00 02 01 4A 00 00 00 00 00 00 14 00 01 79 00 00
000+00:06:24.547 N 13: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
```

MichaelDvP, 2020-11-19:
Thanks, it should be interesting to have also the complete telegram in DHW mode. But so far i see:
ServiceCode in pos 1,2 is always `--`
ServiceCodeNumber in pos 4/5 is 203 idle, 200 heating 
flowtemp in pos 7/8 is the same as in pos 23 (which is publishes in DHW mode), maybe it's boiler temp or storage temp ~50°C?
Statusbits pos 11 are 00 idle, 03 in heating, bit 0 is burner and i think bit 1 is heating pump
I think pos 22 is boilerstate, in heating mode it goes to 09, and 08. If this goes to 0A in DHW mode we have it. 
The other known positions seem to be ok, 
Unknown: 
pos 12 is always 02, pos 21 is always 14, look like configuration bits.
pos 13/14: looks like a temperature. ~ 35°C, Is this the flowtemp? do you have floorheating?

@realthk: Can you please try to catch the DWH telegram and look at your boiler to check what temperatures are around 50°, which around 35° . In the show values i think not all is right.

And i wonder from where you get 
> Warm water current temperature (intern): 31.7 °C

this is only in telegram 0x18.

realthk, 2020-11-19:
That's great! I think the main point is to be able to tell what is the boiler doing, so those 203 and 200 values might help.

I have very simple system: no room for water-storage in the small bathroom, so it is a flow-through DHW, and the heating is done with radiators, no floorheating. Water temp is usually around 50C, but when I captured that data, it was just started for test, so it might have very well been about 35C, you're right! I also have Dallas sensors on the pipes, and checking their log now, it seems the outgoing line was 35C indeed.

Do you mean `watch raw` command?
I've issued that command right after starting hot water:
```
000+06:11:51.207 N 28: [telegram] Rx: 88 00 E5 06 02 AF 44
000+06:11:51.387 N 29: [telegram] Rx: 88 00 E4 17 01 76 D7
000+06:11:51.648 N 30: [telegram] Rx: 88 00 E4 17 01 78 D9
000+06:11:51.828 N 31: [telegram] Rx: 88 00 E5 06 02 B6 5D
000+06:11:52.008 N 32: [telegram] Rx: 88 00 E4 17 01 7C DD
000+06:11:52.310 N 33: [telegram] Rx: 88 00 E4 17 01 80 21
000+06:11:52.489 N 34: [telegram] Rx: 88 00 E4 17 01 82 23
000+06:11:52.699 N 35: [telegram] Rx: 88 00 E9 00 34 01 4B 00 00 00 00 01 4B 46 00 27 11 20 00 31 BA 00 18 8F 00 00 00 34 01 45
```

then EMS-ESP suddenly interrupted telnet connection, and after logging in again:
```
000+00:00:54.361 N 0: [telegram] Rx: 88 00 E5 04 01 2B CE
000+00:00:54.541 N 1: [telegram] Rx: 88 00 E4 17 02 1F B8
000+00:00:54.721 N 2: [telegram] Rx: 88 00 E5 04 01 2A CF
000+00:00:55.443 N 3: [telegram] Rx: 88 00 E5 04 01 27 C2
000+00:00:55.623 N 4: [telegram] Rx: 88 00 E5 04 01 29 CC
000+00:00:57.138 N 5: [telegram] Rx: 88 00 E5 04 01 2D 01 2A 23
000+00:00:58.337 N 6: [telegram] Rx: 88 00 E5 04 01 2F CA
000+00:00:59.509 N 7: [telegram] Rx: 88 00 E5 04 01 31 D4
000+00:00:59.688 N 8: [telegram] Rx: 88 00 E5 04 01 31 D4
000+00:00:59.869 N 9: [telegram] Rx: 88 00 E5 04 01 33 D6
000+00:01:00.261 N 10: [telegram] Rx: 88 0B 14 00 08 80 C2 E2
000+00:01:00.420 N 11: [telegram] Rx: 88 00 E5 04 01 35 D0
000+00:01:00.600 N 12: [telegram] Rx: 88 00 E5 04 01 33 D6
000+00:01:00.868 N 13: [telegram] Rx: 88 0B 16 00 24
000+00:01:01.043 N 14: [telegram] Rx: 88 00 E5 04 01 36 01 34 51
000+00:01:01.299 N 15: [telegram] Rx: 88 0B 19 00 3A
000+00:01:01.578 N 16: [telegram] Rx: 88 0B E6 00 01 46 00 00 46 0A 00 01 06 FA 0A 01 02 64 01 00 00 1E 00 3C 01 00 00 00 01 00 9A
000+00:01:01.739 N 17: [telegram] Rx: 88 00 E5 04 01 38 DD
000+00:01:02.039 N 18: [telegram] Rx: 88 0B EA 00 00 00 3C 04 02 01 34 00 00 00 00 00 46 00 00 01 00 00 00 00 00 64 00 00 00 00 01 3A
000+00:01:02.073 N 19: [telegram] Rx: 88 00 E5 04 01 3A DF
000+00:01:02.253 N 20: [telegram] Rx: 88 00 E5 04 01 38 DD
000+00:01:02.423 N 21: [telegram] Rx: 88 00 E5 04 01 39 DC
000+00:01:02.743 N 22: [telegram] Rx: 88 00 E3 00 04 00 00 00 00 01 00 00 00 00 00 02 1D 27 64 46 01 00 00 BF
000+00:01:02.951 N 23: [telegram] Rx: 88 00 E4 00 00 2D 2D 00 00 C9 34 02 1E 64 38 05 02 01 DC 00 00 00 00 03 45 14 00 02 1E 00 00 19
000+00:01:03.147 N 24: [telegram] Rx: 88 00 E4 23 00 00 00 00 00 27 28 98
000+00:01:03.355 N 25: [telegram] Rx: 88 00 E5 04 01 3D D8
000+00:01:03.526 N 26: [telegram] Rx: 88 00 E5 04 01 3A DF
000+00:01:03.731 N 27: [telegram] Rx: 88 00 E9 00 34 01 F7 00 00 00 00 01 F7 46 00 26 11 20 00 31 BB 00 18 8F 00 00 00 34 01 49
000+00:01:03.986 N 28: [telegram] Rx: 88 0B EA 1A 01 A6
000+00:01:04.017 N 29: [telegram] Rx: 88 00 E5 04 01 3E DB
000+00:01:04.317 N 30: [telegram] Rx: 88 00 E5 04 01 3B DE
000+00:01:04.498 N 31: [telegram] Rx: 88 00 E5 04 01 3D D8
000+00:01:04.839 N 32: [telegram] Rx: 90 0B FF 00 01 A5 00 D7 20 29 00 00 29 24 01 35 03 03 01 01 35 02 C7 00 00 11 01 03 08 65 00 F6
000+00:01:04.868 N 33: [telegram] Rx: 88 00 E5 04 01 40 A5
000+00:01:05.040 N 34: [telegram] Rx: 88 00 E5 04 01 3E 01 3E 7B
000+00:01:05.229 N 35: [telegram] Rx: 88 00 E5 04 01 3F DA
000+00:01:05.666 N 36: [telegram] Rx: 90 0B FF 19 01 A5 06 07 FF E2 00 00 FF 00 46 00 3C 01 FF 01 00 CE
000+00:01:05.691 N 37: [telegram] Rx: 88 00 E5 04 01 42 A7
000+00:01:05.860 N 38: [telegram] Rx: 88 00 E5 04 01 42 A7
000+00:01:06.161 N 39: [telegram] Rx: 88 00 E5 04 01 43 A6
000+00:01:06.361 N 40: [telegram] Rx: 88 00 E5 04 01 41 A4
000+00:01:06.541 N 41: [telegram] Rx: 88 00 E5 04 01 44 A1
000+00:01:06.850 N 42: [telegram] Rx: 90 0B FF 00 01 B9 FF 2E 29 26 24 03 00 FF FF 00 27 01 E1 20 01 0F 05 00 00 01 22 FD
000+00:01:07.008 N 43: [telegram] Rx: 90 00 06 00 14 0B 11 13 33 39 03 00 10 FF 00 92
000+00:01:07.203 N 44: [telegram] Rx: 88 00 E5 04 01 45 A0
000+00:01:07.373 N 45: [telegram] Rx: 88 00 E5 04 01 48 AD
000+00:01:07.547 N 46: [telegram] Rx: 88 00 E5 04 01 46 A3
000+00:01:07.755 N 47: [telegram] Rx: 88 00 E5 04 01 48 AD
000+00:01:08.061 N 48: [telegram] Rx: 90 0B FF 00 01 F5 01 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01 0D
000+00:01:08.316 N 49: [telegram] Rx: 90 0B FF 00 01 3A 00 50 72
000+00:01:08.465 N 50: [telegram] Rx: 88 00 E5 04 01 48 AD
[... 92 lines cut ...]
  Fan: on
  Ignition: off
  Warm water heating: on
  Heating activated: on
  Heating temperature setting: 70 °C
  Burner min period: 10 min
  Burner min power: 1 %
  Burner max power: 0 %
  Temperature hysteresis on: -6 °C
  Temperature hysteresis off: 6 °C
  Warm water starts: 12731
  Warm water active time: 6287
  Burner starts: 7869
  Warm Water active time: 4 days 8 hours 47 minutes
  Total burner operating time: 27 days 16 hours 32 minutes
  Total heat operating time: 18 days 20 hours 21 minutes
  Total UBA working time: 386 days 23 hours 30 minutes
```

After a long cool-down, 20 mins later, "intern" temp decreased nicely, so it must be correct, while "current flow" is still high, I don't know what it could be:
```
  Warm water current temperature (intern): 43.0 °C
  Warm water current temperature (extern): 0.0 °C
  Current flow temperature: 54.0 °C
```

realthk, 2020-11-19:
"Current flow temp" seems to be in line with the value displayed on the boiler itself: even though DHW is set to 52C, after having a shower, at the end the boiler was displaying 61C - and that is what "Current flow temperature" shows still, long after it was reached.

MichaelDvP, 2020-11-20:
>Do you mean watch raw command?
I've issued that command right after starting hot water:

I've meant only a read to the E4, but this watch is better. It's always a good idea to start the watch before changing something to get the change, now we don't see tha start of hot water.  

So far:
- service code number goes to 201 for DHW, thats right. 
- E4 pos 11 is 5, here we can extract the boilerState for heating_active (03) and _tapwater_active (05) and idle (00)

But there is a lot of unknown stuff in that telegrams
- E4 pos 13: looks like another temperature,  48.5° at the end of hot water heating
- E5 publishes unknown temperature on pos 4 and 6, pos 4 starting from 29.9° and at the end 34.1° changing often, but not much.

I'll add service code and heating_active. 
Can you also do a `read 8 E5` to see the complete E5 telegram. Try a few times to get it complete or set `watch on E5` before the read. (There is an issue in read response if the telegram is send often. I'll change the read response to check destination.)

realthk, 2020-11-20:
OK, so I've issued `watch on E5`, then `read 8 E5`
Started in idle mode (though the pump was running, because it's on weather compensated mode now, when the pump always runs, but the gas heating was not) then opened the hot water tap.

```
000+00:00:58.831 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlowPlus(0xE5), data: 01 00 20 00 00 78 00 00 00 00 00 1E EB 00 9D 3E 00 00 00 00 6B 5E 00 06 4C 64 00
000+00:00:59.060 N 3: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlowPlus(0xE5), data: 64 00 00 00 00 8A A3 (offset 25)
000+00:01:18.990 N 4: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 00 20 00 00 78 00 00 00 00 00 1E EB 00 9D 3E 00 00 00 00 6B 5E 00 06 4C 64 00
000+00:01:19.275 N 5: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 00 8A A3 (offset 29)
000+00:02:19.039 N 6: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 00 20 00 00 78 00 00 00 00 00 1E EB 00 9D 3E 00 00 00 00 6B 5E 00 06 4C 64 00
000+00:02:19.264 N 7: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 00 8A A3 (offset 29)
000+00:03:15.952 N 8: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 40 00 00 78 00 00 00 00 00 1E EB 00 9D 3E 00 00 00 00 6B 5E 00 06 4C 00 (offset 2)
000+00:03:16.318 N 9: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 07 (offset 26)
000+00:03:16.588 N 10: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 0F (offset 26)
000+00:03:16.768 N 11: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 14 (offset 26)
000+00:03:16.953 N 12: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 19 (offset 26)
000+00:03:17.180 N 13: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 1E (offset 26)
000+00:03:17.354 N 14: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 23 (offset 26)
000+00:03:17.540 N 15: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 28 (offset 26)
000+00:03:17.801 N 16: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 2D (offset 26)
000+00:03:17.981 N 17: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 32 (offset 26)
000+00:03:18.187 N 18: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 00 40 00 00 78 00 00 00 00 00 1E EB 00 9D 3E 00 00 00 00 6B 5E 00 06 4C 00 37
000+00:03:18.464 N 19: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 00 8A A3 (offset 29)
000+00:03:18.656 N 20: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 43 (offset 26)
000+00:03:18.842 N 21: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 48 (offset 26)
000+00:03:19.043 N 22: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 4D (offset 26)
000+00:03:19.225 N 23: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 50 (offset 26)
000+00:03:19.403 N 24: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 55 (offset 26)
000+00:03:19.683 N 25: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 5C (offset 26)
000+00:03:19.861 N 26: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 61 (offset 26)
000+00:03:20.044 N 27: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 64 (offset 26)
000+00:03:20.351 N 28: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 60 00 00 78 00 00 00 00 00 1E EB 00 9D 3E 00 00 00 00 6B 5E 00 06 4C 64 (offset 2)
000+00:03:20.528 N 29: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 98 (offset 4)
000+00:03:21.143 N 30: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 64 00 01 98 00 D2 (offset 2)
000+00:03:21.329 N 31: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 19 (offset 6)
000+00:03:21.599 N 32: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 52 (offset 6)
000+00:03:21.780 N 33: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 66 (offset 6)
000+00:03:21.959 N 34: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 75 (offset 6)
000+00:03:22.240 N 35: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 7E (offset 6)
000+00:03:22.420 N 36: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 85 (offset 6)
000+00:03:22.811 N 37: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 8F (offset 6)
000+00:03:23.171 N 38: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 6D (offset 2)
000+00:03:23.853 N 39: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 99 (offset 6)
000+00:03:24.683 N 40: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 65 (offset 2)
000+00:03:29.062 N 41: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 02 CA 01 EC (offset 4)
000+00:03:29.241 N 42: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 02 08 (offset 6)
000+00:03:29.701 N 43: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 02 65 (offset 6)
000+00:03:29.881 N 44: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 02 7A (offset 6)
000+00:03:30.102 N 45: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 02 89 (offset 6)
000+00:03:30.462 N 46: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 02 9B (offset 6)
[... cut ...]
000+00:03:51.264 N 54: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 64 00 00 7E 02 19 (offset 2)
000+00:03:51.751 N 55: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 B3 (offset 6)
000+00:03:52.114 N 56: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 79 (offset 6)
000+00:03:52.501 N 57: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE5), data: 01 4B (offset 6)
ems-esp:/# Connection closed by foreign host.
```
(connection was lost at this point. The water was still just lukewarm)

realthk, 2020-11-21:
I've noticed another slight issue with this boiler/thermostat combo: `show values` command displays the _target flow temp_ selected by the boiler:
```
Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05)
  Time: 08:48:24 21/11/2020
  Damped outdoor temperature: 1.0 °C
  Building: medium
  Warm water mode: own_prog
  Heating circuit 1:
    Setpoint room temperature: 19.5 °C
    Current room temperature: 21.2 °C
    Comfort temperature: 20.5 °C
    Eco temperature: 18.0 °C
    Manual temperature: 19.5 °C
    Target flow temperature: 36 °C
    Mode: manual
    Mode type: comfort
```

which could be useful to see why the heating water has a certain temp, but this is not published as an mqtt thermostat sensor.

There is a `sensor.boiler_selected_flow_temperature` but when the boiler is in DHW mode, that changes its value to the domestic hot water of course, while a `sensor.thermostat_target_flow_temperature` would belong to the heating only.

MichaelDvP, 2020-11-21:
The e5 is iteresting, but i can only identfy one value:
N31-N51: a temperature rises from 28.1 to 71.6°C and than drops suddenly in N55-57 to 33.1°C. 
This is the exhaust temperature

N9-N27: within 4min something goes from 7-100% (valve/pump/fan?) 
Can someone with more experience with DHW-flow-systems identify it?

I push a update to dev-build, also include the HA for target flow temperature.

realthk, 2020-11-21:
Thanks, great progress: much more detailed info in show values, and also HA sensors!  `sensor.thermostat_hc1_target_flow_temperature` works, and it's useful

But now heating_active is always off, even though gas burner currently works at 15% in heating mode:
```
Boiler: Bosch Logamax Plus GB122 (DeviceID:0x08 ProductID:234, Version:05.03)
  Heating active: off
  Warm water/DHW active: off
  Service code: --
  Service code number: 200
  Selected flow temperature: 36 °C
  Burner selected max power: 100 %
  Burner current power: 15 %
  Pump modulation: 100 %
  Heat pump modulation: 10 %
  Outside temperature: -2.2 °C
  Current flow temperature: 36.9 °C
  Exhaust temperature: 12.6 °C
  Gas: on
  Flame current: 16.2 uA
  Pump: on
  Fan: on
  Ignition: off
  Heating activated: on
  Heating temperature setting: 60 °C
  Burner min period: 10 min
  Burner min power: 1 %
  Burner max power: 0 %
  Temperature hysteresis on: -6 °C
  Temperature hysteresis off: 6 °C
  Burner starts: 7982
  Total burner operating time: 28 days 8 hours 25 minutes
  Total heat operating time: 19 days 11 hours 9 minutes
  Total UBA working time: 389 days 3 hours 23 minutes
  Warm water selected temperature: 0 °C
  Warm water set temperature: 52 °C
  Warm water disinfection temperature: 70 °C
  Warm water circulation pump available: off
  Warm water circulation pump freq: 0x3min
  Warm water circulation active: off
  Warm water current temperature (intern): 30.2 °C
  Warm water current temperature (extern): 0.0 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: off
  Warm water heating: off
  Warm water starts: 12796
  Warm Water active time: 4 days 9 hours 22 minutes

Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05)
  Time: 22:16:08 21/11/2020
  Damped outdoor temperature: 0.6 °C
  Min ext. temperature: -10.0 °C
  Building: medium
  Warm water mode: own_prog
  Warm water circuit 1 extra: 0
  Warm water circulation mode: own_prog
  Heating circuit 1:
    Setpoint room temperature: 18.5 °C
    Current room temperature: 23.2 °C
    Comfort temperature: 20.5 °C
    Eco temperature: 18.0 °C
    Manual temperature: 18.5 °C
    Nofrost temperature: 5 °C
    Target flow temperature: 36 °C
    Offset temperature: 0 °C
    Design temperature: 45 °C
    Room influence: 5 °C
[... 185 lines cut ...]
  Temperature hysteresis on: -6 °C
  Temperature hysteresis off: 6 °C
  Burner starts: 7983
  Total burner operating time: 28 days 8 hours 52 minutes
  Total heat operating time: 19 days 11 hours 35 minutes
  Total UBA working time: 389 days 4 hours 9 minutes
  Warm water selected temperature: 0 °C
  Warm water set temperature: 52 °C
  Warm water disinfection temperature: 70 °C
  Warm water circulation pump available: off
  Warm water circulation pump freq: 0x3min
  Warm water circulation active: off
  Warm water current temperature (intern): 53.2 °C
  Warm water current temperature (extern): 0.0 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: on
  Warm water recharging: off
  Warm water temperature ok: on
  Warm water heating: on
  Warm water starts: 12797
  Warm Water active time: 4 days 9 hours 23 minutes

```

MichaelDvP, 2020-11-22:
There are a lot of codes for service number, 203 is idle, 204 is idle and temperature above setpoint. Also in normal action there is 283-burner starting, 284-gas-valves opening, 305 delay after dhw-priority. Other codes are error messages.
For checking heating and tap-water it's better to use the (hopefully now working) special publishes.

glitter-ball, 2020-11-22:
> The boiler currently doesn't do anything, but heating active is still on:

Is this still part of the issue? The blue LED on my boiler will light if heating is active. But, the energy saving pump program may shut the pump off (it runs for about 4 minutes in every 10 when the burner is off...) and the burner. So the boiler appears active for heating, but is actually doing nothing. Is this what you're seeing?

realthk, 2020-11-22:
What I'd like to see is when gas is being burnt for heating, then mark the state of the boiler as "heating".
If it's only the pump running (or gas is on but for DHW, not heating), then no.

Now the pump (or rather the whole boiler controlled by a system thermostat) can behave in two very different modes:
- controlled by the room temperature only. In this mode the gas is shut down when the room is warm enough, and the pump works for a few extra minutes only, then it also stops.
- weather compensated mode. This is a bit weird, as the thermostat instructs the boiler to keep a **constant flow**: so in this mode, the pump never stops... The requited water temp depends on the outside temperature, but often so low (around 30C or even less) that a boiler with radiators soon overshoots it even at its lowest power setting, so the gas is switched off, and only the water is circulated until it cools down (more than 5C below the required temp) then the gas is switched back again - and so on...
This might work well with floor heating, but not so much with radiators, so I finally switched back to norm mode, controlled by room temp only.

glitter-ball, 2020-11-22:
If you're using HA, then you could define a new sensor which defines heating as (burner=on) AND (diverter=CH) rather than HW, or whatever your boiler uses to switch between heating and hot water. Then you'd definitely get what you're after, if it's not available in the MQTT info from EMS-ESP.

Your pump stuff makes sense. I'm using weather compensation with room compensation, so probably the first option. With the second one, the pump may run constantly because the system has no way of knowing what the room temperature is and therefore has to circulate water constantly.

You can change the hysteresis settings to stop the boiler cycling too much at low loads. Mine is set to 10ºC / 10 minutes so the flow temp has to drop 10ºC and the boiler wait 10 minutes between cycles. When the boiler's firing, it cuts out when the flow temp exceeds the setpoint by 4ºC. In mild weather, the boiler does short cycles like the one it's just done on the right here...
[image]
There's no avoiding it - the burner just can't turn down any lower - and that's with room compensation on. Good advert for not buying a boiler that is too big! But the hot water tank heats quickly *8).

glitter-ball, 2020-11-22:
Service Code 200 (CH active) on my boiler also mirrors heating pretty well.
[image]

realthk, 2020-11-22:
Ah, your pump has variable speed, that's nice. 
My house is quite small (<75 m2, and from this year with improved insulation) so the minimum 3 kW power of the Bosch 2300 boiler is indeed way too high for maintaining the low temperatures the weather compensated control of the Bosch CW400 thermostat instructs. The boiler switched the gas on-off even during the night many times with steep rise of water temp even at the lowest power:
[image]

To make things worse, this small Bosch boiler cannot modulate its pump, so that runs always at the speed manually set on it - and this CW400 thermostat won't switch off the pump even though its weather compensated mode is always influenced by the room temperature. Perhaps your thermostat or boiler is more advanced, but no such hysteresis/wait timer setting on mine. It keeps on working if the thermostat instructs it to do so.

You are right, for DHW one has to have at least a 24 kW boiler if there is no hot water storage.

So it seems I have to avoid this kind of weather compensated control and stick to a room-based only mode.

Yes, thanks, I'll create HA status based on service code (I used  `{{ is_state('sensor.boiler_gas', 'on') and is_state('sensor.boiler_warm_water_charging', 'off') }}` but that wasn't fool proof, sometimes the gas is on after water charging for a short time)

glitter-ball, 2020-11-22:
That's a pain. We are using the same thermostat (it's called Sense II in the UK) and all the boiler settings are hiding in the Commissioning menu of the CW400 with only one exception for me (hot water eco/preheat). Maybe you can find some of the settings in there or maybe, they're just not there on your boiler *8(. My boiler's minimum output is 30% (7.2kW) so control is difficult in mild weather.

Also, I had to experiment a bit with mine to make it switch off completely overnight (unless it gets very cold). There are setback and summer settings that were very confusing and Worcester-Bosch in the UK couldn't explain them all! Anyway, it does go to sleep completely at night now. 

I use weather compensation with room compensation, what the manual describes as "Weather-compensated with the influence of the room temperature: Automated control of the flow temperature based on the outside temperature and room emperature.
Installation of the controller in the reference room is required."

Sometimes in combi boilers ('flow' boilers?) the burner runs for a short period after the tap is turned off to reheat the plate heat exchanger so that hot water flows immediately when the tap is next turned on. UK boilers have an ECO setting to disable this. That might be why your burner runs after hot water for a short period.

realthk, 2020-11-22:
Wow, you're absolutely right, thanks!
I've just changed the so called "setback type" to "room temp threshold" (from the default "reduced mode") on the CW400, and the pump is finally switched off after its 3 mins run-on time!  With a setback temp the room won't go below during the night, the pump (and the boiler of course) will remain off until morning, splendid!

Yes, the manual is very confusing indeed, with hardly any real-world examples... Bosch Hungary told me they have no direct end-user support, just gave phone number of local service person, who also turned out to be unhelpful (even rude a bit)

The boiler running all night was a show-stopper for me, I was thinking about sending back this thermostat and get a simple room-only one instead - but now it worths to tune its curve settings a bit more, because otherwise the constant flow is not all bad during the day: it is more comfortable with the radiators not getting too cold. 

(yes, my "continental Bosch" boiler also have that eco mode, I'm using it)

glitter-ball, 2020-11-23:
Cool! Stick with it. It took me a few weeks to work out the right settings.

I have radiators too and it’s great to have them at the right temperature rather than on/off. House is much more comfortable.

EMS-ESP and HA is really handy to see what’s happening and tune the settings.

levran, 2020-11-28:
Junkers Cerapur Compact ZWB 24.
Very interesting, now on version v2.1.1b6 the heating is turned off immediately after the start.
An increase in temperature and a command (call boiler heatingactivated on) to turn on the heating does not give a reaction. 
The most interesting thing is that you can now change the temperature on the boiler itself, but the heating remains inactive.

levran, 2020-11-28:
returned the firmware v2.1.0, the heating was immediately activated.

MichaelDvP, 2020-11-28:
Indeed very interesting, ems-esp does not send any commands to the boiler on start. 
Can you provide logs, system info nd values?

levran, 2020-11-28:
on v2.1.0

┌──────────────────────────────────────────┐
│ EMS-ESP version 2.1.0                    │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

ems-esp:/$ show
EMS-ESP version 2.1.0

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:43.05)
  Heating active: on
  Warm water/DHW active: off
  Service code:
  Service code number: 0
  Warm water selected temperature: 51 °C
  Warm water set temperature: 51 °C
  Warm water disinfection temperature: 75 °C
  Selected flow temperature: 38 °C
  Burner selected max power: 100 %
  Burner current power: 24 %
  Pump modulation: 100 %
  Warm water type: flow
  Warm water charging type: 3-way valve
  Warm water circulation pump available: off
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: off
  Warm water current temperature (intern): 50.5 °C
  Warm water current tap water flow: 0.0 l/min
  Current flow temperature: 26.1 °C
  Warm water storage temperature (intern): 50.5 °C
  Warm water activated: off
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: off
  Warm water active: off
  Gas: on
  Pump: on
  Fan: on
  Ignition: off
  Warm water heating: off
  Heating activated: on
  Heating temperature setting: 38 °C
  Circuit pump modulation max power: 100 %
  Circuit pump modulation min power: 10 %
  Circuit pump delay time: 3 min
  Burner min period: 15 min
  Burner min power: 0 %
  Burner max power: 100 %
  Temperature hysteresis on: -5 °C
  Temperature hysteresis off: 0 °C
  Warm water starts: 39063
  Warm water active time: 85027
  Burner starts: 111771
  Warm Water active time: 59 days 1 hours 7 minutes
  Total burner operating time: 384 days 23 hours 56 minutes
  Total heat operating time: 325 days 22 hours 49 minutes
  Total UBA working time: 1366 days 19 hours 58 minutes

{
  "System": {
    "version": "2.1.0",
    "uptime": "000+00:03:35.680",
    "freemem": 66
  },
  "Settings": {
[... 148 lines cut ...]
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 60,
    "#read requests sent": 20,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "HT3 (DeviceID:0x08, ProductID:95, Version:43.05)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:43.05)",
      "handlers": ""
    }
  ]
}

levran, 2020-11-28:
hmm, I was in a hurry.
after 10 seconds it turned out that the heating became active.
Now, as before, I cannot influence it using the command call boiler heatingactivated on / call boiler heatingactivated off

it is interesting that if you press reset on esp32, then the heating icon on the boiler also disappears for a couple of seconds.

MichaelDvP, 2020-11-29:
> Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:43.05)

There seems to be something wrong with the beta in device detection. This should be the same as in v2.1.
> Service code:
> Service code number: 0

Seems your boiler uses some other telegrams. Can you please type in terminal `watch on 08` and post the output of a minute watching.

> Now, as before, I cannot influence it using the command call boiler heatingactivated on / call boiler heatingactivated off

This command does not work on all boilers. For my GB125 the command is accepted by the boiler, but withing a few secondes overwritten by the controller (MC10). Therefore it's a read-only state. Maybe something similar with your system.

levran, 2020-11-29:
┌──────────────────────────────────────────┐
│ EMS-ESP version 2.1.1b6                  │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

ems-esp:/$ watch on 08
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x08
000+00:03:52.858 N 0: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18),                                                                                                                              data: 25 01 15 64 18 09 02 25 C0 01 AB 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
000+00:03:53.068 N 1: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), da                                                                                                                             ta: 33 01 AB 80 00 80 00 00 01 00 01 4C 94 00 98 BF 00
000+00:04:00.049 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), d                                                                                                                             ata: 1E 0E 12
000+00:04:00.157 N 3: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), da                                                                                                                             ta: FF 25 64 00 00 FB 0F 01 03 64 0A 02
000+00:04:00.295 N 4: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), d                                                                                                                             ata: 80 00 80 00 80 00 FF FF 00 64 01 B4 D8 08 77 46 00 00 00 07 2A B2 01 1C 19                                                                                                                              80 00
000+00:04:00.423 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), d                                                                                                                             ata: 80 00 (offset 25)
000+00:04:00.528 N 6: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), d                                                                                                                             ata: 08 00 33 00 00 00 00 02 4B 00 FF FF
000+00:04:02.888 N 7: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 25 01 15 64 18 09 02 25 C0 01 AA 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
000+00:04:03.129 N 8: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 33 01 AA 80 00 80 00 00 01 00 01 4C 94 00 98 BF 00
000+00:04:12.840 N 9: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:04:13.070 N 10: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 25 01 15 64 18 09 02 25 C0 01 A9 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
000+00:04:13.279 N 11: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 33 01 A9 80 00 80 00 00 01 00 01 4C 94 00 98 BF 00
000+00:04:13.509 N 12: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 00 64 01 B4 D8 08 77 46 00 00 00 07 2A B2 01 1C 19 80 00
000+00:04:22.848 N 13: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 25 01 15 64 18 09 02 25 C0 01 A8 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
000+00:04:23.057 N 14: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 33 01 A8 80 00 80 00 00 01 00 01 4C 94 00 98 BF 00
000+00:04:32.844 N 15: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 25 01 15 64 18 09 02 25 C0 01 A6 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
000+00:04:33.053 N 16: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 33 01 A5 80 00 80 00 00 01 00 01 4C 94 00 98 BF 00
000+00:04:42.874 N 17: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 25 01 15 64 18 09 02 25 C0 01 A4 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
000+00:04:43.114 N 18: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 33 01 A4 80 00 80 00 00 01 00 01 4C 94 00 98 BF 00
000+00:04:52.837 N 19: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 25 01 16 64 18 09 02 25 C0 01 A3 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
000+00:04:53.046 N 20: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 33 01 A3 80 00 80 00 00 01 00 01 4C 94 00 98 BF 00
000+00:05:00.092 N 21: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 1E 0E 13
000+00:05:00.225 N 22: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 25 64 00 00 FB 0F 01 03 64 0A 02
000+00:05:00.369 N 23: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 00 64 01 B4 D8 08 77 47 00 00 00 07 2A B3 01 1C 19 80 00
000+00:05:00.495 N 24: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 80 00 (offset 25)
000+00:05:00.602 N 25: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 00 33 00 00 00 00 02 4B 00 FF FF
000+00:05:02.866 N 26: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 25 01 16 64 18 09 02 25 C0 01 A2 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
000+00:05:03.107 N 27: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 33 01 A2 80 00 80 00 00 01 00 01 4C 94 00 98 BF 00
000+00:05:12.818 N 28: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
000+00:07:00.618 N 76: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 00 33 00 00 00 00 02 4B 00 FF FF
000+00:07:02.788 N 77: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 1D 01 25 64 00 00 00 20 C0 01 92 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+00:07:02.998 N 78: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 33 01 92 80 00 80 00 00 01 00 01 4C 94 00 98 BF 00
000+00:07:04.600 N 79: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 1D 01 27 64 00 00 00 20 C0 01 92 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
ems-esp:/$

levran, 2020-11-29:
> This command does not work on all boilers. For my GB125 the command is accepted by the boiler, but withing a few secondes overwritten by the controller (MC10). Therefore it's a read-only state. Maybe something similar with your system.

since I do not have a thermostat, I really hope this command will work.

MichaelDvP, 2020-11-30:
> since I do not have a thermostat, I really hope this command will work.

On my system it's not the thermostat, it's the boiler control panel that overrides any setting of heatingActivated.
( look at [this image](https://bbqkees-electronics.nl/wp-content/uploads/ems-devices/bc10-e1581426090307.jpg), the right rotary control set heating on/off and nothing by software can change it).
If my thermostat wants to stop heating it sets the flowtemp to zero. I think this should work always.

The 0x18 telegram indeed does not publish a service-code. Is your controlpanel capable to show the service code?

levran, 2020-11-30:
there is sample of control panel [Panel

](https://www.google.com/url?sa=i&url=https%3A%2F%2Fallegro.pl%2Foferta%2Fjunkers-cerapur-compact-zwb-24-1-de-7736900763-9523380320&psig=AOvVaw3TQSSkkG64e8KCknr3i05j&ust=1606821154449000&source=images&cd=vfe&ved=0CAIQjRxqFwoTCMCUhrSRqu0CFQAAAAAdAAAAABAR)

> The 0x18 telegram indeed does not publish a service-code. Is your controlpanel capable to show the service code?
I don't understand how to find out.
Is this the device ID on the EMS network?

levran, 2020-11-30:
[image]

proddy, 2020-11-30:
>> Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:43.05)

> There seems to be something wrong with the beta in device detection. This should be the same as in v2.1.

I made a test to simulate your HT3 boiler and gateway and can't reproduce why the name's are switched. 
Do you get `Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:43.05)` all the time?

levran, 2020-12-01:
> I made a test to simulate your HT3 boiler and gateway and can't reproduce why the name's are switched.
> Do you get `Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:43.05)` all the time?
I got Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:43.05) on 2.1.1b6 firmware
and Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:43.05) on EMS-ESP version 2.1.0

levran, 2020-12-02:
> call boiler flowtemp 0

I am really HAPPY!!!
THANK YOU SO MUCH!!!!

call boiler flowtemp 0
[image]

call boiler flowtemp 33
[image]

it turns out that **I don't need the command "heatingactivated "** to emulate relay.

closed 2021-01-05 by proddy