emsesp/EMS-ESP #621: syslog - missing events

susisstrolch, 2020-11-19:
syslog is missing its purpose. IMHO it should be seen as a service independand of web, telnet and mqtt and send meaningfull information about system status. As those are:

- log info on OTA / WEB update
- EMS protocol independant of "log xx" or "watch yy"

proddy, 2020-11-19:
probably worth to write up how logging works today and agree on a format. syslog is an extension to the logger component. So what you see in the Console is echo'd to syslog. And the inheritance works like all->trace->debug->info->notice->warning->err->alert->crit->emerg->off

Some thoughts
 TRACE: show Rx and Tx in their raw format. No additional text
 DEBUG: only show messages of the internal workings, used to troubleshoot issues
 INFO: used to show services on/off, settings changed
 WARNING: used when something can't be parsed and is ignored, e.g. failed CRC check on incoming Rx
 ERR: the bad things

We keep `watch` and this only used in TRACE

problems:
1. what happens to the 'pretty' format (e.g. Boiler -> Thermostat: UBAFast ....)
2. turning on TRACE will dump all the DEBUG messages too, and there may be many

susisstrolch, 2020-11-20:
> DEBUG: only show messages of the internal workings, used to troubleshoot issues

I'd include the Rx/Tx in debug, because that's what you need for troubleshooting

> TRACE: show Rx and Tx in their raw format. No additional text

DEBUG and also the the descriptive text.

MichaelDvP, 2020-11-20:
@proddy 
> turning on TRACE will dump all the DEBUG messages too, and there may be many

apart from mqtt and other stuff, a lot of the debug messages are incomming telegrams in `EMSdevice::handle_telegram` `Received:...`
In log trace the same telegram is displayed in pretty format with the telegram name, this is double. 
```
000+00:24:56.084 T 6: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 30 01 E2 7D 00 21 00 00 03 00 01 1C 0B 00 18 F2
000+00:24:56.084 D 7: [emsesp] Received UBAMonitorWW
```
Maybe skip the telegram name from debug (line 335 in emsdevices.cpp), than debug shows all except incomming telegrams, trace add these?

susisstrolch, 2020-11-20:
Yep, but that's the red pill you have to swallow if you set logging to "TRACE".
And - you would see non-registered and registered telegram types.
Imho, TRACE and DEBUG are really what they are meant for: debugging and analysing.

proddy, 2020-11-20:
@susisstrolch not sure you understand, when the log level is TRACE you *will* also see all the DEBUG messages. So if you want a nice dump of just raw telegrams in the syslog, forget it :-)

Then how about we use DEBUG->INFO->NOTICE->WARNING->ERR->OFF and remove the other log levels?

DEBUG - is what is debug now
INFO - shows the Rx telegrams (choose between raw or pretty)
NOTICE - all other messages, like "MQTT connected"
...

MichaelDvP, 2020-11-21:
I usually use log info and add sometimes a watch with filtering. 

Log debug mainly for mqtt messages, for testing the publish on change, etc. The telegram messages in log debug gives me no value, it's only name but no telegram info.  In log debug + watch the the telegram names are double. 

But this in terminal with changing the watches interactive. 
For syslog, if i check later, it is usefull to have all telegrams in a loglevel all and the debug informations.

proddy, 2020-11-21:
then let's keep it at that. Change the default in telnet to INFO, timeout 1hr and remove the telegram handler names from DEBUG, for everyone. And have a look at other DEBUG messages that are not very useful to show which could be removed (unless compiled with -DEMSESP_DEBUG).

I also noticed that there's a keep-alive option in ptty that could also prevent the telnet session from ending.

MichaelDvP, 2020-11-22:
Ok, i've checked the debug-messages and find all usefull. Added a message for dallas, and maybe the mqtt-commands (line 233 in mqtt.cpp) can be usefull (leave it for now as EMSESP_DEBUG).

proddy, 2020-11-22:
looks great, I ran a few scenarios (log trace with watch raw) and it works well. Thanks Michael.

closed 2020-11-29 by proddy