emsesp/EMS-ESP #622: New Feature: Configuration service

hpanther, 2020-11-20:
This ticket is to solicit comments on offering configuration services for EMS devices.

Specifically, I would like to implement a configuration page for the solar module MS-100(SM100). The approach could then be extended to other devices that require configuration.

The solar module MS-100 has a number of parameters that must be set for it to work properly. I would like to add a section "solar configuration" to the web UI settings. Any setting made here would then be stored on the gateway. On device startup, these settings would be programmed to the solar module.

proddy, 2020-11-20:
Which settings in the Solar module would be configurable?

Looking at the other devices I can't think of any settings that would be useful to override when EMS-ESP restarts. Perhaps possibly setting the Master Thermostat which is only available via the Telnet.

glitter-ball, 2020-11-22:
> The solar module MS-100 has a number of parameters that must be set for it to work properly (Collector Size, Climate Zone, Maximum Tank Temperature, and more.

Would these parameters normally be stored in, and set by, the thermostat? If so, is this to cover use cases where the MS-100 is being used without a thermostat? Just trying to understand the scenario to see how it fits with other kit.

hpanther, 2020-11-22:
The MS-100 normally requires a solar controller, such as the CS-200 (standalone system) or the CW-400 (system controller) to configure it. Adding this support would allow to go without the solar controller.
Why is this useful? If you use the CS-200, you cannot connect it to the boiler, the MS-100 will refuse to work in that configuration. The CW-400 is your only option, but if you don't have a CW-400 you're out of luck. Hence the addition to the gateway.

glitter-ball, 2020-11-22:
OK. I understand that. CW400 (Sense II) is not cheap (£150+ in the UK) so loading parameters via EMS-ESP avoids that. Interesting.

proddy, 2020-11-30:
Sounds like a good feature to add then. Let me know if you need help @hpanther

proddy, 2021-03-08:
hi @hpanther did you make any progress on ths config screen for the Solar module settings?

hpanther, 2021-03-08:
Not yet. I hope to pick it up again come spring.

proddy, 2021-05-04:
ok, let us know your ideas. This will be for v3.