emsesp/EMS-ESP #624: mqtt publish 'system send "xx xx xx" results in multiple Tx telegram on ems-bus

susisstrolch, 2020-11-21:
Build: 963a08cfaa7ebf99d5184d433cf3e28a9bbf586f
publishing a 'ems-esp/system send  "0B 08 05 00 5A" results in multiple Tx telegram on the ems-bus:

```
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:04.783 D 63805: [telegram] Sending write Tx [#118], telegram: 0B 08 05 00 5A BE
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:04.847 D 63806: [emsesp] Last Tx write successful
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:04.847 D 63807: [telegram] Sending post validate read, type ID 0x05 to dest 0x08
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:04.964 D 63808: [telegram] Sending read Tx [#119], telegram: 0B 88 05 00 20 A0
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.051 D 63809: [emsesp] Last Tx read successful
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.051 T 63810: [telegram] Rx: 08 0B 05 00 66
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.051 N 63811: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0x05), data: <empty>
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.051 D 63812: [emsesp] No telegram type handler found for ID 0x05 (src 0x08)
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.142 D 63813: [mqtt] Publishing topic ems-esp/response (#11531, retain=0, try#1, size 77, pid 1)
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.163 D 63814: [telegram] Sending write Tx [#117], telegram: 0B 08 05 00 5A BE
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.246 D 63815: [emsesp] Last Tx write successful
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.358 D 63816: [telegram] Sending write Tx [#116], telegram: 0B 08 05 00 5A BE
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.445 D 63817: [emsesp] Last Tx write successful
Nov 20 13:24:46 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.553 D 63818: [telegram] Sending write Tx [#115], telegram: 0B 08 05 00 5A BE
Nov 20 13:24:47 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.646 D 63819: [emsesp] Last Tx write successful
Nov 20 13:24:47 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.780 T 63820: [telegram] Rx: 08 00 18 00 34 01 78 64 00 81 01 20 00 83 00 7D 00 80 00 01 CA FF 30 43 00 00 FF 00 00 83 00 25
Nov 20 13:24:47 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.780 T 63821: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 34 01 78 64 00 81 01 20 00 83 00 7D 00 80 00 01 CA FF 30 43 00 00 FF 00 00 83 00
Nov 20 13:24:47 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.780 D 63822: [emsesp] Received UBAMonitorFast
Nov 20 13:24:47 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:05.947 D 63823: [mqtt] Publishing topic ems-esp/boiler_data (#11532, retain=0, try#1, size 600, pid 1)
Nov 20 13:24:47 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:06.060 T 63824: [telegram] Rx: 08 00 18 1B 00 00 00 00 00 00 00 30 00 00 13
Nov 20 13:24:47 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:06.060 T 63825: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 30 00 00 (offset 27)
Nov 20 13:24:47 192.168.254.81 1 - ems-esp - - - - ﻿000+03:32:06.060 D 63826: [emsesp] Received UBAMonitorFast

```
Those multiple Tx shows up in 63818, 63814, 63816 and 63818.
As a result (with this specific command) the boiler goes to "5P" - a locking error (too much consecutive RESET commands) which only can be reset via the SAFe reset button.

proddy, 2020-11-22:
I've reproduced this. Now need to find out why

MichaelDvP, 2020-11-23:
I've also reproduced this with this findings:
- it's multible calls of mqtt::on_message, verfied by using `LOG_DEBUG(F("[DEBUG] Received %s => %s (length %d)"), topic, message, len);`  (Maybe we should add this to normal debug.)
- changing QoS does not help
- every change in mqtt-settings causes some more repeated messages

The AsyncMqttClient adds with every call of  `Mqtt::start()` a new instance of `on_message`. 
A quick solution is to insert in `AsyncMqttClient.cpp` line 155 ` _onMessageUserCallbacks.clear();` to allow only one callback.
But a clean solution should ensure that the initialisation is done correctly ond only once for all parameters.
But i have no clean solution for now.

proddy, 2020-11-23:
yes, that's it. I found the same but didn't finish the workaround.

MichaelDvP, 2020-11-23:
Clean solution: 
`Mqtt::start()` is called on startup and by MqttSettingsService. To avoid registering the callbacks (not only on_message) add a static bool `initialized_` and in mqtt.cpp line 362: 
```
    // if MQTT disabled, quit
    if (!mqtt_enabled_ || initialized_) {
        return;
    }
    initialized_ = true;
``` 
testing now.

MichaelDvP, 2020-11-23:
Working and frees some memory too.

proddy, 2020-11-23:
great. can I make the change? I can bundle it with the other changes I have pending in that library

MichaelDvP, 2020-11-23:
Please make the change, my dev folder is not in sync, i'm testing some thermostat parameters and not sure if they are working and usefull.

proddy, 2020-11-24:
I broke something last night, no Rx anymore. I have to go through all the changes and see where the bug is.

proddy, 2020-11-24:
the change is in anyway, so closing this

closed 2020-11-24 by proddy