emsesp/EMS-ESP #625: thermostat mode

florin-laurentiu, 2020-11-22:
Hello,

for Development Build v2.1.1b4, i have this situation with RC 35 thermostat - see attached picture also:
- is afternoon time, thermostat is in AUTO MODE , on the boiler himself is DAY TIME, on the EMS -ESP is night time.
[image]

proddy, 2020-11-22:
did it work before in previous versions? Can you capture the 0x3E telegram?

florin-laurentiu, 2020-11-22:
i will chek .
thank you.

MichaelDvP, 2020-11-22:
Now i know why Bosch in newer thermostats renamed the day/nightmode to heat/eco. 

@florin-laurentiu: nightmode means the thermostat sets the lower temperature according to timer-program, it's not related to nighttime. The target flow temperature of 5°C also fits to nightmode. The modes are named as day/nightmode due to the symbol on the keys with sun and moon on it.

florin-laurentiu, 2020-11-22:
hi Michael,

please, let take it more slow to be sure that i can understood correctly:
- mentioned issue is present also in previous release xxxx.b3; at that
release, if Thermostat is OFF , is remain indication NIGHT or DAY according
with last state that was checked by EMS-ESP interface and never change
according with time that is set for DAY or NIGHT in RC thermostat. for this
reason, EMS-ESP can show NIGHT time and in fact is DAY time or vice versa.
- as soon as Thermostat is ON, the indicated time on EMS-ESP will come back
to the status that is indicated in RC Thermostat for example, and will
respect indication for DAY or NIGHT according with RCThermostat.
- i don t know if this aspect is affected the running  program that is
inside RC Thermostat related to the DAY or NIGHT time ....?

thank you,'
laurentiu

N.B.
as far i remember about setting temperature from BUDERUS specification, is
possible to set as for example 5 Celsius ( external temperature)  degree
for freezing temperature; is rezonable as value and beginning with this
temperature and considering that is follow down  that external temperature
, will be activated the frozen protection that means running circulation
pump of boiler in order to be sure that water inside heating circuit/boiler
is not frozen.
 this is related to the fact that boiler is not running and truly is not
affected by time DAY or NIGHT.
but , at this moment i m not sure that you was refer at this temperature,
so you can be more specific, please?

realthk, 2020-11-24:
> The modes are named as day/nightmode due to the symbol on the keys with sun and moon on it."

BTW, if there are no dedicated keys (like on my CW400) I guess there is no way to manually select these modes, right?

It would a convenient way to switch to "eco mode" when leaving the house, especially because lowering the setpoint is not the same in weather compensated mode (lowering the setpoint only lowers the required flow temp, but the boiler remains is heating mode, with the pump running. In eco mode though, the boiler/pump is switched off, if the room is still warm enough)

But I see no way in the manual to select this, seems as if it could be selected by the thermostat only.

MichaelDvP, 2020-11-25:
@realthk The CW400 has auto/manual keys. And you're right, lowering the setpoint only lowers the flowtemp and does not stop the pump. In RC35 the nightmode normally also lowers the setpoint/flowtemp and does not stop the pumpe, but there is a setting `reducemode` how this is handled: reduce, room-hold, outsidetemp-hold. In the hold-modes the pump is off until the temperature is reached. I havn't tested these modes yet.

realthk, 2020-11-25:
@MichaelDvP My hours away from home are not the same every week, otherwise it won't be a problem, with a schedule selecting "eco mode" for those hours. I've created a 2nd schedule-program with all-day in eco mode, that would do fine if selected when leaving the house - but switching between programs is deep in the menu of the CW400 so won't bother with it (it cannot have "favs" shortcut as well)

I guess you cannot switch between schedule programs by EMS telegrams, right? 
Or selecting "room temp only" mode for the thermostat instead of weather compensated by EMS telegrams?
Or any chance that `reducemode` you mention for RC35 could be used on CW400 as well? 
Or putting the thermostat to an "off" mode (but setting it to off in Home Assistant results "manual mode", so I think this thermostat does not support it) 

If none of these, then I'll stick with a lowered setpoint (and running pump) for regular away hours and manually selecting the all-day long eco schedule program when leaving the house for several days.

MichaelDvP, 2020-11-26:
@realthk Changing the program by ems should be possible. For RC35 (3F-offset84) and RC20(AD offset 13) the telegrams are known, but for CW400 you have to fetch it: Set `watch on 10` and change the schedule-program through some settings.

florin-laurentiu, 2020-11-28:
hi Michael,
please, find my comments that i remarq in my log, regarding interface
EMS-ESO:

1.       First of all, need to be defined very clear, from very beginning,
which kind of boiler we use:

Gas boiler

Electric boiler

Heat pump in every version that we can found on market, means air to air,
air to water or water to water

There is another kind also, this kind of hybrid boiler, based on a heat
pump air to water for example that is combined with a gas boiler.

At this point, in my opinion, it will be great that the end user can choose
which kind of boiler is using, another way will be confused by the
parameter that exists for boiler devices and is not used all for one kind
or another of boiler.

As soon as the user will decide which kind of boiler to use, I'm sure that
will be for each type of boiler , less parameters that will be more easy to
understand or set.

2.       For solar module, it will be great , if possible, to have as for
example in RC 35 daily reports of energy gained from panels and more, also
if possible energy gained per week/day/hour.

3.       For thermostat, we need to know more from interface, if possible:

Where is sensor no.1, no2...this to can be edited for example by user

Which kind of correction is used, means, external, room temperature.....

Just now, I need to explain my wishes to this great piece of art that is
quickly growing up in the good sense of an end user, in the experience way
of last stable firmware and all released intermediare firmware, till
V2.1.1B6 .

Integration on HOMEASSIST:

-          Each time when is performed an update for EMS-ESP firmware, the
only thermostat is appear most of the time with different number of
firmware in HOMEASSIST: if it was last stable release firmware for example
2.1.0 and We go to next superior one or latest one can remain in HOMEASSIST
2.1.0; need to up date or downgrade more time the firmware and finally will
respect the last firmware writed, as for the devices that is discover in
MQTT.

-          Beginning with 2.1.1 firmware, status of EMS-ESP device in MQTT
of HOMEASSISTANT is continuously/all time UNAVAILABLE.

-          In the overview mode page, looking to thermostat associated to
boiler thermostat, nothing is working beginning with superior update of
revision 2.1.0; more of this even in the last firmware which is running
more of less( 2.1.0) setting temperature in mentioned page of
HOMEASSISTANT, will switch night mode to day mode even you declare that
temperature must have same value day and night. More of this , the
temperature that is set at that time in the thermostat will remain seated
on the thermostat -for example RC35; need to set again in Homeassistant
temperature to put a different value there.

 i hope that is intelligible what i want to explain; I wait more confident
in the new firmware that will be released to improve the functionality of
interface and his integration on HOMEASSIST.

thank you,

laurentiu

florin-laurentiu, 2020-11-30:
Hello Michael,

I need a favor from you:

Need to use EMS-ESP interface as it was made , but with an extra option
that need to be hardware , not software, because for software part , can be
done by HOMEASSISTANT.

I need to START /STOP Buderus boiler from external hardware switch/contact
which is ON/OFF; ON contact will start boiler with temperature that is set
by RC 35 in my case, and same contact in OFF position, will stop running
 the boiler. This hardware switch/contact it is belongs to the floor
heating system, and is far away from the position of boiler and EMS-ESP
interface, so transmission to the EMS-ESP need to be wireless.

This option I need from some reason:

-        As soon as is present a thermostat in the boiler system, control
will be from this one or from the internal temperature of reference room;
there is a contact in fact as option for START/STOP the boiler, but is
invalidated as soon as control of boiler is taken by thermostat.

-         need to START/STOP the boiler , according with the request of
this external switch that is the heat demanding of a hydronic heating/floor
heating system.

What do you think about my request ? It is reasonable and can be done ?

best regards,

laurentiu

MichaelDvP, 2020-11-30:
That are a lot of points. Regarding Homeassistant i'm complete clueless and can not help.
The other enhancements: Keep in mind that ems-esp is a very small processor with low resources. 
The task is mainly a gateway from ems to mqtt with some extra goodies and for me as user it do this job very good.

ems-esp publishes only data that are read from ems-bus and different boiler types publish different values. You will never get gas/fan/ignition values from a heatpump. So there is no need to select this by user. 
Energy values from solar are published by th newer SM100/200 modules, but SM10 does not have these values. Also it's not possible to calculate, cause there are some parameters missing. You can add dallas sensors and measure solar flow temperatures and try to calculate by your automation system. Numbering/naming/calibration of dallas sensors is discussed earlier and also better to do this in the higher level automation system.

For your switch hardware i think this is also something for your automation system. To read a contact with safety-low-voltage you can use a separate wemos with tasmota/espurna/espeasy software. If it switches 230VAC it's better to use something insolated like Shelly i3 or Shelly one / Sonoff mini.

florin-laurentiu, 2020-11-30:
hi Michael,

clear understood your comments.
for the moment, i was continue working with EMS-ESP; i was stoping main gas pipe for boiler, boiler stop giving alarm 6A .
after reseting alarm on boiler, everithing was OK, but in the report of EMS-ESP is not any new alarm, last one is from MAY 2020 when it was also a gas stopage. 
Sorry, just restart  EMS-ESP and now is last alam coming.

florin-laurentiu, 2020-11-30:
hello,

returning with one good news regarding EMS-ESP integrate in Homeassistant:
- as i pointed out in precedent posts, it was impossible to have EMS-ESP thermostat connected on Homeassistant; just at restart of interface ,during 30 seconds interface it was appeared connected and after is showing disconnected.
- to can investigate this, i need to install MQTT explorer and check what is happened. 
- to fix this problem it was a problem of settings: EMS-ESP - settings - EMS Bus - Tx Mode = 3. with this setting , now EMS-ESP thermostat is appear connected and available in Homeassistant.

proddy, 2020-11-30:
@florin-laurentiu please comment on #643 so I can address your Home Assistant issues.

closed 2021-01-04 by proddy