emsesp/EMS-ESP #626: Question: how to get 3.3v from ems bus?

Chuckame, 2020-11-22:
Hello there,

I've ordered the gateway last year. 

Do you know if it is possible to get 3.3v from ems bus? I've got a lm2596s voltage regulator, can I use it?

Thanks if you can help me 😁

bbqkees, 2020-11-23:
Which board revision of the Gateway do you have?

Chuckame, 2020-11-23:
V0.9, with an esp8266 or 12f

bbqkees, 2020-11-23:
Ok then you don't have a Gateway, you have an EMS interface board.
The V0.9 interface board is not designed to draw power from the bus.
There is no pin to connect to and the resistor values of R9 to R12 are not compatible with powering an ESP via the bus.
However you can use the service jack to draw power from if your boiler has one.
You can use header J5 on the board for that purpose with the voltage regulator you suggest.

Chuckame, 2020-11-23:
I haven't any jack on my boiler. But the j5 header is perfect! I will try
it

bbqkees, 2020-11-23:
The service jack has a third pin with power that is wired to the board on J5.
If you do not have a service jack and are using the EMS bus wires via the screw terminal, you do not have this third pin available so it will not work.

Chuckame, 2020-11-23:
Oh no, snif.
So no on-line power available for me :/
But the thermostat is working without battery, I hope I'll find how to get
power without a 220v adapter 😄

bbqkees, 2020-11-23:
It's possible or course. The Gateway also works on just the bus but you have to modify the whole interface board to get it working in your case.

Chuckame, 2020-11-23:
Too bad!

Oderik, 2020-11-30:
@bbqkees do you plan on evolving your modules to support bus powering? I would really like that feature. It seems a bit odd that my super cool gateway + nodemcu need power outlet while the stupid thermostat is just powered by the two wires it's connected to... :)

bbqkees, 2020-11-30:
@Oderik Yes that evolution has been available for a year now and it's called the Gateway Premium II.

You can't just draw an unsupervised amount of power from the bus for your own needs, you'll mess op the bus.
TX on the EMS bus works with a current signal, not a voltage signal. To work reliable, the EMS TX circuit must be tailored to a specific design each time.

The interface board is designed to provide a reliable bus interface for all use cases.
But being a bus-safe one-for-all solution, this does not include drawing power from that bus.

If you really want you can still modify an existing EMS interface board though and add some circuitry to power your NodeMCU from the bus.

Chuckame, 2020-12-27:
Found this page that can be useful to get 15v, and after step down to 5v or 3v3: https://emsesp.github.io/docs/#/Building-your-own-test-circuit

closed 2020-12-27 by Chuckame