emsesp/EMS-ESP #627: WLAN inactive - flashing to new version with ESP8266Flasher was not succesful

tp1de, 2020-11-23:
After 2 weeks of testing suddenly wlan was not working anymore (V2.1). So I tried to flash with ESP8266Flasher on my PC. Flashing 
itself was successful but WLAN was still not working. When I started the blue led was permanent on and not blinking for the first seconds. I tried different versions of 2.xx Wemos bin files but without success.

Finally I tried version 1.95 and this version was working fine. I could then upgrade to 2.1 with the web interface.

proddy, 2020-11-23:
are you using Home Assistant? There are some issues with 2.1, which is why we're working on getting the patch 2.1.1 patch out. Can you try that?

tp1de, 2020-11-23:
Sorry no Home Assistant. I am using my own mqtt server to connect. This is working fine and stable.

closed 2020-12-04 by tp1de