emsesp/EMS-ESP #628: ems-esp is disabeling the internet interface (integrated / km200)

tp1de, 2020-11-23:
When I connect to the ems-bus (by service jack) the km200 interface does not work anymore. So I can't access the heating system by App or Web-Interface anymore.

As far as I can recognize this happens when I restart the heating and the boiler / controller scans the bus for devices.
When the heating system is running and I connect on a running system, then everything is working fine and the km200 interface keeps running. At the end I want to connect hardwired to the ems-bus and when switching of for maintenance, the system including ems-esp should work stable. For the moment this is not secured. I feel that using the service jack address is bringing some problems as well. (....I am not sure...)

{
  "System": {
    "version": "2.1.0",
    "uptime": "003+13:22:21.021",
    "freemem": 44,
    "fragmem": 5
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 0,
    "publish_time_mixer": 10,
    "publish_time_other": 0,
    "publish_time_sensor": 10,
    "mqtt_format": 2,
    "mqtt_qos": 0,
    "mqtt_retain": "on",
    "tx_mode": 2,
    "ems_bus_id": 15,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "off",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 926224,
    "#read requests sent": 56359,
    "#write requests sent": 0,
    "#incomplete telegrams": 12,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 99,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus GB125/Logamatic MC110 (DeviceID:0x08 ProductID:133, Version:01.44)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.03)",
      "handlers": "0xA3 0x06 0xA2 0x2A5 0x2B9 0x2AF 0x2A6 0x2BA 0x2B0 0x2A7 0x2BB 0x2B1 0x2A8 0x2BC 0x2B2 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x21, ProductID:160, Version:24.05)",
      "handlers": "0x2D8"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)",
      "handlers": ""
    },
    {
      "type": "Controller",
      "name": null,
      "handlers": ""
    }
  ]
}

proddy, 2020-11-23:
This problem rings a bell, think we've seen it before. Need to refresh my memory.

tp1de, 2020-11-23:
In general I would recommend to implement a "waiting time / start delay" when restarting the wemos before scanning the bus (1-2 minutes). I had similar problems while restarting that my mm100 mixer went into error state (led blinking).

There might be interferences with the controller ems-bus scan when restarting and the ems-esp interface when running same time.

MichaelDvP, 2020-11-23:
Is there a special reason why you use 
> "ems_bus_id": 15,
This is a clock-module and maybe the KM200 tries to request a time from it. 
If you change to standard 11 (service-key) does it give the same issue?
Do you have a log (with watch) from the blocked start?

tp1de, 2020-11-23:
I started with standard 11 (service key) but had problems with mm100 module running into error state (led blinking) and errors displayed in RC310. With other id's this did not happen ... but I will test again with this new development version.

> _Do you have a log (with watch) from the blocked start?_

How shall I watch on hardware reset?

tp1de, 2020-11-23:
By the way: 

> A scan devices 
kills the ems-esp interface working (error state) and my Buderus heating is in error state and needs a hardware reset.

tp1de, 2020-11-23:
So I tried different bus id's. Same behavior independent from the id:

When ems-esp is connected **while restarting heating (including bus scan)** then km200 internet gateway is disabled.
When connected **after bus scan**, then km200 is working fine.

proddy, 2020-11-23:
@tp1de what do you mean by "bus scan". Are you manually hitting the 'SCAN DEVICES" button in the web UI? Why?

tp1de, 2020-11-23:
Just by instance in the web UI driven by curiosity while have ems-esp newly bought. 
Any ems-bus scan seems to produce some trouble. Either initiated by the heating device (by switching on e.g. after maintenance) or by the web UI.

bartplessers, 2020-11-28:
same over here.
Hardware:

- Buderus GB192iT40
- RC310 thermostat
- KM200 gateway
- EMS-ESP v2.1.0

From time to time, I can not use the app connected to the gateway.
Seems that rebooting the ems on a running boiler induces this problem, but I'm not sure.

kind regards,
Bart

proddy, 2020-11-28:
when EMS-ESP starts it sends one message to the UBAMaster (boiler on device ID 0x08) requesting a telegram which shows us which other devices have been registered on the EMS bus. Then for each device we send a Tx telegram asking it for more details (which is where we get the version number and product ID).  We can try excluding the KM200 gateway from this batch of requests and see if that helps?

tp1de, 2020-11-29:
I believe that we should try to exclude KM200 gateway (0x48) from batch request. I would like to test this.
Next to the km200 problematic I have for the moment some additional severe stability problems with the mm100 mixer as well.

Actually the EMS-ESP looses connection to ems-bus every 3-4 days. Then the led on the mm100 (0x21) is blinking and a heating device failure is displayed on my RC310 Thermostat. Floorheating circuit (hc2) is not working anymore ! Sometimes while trying to reconnect to WEB-UI a reboot of the EMS-ESP device takes place. Seems to be a software stability problem. Could it be related to the mqtt interface to my mqtt server? 

A restart from WEB-UI does not solve the problem to reconnect ems-bus again. While disconnceting the hardware from service jack and then reconnecting, it works again. Should the restart option from WEB-UI not be the same as the hardware reset? (rebooting the device). While uploading the firmware again by WEB-UI the restart works fine as well (connected to ems bus). Just a restart is not reconnection to ems-bus.

Nevertheless the EMS-ESP should not stop my heating system (hc2) every 4-5 days. How could I help to debug?
For the moment I disconnected the device to have my central heating operating stable.

bartplessers, 2020-11-29:
Hi,

Situation:
- ems + mqtt + homeassistant --> ok
- gateway --> app NOT working

Now I rebooted gateway: unplugged service jack, waited few seconds and plugged in again.

- ems + mqtt + hass --> ok
- gateway + app --> ok

I will keep an eye on it for the next days.

Kind regars
Bart

glitter-ball, 2020-11-29:
Plugging and unplugging the service jack with the boiler live can short the EMS bus - may not be relevant to this problem but worth knowing.

I did it once and it caused my thermostat to cold start, presumably because it lost power due to the short. Now I'll only connect and disconnect EMS-ESP with the boiler off (which also cold starts the thermostat, but at least it isn't shorting the bus out!)

A quick short is probably OK and Bosch must have known this would happen with a 3.5mm jack plug but you might want to avoid it.

tp1de, 2020-11-30:
To add some more findings:

1. Yes I have the same experience as Bart: EMS-ESP must be connected some 30 seconds **after** start of heating system
2. I connected the EMS-ESP to USB-power and the ems-cable to the mm100 mixer, so that it keeps up running when restarting the heating system
3. The disabling of some parts of the internet interface (heating circuits / warm water) stays the same.
    Some other data is still transferred by the km200 web interface -> system preasure and temp

I might believe that the scan of the ems-bus by the master controller (mc110) is disabling km200.
Is EMS-ESP answering to the poll request of the master controller? How? 
Could this be the reason for the km200 problems?

Best regards
Thomas

proddy, 2020-11-30:
@tp1de if you're comfortable with building the firmware from source we can try and few things out, like tracing the polls and blocking the bus scan on startup. One method is to compile with the -DEMSESP_DEBUG flag set and set logging to trace and use a syslog server to see what is happening. Or if you have an ESP32 it'll be about 10 times easier to debug. Let me know and I'll help

tp1de, 2020-11-30:
@proddy  since I am retired I am fine to spend quite some time for debugging and hopefully adding more functionality. I just need some help how to do the firmware building from source and how to do the debugging. The ESP32 is new for me since I used Raspberry PI's or VM's before.

I do have the BBQKees EMS-gateway with the ESP32 since  2 weeks because I came to limitations with the km200 interface for my homeautomation system (ioBroker). It is important for me to use both in parallel and to keep it longtime stable working.

Just tell me what to do ...

MichaelDvP, 2020-12-01:
> Just tell me what to do ...

Only to analyse what's blocking, use a syslog server and connect ems-esp to this server with syslog-setting `log all`. 
And post the logs here.

To build the firmware: [https://emsesp.github.io/docs/#/Building-firmware](https://emsesp.github.io/docs/#/Building-firmware)

To try a delay on startup for the ems-bus activity you can add in `emsesp.cpp` line 926 after the `if (length == 1) {`
```
       // first send after 60 sec
        if (uuid::get_uptime_ms() < 60000ULL) {
           return;
        }
```

tp1de, 2020-12-01:
I switched off my heating system. The EMS-ESP kept on running powered by USB.
I then switched on syslog-setting to all and after a few seconds I started the heating system again.
EMS-ESP kept running and was the whole time connected to the ems-bus.

I tried to connect to km200 web interface - connection itself was succesfull but no data available and some error messages were shown. I then ended syslog-setting. But when trying to reconnect to EMS-ESP a reboot of the EMS-ESP gateway took place. 

When restarting the heating without EMS-ESP beeing powered up everything waorked fine. Connecting the gateway some 1-2 minutes later.

Here is the syslog of the startup-procedure out of my synology diskstation syslog server:

[All_2020-12-1-17 2 13.zip](https://github.com/proddy/EMS-ESP/files/5623830/All_2020-12-1-17.2.13.zip)

MichaelDvP, 2020-12-01:
If you switch ems-esp to `tx_mode 0`, only listening, is the KM200 than accessible?

tp1de, 2020-12-01:
with tx_mode 0 km200 is working but:

In dashboad recieved telegrams are shown - but no devices !
After a while WEB-UI is not accessible anymore and blue led is blinking.

A hard reset does not work.

tp1de, 2020-12-02:
Some new findings:

The km200 internet gateway itself does not seems to be the problem. The gateway is accessible and data like boiler temp, pressure etc is provided. Just all data from heating circuits and ww circuit is not available. This data comes from RC310 thermostat. The communication between gateway (0x48) and thermostat (0x10) seems to be blocked.

Going into the installers menu on the RC310 and starting a new configuration scan makes the km200 working again.
Somehow the presence of the ems-esp disturbes the communication between km200 and rc310.
Any idea why?

MichaelDvP, 2020-12-02:
Too many clients on the ems-bus? Bad bus-topologie? Just a guess.

tp1de, 2020-12-02:
I don't believe .... just the boiler with master controller, mm100 mixer 1 m distance and rc300 Thermostat with 3-4 m cable distance.
Gateway was in service jack now with 1m cable connected to ems-out on the mm100. This is not a bus topology problem. 

Too many telegrams per time send by the ems-esp gateway that blocked communication during startup?

bartplessers, 2020-12-02:
> I will keep an eye on it for the next days.
> 
> Kind regars
> Bart
> 

Small update: everthing keeps running for now, but noticed in HomeAssistant that "set temperature" is not displayed anymore:
[image]

Regards,
Bart

proddy, 2020-12-03:
@bartplessers could you do a http://ems-esp/api?device=thermostat&cmd=info

proddy, 2020-12-03:
@tp1de 

> with tx_mode 0 km200 is working but:
> 
> In dashboad recieved telegrams are shown - but no devices !
> After a while WEB-UI is not accessible anymore and blue led is blinking.
> 
> A hard reset does not work.

a TxMode of zero prevents any Tx traffic being sent out, and we need to send version requests to the Master to find out which devices are on the bus. This is why you'll see no devices.

Are you able to compile the code yourself? If so I can suggest some lines to change to see what is causing the interference with the km200 gateway

bartplessers, 2020-12-03:
> @bartplessers could you do a http://ems-esp/api?device=thermostat&cmd=info

[image]

proddy, 2020-12-03:
wow, that was quick (and fancy!). I'll create a new bug for this - don't think it has anything to do the km200.

tp1de, 2020-12-03:
@proddy 

> Are you able to compile the code yourself? If so I can suggest some lines to change to see what is causing the interference with the km200 gateway

Yes I think I should have everything installed on my PC. Visual Studio Code I am using quite some time but basically for edititing Javascript / NodeJS. I have now installed the PlatformIO plugin but I need some advice how to proceed:

- Shall I open a new project? Which board / framework?
- Then downloading the sourcecode and putting in the working directory?
- How should I start compile / build? I would like to create a bin file - no ota upload?

Please advice on how to setup the pio subsystem and what source code changes to do ... I am ready to try

bbqkees, 2020-12-03:
I have someone else with a GB192i and a RC300 who also has similar communication problems with the bus. This boiler also has a (built-in) KM200 connected.
A while ago I had another one with similar problems. So there might be a pattern here.

tp1de, 2020-12-04:
@proddy @bbqkees @MichaelDvP 
Since my gateway with the new esp32 d1 board is now working stable (#649) I am on standby to test some coding changes to avoid km200 disabeling on startup. I found out how to compile new firmware with platform io.

I wrote that I need external usb power supply for the gateway to get it in stable operations mode (#649 ).
When disabeling tx_mode = 0 while starting up the heating system -->then km200 gateway is working.

**Therefore ems-esp firmware should recognize:**
1. when heating system / ems-bus is down - but ems-esp gateway still operating because of own power supply. 
No telegrams received for 30 ---60 seconds         **--> ems-bus inactive but still connected**
On gateway startup tx_mode should be disabled **--> ems-bus inactive but connected**
2. when starting the ems-esp gateway and / or ems-bus active again (telegrams received) -**-> ems-bus active**

**For (1) tx_mode should be set temporary to 0 
For (2) tx_mode should be set back to original value of setup after some time-delay (fix 60 seconds - better parameter in setup)** 

Might this be the right logic? Please advice how I could help testing

MichaelDvP, 2020-12-04:
In [https://github.com/proddy/EMS-ESP/issues/628#issuecomment-736635450](https://github.com/proddy/EMS-ESP/issues/628#issuecomment-736635450) i told you how to delay the tx-start after powerup for 60 sec.
If you want to check the connection state (if powerd by usb) use that code in same place:
```
    static uint64_t delayed_tx_start_ = 0;
    // check for poll
    if (length == 1) {
        if (!rxservice_.bus_connected()) {
            delayed_tx_start_ = uuid::get_uptime_ms();
        }
        EMSbus::last_bus_activity(uuid::get_uptime()); // set the flag indication the EMS bus is active
        // first send delay 60 sec after connect
        if (uuid::get_uptime_ms() - delayed_tx_start_ < 60000ULL) {
            return;
        }
```
replacing these lines
```
    // check for poll
    if (length == 1) {
        EMSbus::last_bus_activity(uuid::get_uptime()); // set the flag indication the EMS bus is active
```

tp1de, 2020-12-04:
I was looking for line 926 in emsesp.cpp but I could not find the right position...
Could you help by uploading the cpp file?

MichaelDvP, 2020-12-04:
I'm refering to the dev build, in main it's around line 853.

tp1de, 2020-12-04:
Ok I tried. For a new start of the gateway tx is delayed for 60 secs. OK.
When switching boiler off , then a restart is not  recognized and tx is not delayed.  KM200 problem existent.

MichaelDvP, 2020-12-05:
The boiler switch off/on is only detected if the communication stops for 30 sec, defined in telegram.h line 191 `EMS_BUS_TIMEOUT = 30000;`  You can reduce the value, minimum to prevent false detection should be around 2 sec.

tp1de, 2020-12-05:
I switched off for 3 minutes and watching the telegrams. No telegrams recieved or transmitted.  
Switching on again after around 10 secs first tx messages were sent. No 60 secs delay.

MichaelDvP, 2020-12-05:
Sorry, i can not reproduce. If i disconnect the bus i get after some time led slow blinking and EMS-Status offline in web-ui.

tp1de, 2020-12-05:
@MichaelDvP 
I will check again. Does disconnecting means disconnecting the wires? Or switching off the heating system like I did?

MichaelDvP, 2020-12-05:
Tested with some extra debug messages, esp32 usb-powered, bus connect via servicejack, v2.1.1b6, tx-delay 30 sec, bus-timeout 5 sec.

1. unplug servicejack, the contact produces a lot of garbage on rx:
```
Dec  5 11:20:49 ems - 000+00:02:00.725 E 266: [telegram] Rx: FF 41 1E 78 60 20 7F 62 (CRC 62 != 99)
Dec  5 11:20:49 ems - 000+00:02:00.781 E 267: [telegram] Rx: 10 30 1C F8 (CRC F8 != 3C)
Dec  5 11:20:49 ems - 000+00:02:00.927 E 268: [telegram] Rx: 1E 1E 0F 3F FF 1F (CRC 1F != B4)
Dec  5 11:20:49 ems - 000+00:02:01.238 E 269: [telegram] Rx: E0 78 00 FF C0 80 F0 (CRC F0 != 24)
Dec  5 11:20:49 ems - 000+00:02:01.615 E 270: [telegram] Rx: 02 FC 80 FF 02 02 (CRC 02 != 58)
Dec  5 11:20:49 ems - 000+00:02:01.694 E 271: [telegram] Rx: 28 01 FE 61 (CRC 61 != 5C)
Dec  5 11:20:49 ems - 000+00:02:01.766 E 272: [telegram] Rx: FC A7 52 01 10 04 0F 20 A3 F5 96 44 1A 07 03 (CRC 03 != C9)
Dec  5 11:20:49 ems - 000+00:02:01.798 E 273: [telegram] Rx: 5A 98 FF 86 41 02 01 00 2E D8 FF 1B B7 82 26 (CRC 26 != 90)
Dec  5 11:20:49 ems - 000+00:02:01.860 E 274: [telegram] Rx: 00 CD C0 BD 06 02 7C 14 D7 FF FB 3B 01 2A DF CE FF 6F FE (CRC FE != D0)
Dec  5 11:20:50 ems - 000+00:02:01.938 E 275: [telegram] Rx: E0 FC FF FF 10 B8 2F FF 05 E8 EE 7F 03 F4 0F FF 7F (CRC 7F != D0)
Dec  5 11:20:50 ems - 000+00:02:01.998 E 276: [telegram] Rx: 00 FF FF 07 08 FF 01 C0 FF FF (CRC FF != E3)
Dec  5 11:20:50 ems - 000+00:02:02.858 E 277: [telegram] Rx: FF FF 3F FF (CRC FF != 0F)
```
2. about 10 sec later i get the disconnect message from system, System polls only every 5 sec, so we have Murphy and Shannon who expand this to 10 sec.
```
Dec  5 11:20:57 ems - 000+00:02:10.361 E 278: [system] Error: No connection to the EMS bus
Dec  5 11:21:17 ems - 000+00:02:30.077 D 279: [mqtt] Publishing topic ems/boiler_data (#16, retain=0, try#1, size 292, pid 17)
Dec  5 11:21:17 ems - 000+00:02:30.178 D 280: [mqtt] Publishing topic ems/boiler_data_ww (#17, retain=0, try#1, size 263, pid 18)
Dec  5 11:21:17 ems - 000+00:02:30.279 D 281: [mqtt] Publishing topic ems/thermostat_data (#18, retain=0, try#1, size 30, pid 19)
Dec  5 11:21:17 ems - 000+00:02:30.380 D 282: [mqtt] Publishing topic ems/dallassensor_data (#19, retain=0, try#1, size 176, pid 20)
```
3. plug servicejack again, first telegram is garbage, delay started and first tx is queued 23 sec later and send out 30 sec later.
```
Dec  5 11:21:23 ems - 000+00:02:36.708 E 283: [telegram] Rx: E3 00 8B 00 90 (CRC 90 != 58)
Dec  5 11:21:23 ems - 000+00:02:36.712 D 284: [emsesp] Tx delay started
Dec  5 11:21:47 ems - 000+00:03:00.006 D 285: [emsesp] Fetching values for device ID 0x08
Dec  5 11:21:47 ems - 000+00:03:00.006 D 286: [telegram] Tx read request to device 0x08 for type ID 0x14
Dec  5 11:21:47 ems - 000+00:03:00.006 D 287: [telegram] Tx read request to device 0x08 for type ID 0x16
Dec  5 11:21:47 ems - 000+00:03:00.006 D 288: [telegram] Tx read request to device 0x08 for type ID 0x19
Dec  5 11:21:47 ems - 000+00:03:00.006 D 289: [telegram] Tx read request to device 0x08 for type ID 0x33
Dec  5 11:21:47 ems - 000+00:03:00.006 D 290: [telegram] Tx read request to device 0x08 for type ID 0xE6
Dec  5 11:21:47 ems - 000+00:03:00.006 D 291: [telegram] Tx read request to device 0x08 for type ID 0xEA
Dec  5 11:21:47 ems - 000+00:03:00.006 D 292: [emsesp] Fetching values for device ID 0x09
Dec  5 11:21:47 ems - 000+00:03:00.006 D 293: [emsesp] Fetching values for device ID 0x10
Dec  5 11:21:48 ems - 000+00:03:00.006 D 294: [telegram] Tx read request to device 0x10 for type ID 0xA5
Dec  5 11:21:48 ems - 000+00:03:00.006 D 295: [telegram] Tx read request to device 0x10 for type ID 0x37
Dec  5 11:21:48 ems - 000+00:03:00.006 D 296: [emsesp] Fetching values for device ID 0x21
Dec  5 11:21:48 ems - 000+00:03:00.006 D 297: [telegram] Tx read request to device 0x21 for type ID 0xAB
Dec  5 11:21:48 ems - 000+00:03:00.053 D 298: [mqtt] Publishing topic ems/boiler_data (#20, retain=0, try#1, size 293, pid 21)
Dec  5 11:21:48 ems - 000+00:03:00.154 D 299: [mqtt] Publishing topic ems/boiler_data_ww (#21, retain=0, try#1, size 263, pid 22)
Dec  5 11:21:48 ems - 000+00:03:00.256 D 300: [mqtt] Publishing topic ems/thermostat_data (#22, retain=0, try#1, size 30, pid 23)
Dec  5 11:21:48 ems - 000+00:03:00.357 D 301: [mqtt] Publishing topic ems/dallassensor_data (#23, retain=0, try#1, size 176, pid 24)
Dec  5 11:21:48 ems - 000+00:03:00.458 D 302: [mqtt] Publishing topic ems/heartbeat (#24, retain=0, try#1, size 139, pid 25)
Dec  5 11:21:54 ems - 000+00:03:07.423 D 303: [emsesp] Toggling fetch for device ID 0x10, telegram ID 0x48 to 1
Dec  5 11:21:54 ems - 000+00:03:07.423 D 304: [emsesp] Toggling fetch for device ID 0x10, telegram ID 0x47 to 1
Dec  5 11:21:54 ems - 000+00:03:07.423 D 305: [emsesp] Toggling fetch for device ID 0x10, telegram ID 0x49 to 1
Dec  5 11:21:54 ems - 000+00:03:07.688 D 306: [telegram] Sending read Tx [#14], telegram: 0B 88 19 00 20 D0
Dec  5 11:21:54 ems - 000+00:03:07.778 D 307: [emsesp] Last Tx read successful
Dec  5 11:21:55 ems - 000+00:03:07.872 D 308: [telegram] Sending read Tx [#44], telegram: 0B 88 19 19 20 E2
Dec  5 11:21:55 ems - 000+00:03:07.951 D 309: [emsesp] Last Tx read successful
Dec  5 11:21:55 ems - 000+00:03:08.107 D 310: [telegram] Sending read Tx [#15], telegram: 0B 88 33 00 20 78
Dec  5 11:21:55 ems - 000+00:03:08.163 D 311: [emsesp] Last Tx read successful
Dec  5 11:21:55 ems - 000+00:03:08.287 D 312: [telegram] Sending read Tx [#16], telegram: 0B 88 E6 00 20 07
Dec  5 11:21:55 ems - 000+00:03:08.349 D 313: [emsesp] Last Tx read successful
```

tp1de, 2020-12-05:
OK understood you disconnect the wires. Have you removed the jumper for selecting service-jack vs. ems-bus power completely for usb-powering? (as I did like Kees told me). What happens if you keep the gateway usb-powered and connected to the service jack and then switching off the boiler?

I do not disconnect the wires but switching off the power for the heating system and keeping usb-power on for the gateway. Gateway keeps on running, devices are seen, EMS-Bus keeps to be active and web-api request are beeing kept answered.
Just no new tx/tx telegrams so values do no change. 

In case for repairs / service on my heating system power is sometimes switched of or the central power switch on the boiler is switched off for maintenance (EMS-bus down due to master contoller off).

But gateway keeps wired to the MM100 ems-bus and is usb-power stay on. This situation is not recognized as ems-bus down.

Are there any changes to the latest development version 2.1.1b6 vs. 2.1.0 I seem to use?
How do I download the latest development version for platform io? (see above)

MichaelDvP, 2020-12-05:
> Have you removed the jumper for selecting service-jack vs. ems-bus power completely for usb-powering? 

I've removed the dc-dc (older board version), result is: the esp is powered only by usb.
> What happens if you keep the gateway usb-powered and connected to the service jack and then switching off the boiler?

Bad waf. But for ems-esp that should be the same, no incomming telegrams and polls.
 > Are there any changes to the latest development version 2.1.1b6 vs. 2.1.0 I seem to use?
How do I download the latest development version for platform io? (see above)

There are a lot of changes (228 commits), but i can't remember anything in bus-detection. For the code go to [https://github.com/proddy/EMS-ESP/tree/dev](https://github.com/proddy/EMS-ESP/tree/dev) select the code-button and clone or download,

tp1de, 2020-12-05:
> For the code go to https://github.com/proddy/EMS-ESP/tree/dev select the code-button and clone or download,

That's what I have done ... but code is different then seen online in dev tree in github. 
(e.g. emsesp.cpp: Online 1023 lines vs. 953 lines in download) - Download seams to be last main version - Any idea ?

> > What happens if you keep the gateway usb-powered and connected to the service jack and then switching off the boiler?
> 
> Bad waf. But for ems-esp that should be the same, no incomming telegrams and polls.

Yes no telegrams but what is happening when you switch on again? Do you have 60 seconds delay for tx?

MichaelDvP, 2020-12-06:
> That's what I have done ... but code is different then seen online in dev tree in github.

Yes, there is something wrong, download is always the 2.1.0 main source. My forc is in sync, try download from [here](https://github.com/MichaelDvP/EMS-ESP)
> Do you have 60 seconds delay for tx?

Yes.

tp1de, 2020-12-06:
> Yes, there is something wrong, download is always the 2.1.0 main source. My forc is in sync, try download from [here](https://github.com/MichaelDvP/EMS-ESP)
- Download was working - I do have now the actual dev version 2.1.1b6
- I replaced the lines as proposed in emsesp.cpp (now it is line 923 👍 )
- Compiling and uploading now gives version 2.1.1b6 - with the changes done

But usb-powered (independent from heating system) the status is kept connected when heating system is switched of.
No TX or RX telegrams received. I waited 5 minutes - but status stays connected.
No new telegrams received is not being interpreted as "disconnected".

When switching heating on again, tx is send after approx. 7 seconds and internet gateway is blocked again.

This actual solution only works if usb-power is connected to boiler - being switched off when the boiler is switched off, or powered by service jack or ems-bus. In this case on a restart the 60 secs tx delay works. 
(But I do not want to connect a new power-plug internally to the boiler.)

Any other idea?

P.S.: In pio there are different project task compared to proddy's github account: env:esp32-ci and env:esp32
What are the differences and which one should I use?

MichaelDvP, 2020-12-06:
Hmm, maybe the floating ems-bus produces some random bytes. Or:  Is the KM200 powered seperatly (and sending something) or by the boiler? 
In first case: 
Try reducing the timeout in telegram.h to, let's say, 5 sec: `EMS_BUS_TIMEOUT = 5000;` and reset only on poll for us in emsesp.cpp: 
replace ` EMSbus::last_bus_activity(uuid::get_uptime()); // set the flag indication the EMS bus is active` by:
```
        if ((first_value ^ 0x80 ^ rxservice_.ems_mask()) == txservice_.ems_bus_id()) {
            EMSbus::last_bus_activity(uuid::get_uptime()); // set the flag indication the EMS bus is active
        }
```
> What are the differences and which one should I use?

Esp32-ci does not build the webinterface, it's faster if there is no change in interface. Use esp32 to build the complete software including web-interface and remember to reload the interface in browser after update.

tp1de, 2020-12-06:
I kept EMS_BUS_TIMEOUT unchanged. And did the coding changes.

**IT IS WORKING AS EXPECTED ! After 30 secs EMS-Bus is disconnected when heating is powered off and any restart of the heating system and / or the gateway has a delay of 60 secs for tx telegrams
KM200 problems are solved**  

Coding in ems.cpp is now from line 923 onwards: (correct?)

```
static uint64_t delayed_tx_start_ = 0;
// check for poll
 if (length == 1) {
        if (!rxservice_.bus_connected()) {
            delayed_tx_start_ = uuid::get_uptime_ms();
        }
        //EMSbus::last_bus_activity(uuid::get_uptime()); // set the flag indication the EMS bus is active
        if ((first_value ^ 0x80 ^ rxservice_.ems_mask()) == txservice_.ems_bus_id()) {
            EMSbus::last_bus_activity(uuid::get_uptime()); // set the flag indication the EMS bus is active
        }
        // first send delay 60 sec after connect
        if (uuid::get_uptime_ms() - delayed_tx_start_ < 60000ULL) {
            return;
        }

```

Are you adding this to the standard coding (new dev version)?

Topic can be closed.

MichaelDvP, 2020-12-06:
I'll set an option to the setting page for a user defined delayed tx-start ( 0 - 120 sec) in beta after new release. 
(see [https://github.com/proddy/EMS-ESP/issues/632#issuecomment-738741849](https://github.com/proddy/EMS-ESP/issues/632#issuecomment-738741849))

@bartplessers and other KM200 owners: if you need a binary in advance, tell us.

MichaelDvP, 2020-12-08:
@tp1de thanks for your help and tests, i added a tx-delay-setting to b7 settings page. 
@bartplessers can you check with your KM200 if this is a general solution.

tp1de, 2020-12-08:
@MichaelDvP 
I was not sure if the b7 bin file has all changes included. So I downloaded from your github directory and compiled myself. All working just b6 shown in web ui. 

I may recommend to set initial tx delay to  60 secs in settings.

bartplessers, 2020-12-09:
@MichaelDvP : 
- currently running EMS-ESP  v2.1.1b7
- TX delayed start: 60 seconds
- repowered my whole installation

for now, everything looks fine.
I will keep an eye on this and keep you informed.

kind regards
B

bartplessers, 2020-12-09:
addendum:
- running EMS-ESP v2.1.1b7
- TX delayed start --> back to 0 seconds
- rebooted ems-espx (NOT my whole installation, only esm-esp)

Now:
- devices found again
[image]

- API is working again
[image]

- Buderus app NOT working yet (no current temp, no schedule):

[image]

MichaelDvP, 2020-12-09:
"No EMS device found" will diappear after the  tx-delay. Without tx the devices can not be found (only if device is sending it's version information). If you set a delay you have to be patiently.

tp1de, 2020-12-09:
I am a little confused what you have done in what sequence. A reboot of ems-esp does not have any influence on km200 internet gateway. The gateway is blocked when the boiler restarts after beeing powered off and scans for devices and ems-esp sends tx telegrams are send during this time (within the first minute).

The internet gateway nedds about 5-7 minutes to send the data to the buderus server. 
Have you waited long enough?

bartplessers, 2020-12-09:
1
"No EMS device found" will disappear after the tx-delay.
I definitly waited longer than 60 seconds.

2
_The internet gateway needs about 5-7 minutes to send the data to the buderus server._
I did another test: rebooted the whole system without ems-esp attached.
Everything (including buderus app) worked fine within a minute.

On this test I attached the ems-esp AFTERWARDS. To be clear:
1. power off everything
2. power on boiler (thermostat, gateway and external sensor attached)
3. waited +/-1 minute
4. attached bbqkees ems-esp device

Everythings works fine now
- Rx/Tx: good quality
- buderus app works fine
- mqtt + hass works fine

So in my experience: the Tx delay does not have the same impact as starting the ems-esp device.

Some other thing: since I use ems-esp I do have an error on my thermostat on regulary base:
[image]
According to documentation this has something to do with external temp sensor, but I doubt that: I checked the connections and the installation is brand new.
Maybe there is some interference between some devices?

If I can be of any help to debug this: just tell me what to do.

kind regards,
Bart

tp1de, 2020-12-09:
Do you still have connected the ems-esp to the thermostat?

bartplessers, 2020-12-09:
indeed. ems-esp connected directly to thermostat
[image]

You think this will be a power issue? Should I try USB powering the device?

tp1de, 2020-12-09:
Without usb power it works for me only if connected to service jack - which I do not want.
Without USB power I have the same errors on ems bus and same messages on the display.

bartplessers, 2020-12-09:
Is it worth trying the other EMS bus? I have an orange one (attached to the thermostat) and a white one (unused). Or are they internally connected to the same power supply?

tp1de, 2020-12-09:
Additionally I got longterm stability problems with the wemos d1 mini board (esp8266) and bought an esp32 d1 mini board.
SInce then everything is fine and stable.

[12 more replies]

closed 2021-01-14 by proddy