emsesp/EMS-ESP #629: MM100 Mixer - Pumpstatus is wrong

tp1de, 2020-11-23:
I installed the last development version. But this error was existing in 2.1 stable version before.
The pumstatus for the MM100 ist wrong:

current pump status: on -> but pump is off 
                                  off -> but pump is on 

[image]

proddy, 2020-11-23:
Could you post the output of the 0x02D8 telegram? `read 21 2D8`

tp1de, 2020-11-23:
Yes, if you tell me how to do it ....

MichaelDvP, 2020-11-23:
Yes there is something wrong, pump and valve are read from the same address, pump should be pos 0 (according to Norbert).

tp1de, 2020-11-23:
Is this what you mean for the telegram?

[image]

proddy, 2020-11-23:
yup, thats it. We're picking up the pump status from the 3rd byte 00 02 **00** 01 3A...
which looks incorrect.

tp1de, 2020-11-23:
In general that is the pump status but just opposite way around: off means on and vice versa

tp1de, 2020-11-23:
Just loaded new binary ....b5. Pump status is now ok

closed 2020-11-23 by tp1de