emsesp/EMS-ESP #630: Junkers Cerapur + CW400 unable to start one time heating

jakubjakubik, 2020-11-23:
**Bug description**
One time loading doesn't work as expected. While the EMS-ESP accepts the boiler command to start one time loading, and the boiler one time loading status changes to enabled (and back to off when the opposite command is sent) this doesn't affect the state of the heating set temperatures.

**Steps to reproduce**
Send the boiler wwonetime command on Junkers Cerapur with CW400

**Expected behavior**
The one time loading of warm water starts.

**Device information**
`{
  "System": {
    "version": "2.1.0",
    "uptime": "000+00:57:05.632",
    "freemem": 42,
    "fragmem": 47
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 3,
    "mqtt_qos": 1,
    "mqtt_retain": "off",
    "tx_mode": 2,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 2680,
    "#read requests sent": 744,
    "#write requests sent": 5,
    "#incomplete telegrams": 0,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08, ProductID:123, Version:06.03)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05)",
      "handlers": "0xA3 0x06 0xA2 0x2A5 0x2B9 0x2AF 0x2A6 0x2BA 0x2B0 0x2A7 0x2BB 0x2B1 0x2A8 0x2BC 0x2B2 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Controller",
      "name": "Controller (DeviceID:0x09, ProductID:152, Version:03.01)",
      "handlers": ""
    }
  ]
}`

**Additional context**
When I manually start (as in on the control panel) the CW400 one time loading program, the 'Warm water one time charging' does not change state to enabled - even though the heating starts.
I suspect that the boiler setting of wwonetime is ignored and the thermostat controls the whole one time loading cycle by adjusting the temperatures accordingly.
The api documentation lists that there should be an option to do the wwonetime command on the thermostat as well, but I can't call it nor find bindings to it in the code.

jakubjakubik, 2020-11-23:
Browsing a bit more it seems that 

https://github.com/proddy/EMS-ESP/commit/8c62977ad3b71253e68726a527f59460c0ecc6a4 

actually adds this command to the thermostat.

@MichaelDvP are my suspicions correct, and this implementation solves the problem?

MichaelDvP, 2020-11-24:
No, this was only a step, not the solution. Read #589

jakubjakubik, 2020-11-24:
Closing this as duplicate of #589 and will follow up there

closed 2020-11-24 by jakubjakubik