emsesp/EMS-ESP #633: Extension of boiler data with usage data

edejongvandenbrand, 2020-11-25:
Attached is an Excel file showing the information menu structure off an HMS300 and the values at that time and mapping to the values from telegrams 0x494 and 0x495. The data is for a boiler identified as "Enviline/Compress 6000AW/Hybrid 7000iAW". 

I have added the mapping in a local fork of release 2.1.0 from which i have attached the changed boiler and locale files. Time and energy (kWh) values are read as a 4byte/32bit value and all are added to the boiler_data topic. Telegram 0x495 contains additional values at indexes 1-3 and 22-23 of which i cannot determine what they represent. I choose to output the time in seconds instead of minutes as they are supplied and handling translation in the Domoticz plugin.

Enhancement request is to add the data to the baseline with any changes as required for consistency.

[HMC300.xlsx](https://github.com/proddy/EMS-ESP/files/5595444/HMC300.xlsx)
[boiler.cpp.txt](https://github.com/proddy/EMS-ESP/files/5595449/boiler.cpp.txt)
[boiler.h.txt](https://github.com/proddy/EMS-ESP/files/5595450/boiler.h.txt)
[locale_EN.h.txt](https://github.com/proddy/EMS-ESP/files/5595452/locale_EN.h.txt)

viimD, 2020-11-25:
wow, great find! In my Nefit heatpump app i can see the energy usage (only in a very not useful graph), so the figures must be there indeed!

MichaelDvP, 2020-11-25:
I've merged it to my local dev. Needs some smaller changes to fit to the dev branch and some corrections, but compiles now. 
@proddy should i push?
@replayed can you provide some raw telegrams for the test-routines.

proddy, 2020-11-25:
@MichaelDvP could you wait a day. I have about 30+ changes waiting to push

replayed, 2020-11-26:
@MichaelDvP Attached are two files for raw data for the 494 and 495 telegrams. Let me know if you need additional data.

[ems-esp - raw data telegram 0x495.txt](https://github.com/proddy/EMS-ESP/files/5601485/ems-esp.-.raw.data.telegram.0x495.txt)
[ems-esp - raw data telegram 0x494.txt](https://github.com/proddy/EMS-ESP/files/5601486/ems-esp.-.raw.data.telegram.0x494.txt)

proddy, 2020-11-26:
> @MichaelDvP could you wait a day. I have about 30+ changes waiting to push

I did my push, probably broke lots of things but we'll see. @MichaelDvP over to you....

MichaelDvP, 2020-11-27:
Did not find broken things till now. Works good.
Trying to merge the boiler extension now (needs some manual editing). 
Not sure how to format the times, see #632

MichaelDvP, 2020-11-27:
@replayed , @viimD 
Can you please check the new dev build. My boiler does not have these values.

replayed, 2020-11-27:
@MichaelDvP 
Just uploaded the b6 development build. It's running now and all values are being output correctly. I will have to let it run a while to check changes in more values.

Probably unrelated but the Dashboard / Devices & Sensors pages does not show latest values anymore. Only separator lines are visible (see image below).

[image]

MichaelDvP, 2020-11-27:
Maybe confusion with cached b5 page? Try reload the webpage. 
Have you flashed the bin from [https://github.com/proddy/EMS-ESP/releases/tag/dev](https://github.com/proddy/EMS-ESP/releases/tag/dev) or compiled self?

replayed, 2020-11-27:
Correct, a forced page update fixed the issue.

And i used the compiled bin from https://github.com/proddy/EMS-ESP/releases/tag/dev.

viimD, 2020-11-28:
@MichaelDvP I just checked the values that are shown in EMS-ESP and compared them to the values I see on the boiler: exactly the same!!!

replayed, 2020-11-30:
@MichaelDvP 
After monitoring for a few days all values are consistently being handled correctly. Small issue is that all uptime source values are in seconds and the webpage shows the values interpreted as minutes.

MichaelDvP, 2020-11-30:
@replayed keep the unit minutes and show value/60, is this ok, or display the values with unit seconds?

replayed, 2020-11-30:
@MichaelDvP
Indeed keep the unit in minutes and show value/60.

swa72, 2021-03-06:
Is the usage data feature already part of a firmware package? My ESP says `esp32/v3.2.3-14-gd3e562907`, HA shows only shows zeros and I have no clue how to find the info in github.

proddy, 2021-03-06:
@swa72 some of it is there. Check with the console using the `show` command. Top line is the version of EMS-ESP.

swa72, 2021-03-06:
Yupp, After replacing the ESP8266 with the ESP32 I'm on `EMS-ESP version 3.0.0b1` and the device just rocks :-)

glitter-ball, 2021-03-06:
That’s good news. Bought an ESP32 to put in my gateway - maybe this week. Just need some time to solder the headers on and flash it.

swa72, 2021-03-08:
I've seen the solar energy data and some operating times. 

Would be cool if users could read out energy consumption of the burner/warm-water (the same data as the RC300 shows, if you tap on "Info" and scroll down to last entry "usage data" or something. It lists you the energy consumed for the boiler (gas, electricity), warm water (gas, electricity) among others for the last 24h and the last 30d. Plotting that data would be really cool :-)

proddy, 2021-03-08:
> I've seen the solar energy data and some operating times.
>
> Would be cool if users could read out energy consumption of the burner/warm-water (the same data as the RC300 shows, if you tap on "Info" [...]

that would be interesting. There's been some talk about this before (#91) and also quite a bit in the old Gitter chat room. In theory if the RC300 can do this, so can EMS-ESP. It'll be a nice project. I only have a old RC20 so can't help much!

swa72, 2021-03-08:
I‘m wiling to beta test.  Question is, does the RC300 receive the energy consumption data or does it calculate it given other data? And if so, which?

swa72, 2021-05-15:
I've had a chat with a buderus guy on the the phone the other day: usage data in RC300 is guesstimated - so no reliable info. In the meantime, I've setup an ESP32 device to monitor my gas meter (https://github.com/swa72/home-assistant/blob/main/esphome/README-gas.md).