emsesp/EMS-ESP #635: Switch to another schedule-program on CW400 thermostat

realthk, 2020-11-28:
As mentioned in another issue, it would be very useful to be able to switch the CW400 thermostat (and perhaps some other types) to another schedule program with MQTT message sent from Home Assistant.

The reason is because weather compensated mode on the CW400 keeps the pump running as it requires a constant water-flow. Even lowering the setpoint only means lowering the required flow temp, but it does not stop the pump ever.
In the so-called eco-mode though it stops the pump if the room temp is high enough (I think 1C above the setback temp), but this "eco-mode" cannot be selected manually on the thermostat, only by a schedule program. So for example when leaving the house, selecting another program (with a schedule that has "eco-mode" for the whole day) could be a solution. But there is no dedicated key for this on the CW400, so it would be much better to select it by a HA automation.

I've captured some data with `watch on 10` when switching between modes, hopefully these have some info about the programs selected.

This was when switching from "Prog. 1" to "Prog. 2":
```
003+08:13:58.854 N 1: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xE7), data: 00 02 00
003+08:13:58.888 N 2: [emsesp] Thermostat(0x10) -> All(0x00), RC300Floordry(0x267), data: 00 00
003+08:13:59.079 N 3: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2E0), data: 01 1E 64 00 01
003+08:13:59.111 N 4: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2EA), data: 00
003+08:13:59.149 N 5: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 05 07
003+08:14:12.016 N 6: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode(0x2F5), data: 01 00 02 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01
003+08:14:12.272 N 7: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300OutdoorTemp(0x23A), data: 00 0A
003+08:14:12.797 N 8: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Settings(0x240), data: 32 00 05 00 02 00 00 00 FF 03 F2 06 FF 00 00 00 00 00 00
003+08:14:18.629 N 9: [emsesp] Thermostat(0x10) -> All(0x00), RC300Set(0x2B9), data: 00 (offset 18)
003+08:14:19.030 N 10: [emsesp] Thermostat(0x10) -> All(0x00), RC300Set(0x2B9), data: 02 (offset 11)
003+08:14:20.498 N 11: [emsesp] Thermostat(0x10) -> All(0x00), RC300Set(0x2B9), data: FF (offset 8)
003+08:14:26.111 N 12: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0B 13 1C 1B 3B 05 00 10 FF 00
003+08:14:29.172 N 13: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 D4 02 B0 (offset 13)
```

And this was switching back to "Prog. 1":
```
003+08:15:02.741 N 14: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xE7), data: 00 02 00
003+08:15:02.775 N 15: [emsesp] Thermostat(0x10) -> All(0x00), RC300Floordry(0x267), data: 00 00
003+08:15:03.011 N 16: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 E4 21 2B 1E 00 2B 25 00 D4 03 03 01 00 D4 02 B0 00 00 11 01 03 08 E8 00
003+08:15:03.438 N 17: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 06 04 FF CE 00 00 FF 64 37 00 33 04 FF 01 02 (offset 25)
003+08:15:03.632 N 18: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2E0), data: 01 1E 64 00 01
003+08:15:03.664 N 19: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2EA), data: 00
003+08:15:03.694 N 20: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 05 07
003+08:15:11.878 N 21: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode(0x2F5), data: 01 00 02 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01
003+08:15:12.011 N 22: [emsesp] Thermostat(0x10) -> All(0x00), RC300Set(0x2B9), data: 00 (offset 18)
003+08:15:12.318 N 23: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300OutdoorTemp(0x23A), data: 00 0A
003+08:15:12.611 N 24: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Settings(0x240), data: 32 00 05 00 02 00 00 00 FF 03 F2 06 FF 00 00 00 00 00 00
003+08:15:12.741 N 25: [emsesp] Thermostat(0x10) -> All(0x00), RC300Set(0x2B9), data: 01 (offset 11)
003+08:15:14.249 N 26: [emsesp] Thermostat(0x10) -> All(0x00), RC300Set(0x2B9), data: FF (offset 8)
003+08:15:26.495 N 27: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0B 13 1C 1D 00 05 00 10 FF 00
003+08:15:30.220 N 28: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 D3 02 B1 (offset 13)
```

realthk, 2020-11-29:
Another solution would be to switch the thermostat to room controlled mode instead of weather compensated.

This was captured during a switch to room controlled:
```
003+21:11:53.931 N 1: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xE7), data: 00 02 00
003+21:11:53.965 N 2: [emsesp] Thermostat(0x10) -> All(0x00), RC300Floordry(0x267), data: 00 00
003+21:11:54.157 N 3: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2E0), data: 01 29 64 00 01
003+21:11:54.188 N 4: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2EA), data: 00
003+21:11:54.218 N 5: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 05 07
003+21:12:11.521 N 6: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF6), data: FF 01 A5 02 2F
003+21:12:12.161 N 7: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 A5 07 (offset 3)
003+21:12:12.561 N 8: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 A5 07 (offset 3)
003+21:12:12.751 N 9: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 AF 00
003+21:12:12.952 N 10: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 C3 FF FF FF FF FF FF FF FF FF FF FF
003+21:12:14.195 N 11: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 03 49 FF FF FF FF FF FF FF FF FF FF FF
003+21:12:14.404 N 12: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 05 83 00 00 00 00 00 00 00 00 00 00 00
003+21:12:14.877 N 13: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 05 8D 00 00 00 00 00 00 00 00 00 00 00
003+21:12:15.063 N 14: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 AF 00
003+21:12:15.521 N 15: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 A5 9F (offset 4)
003+21:12:15.705 N 16: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 9B 3B
003+21:12:15.890 N 17: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 03 21 0C
003+21:12:16.321 N 18: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2E0), data: 01 00 00 00 01
003+21:12:16.353 N 19: [emsesp] Thermostat(0x10) -> All(0x00), RC300Curves(0x29B), data: 04
003+21:12:16.550 N 20: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 07 00 00 00 00 FF 00 (offset 26)
003+21:12:17.069 N 21: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode(0x2F5), data: 01 00 02 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01
003+21:12:17.451 N 22: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 D8 21 2B 00
003+21:12:17.635 N 23: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 (offset 39)
003+21:12:18.135 N 24: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300OutdoorTemp(0x23A), data: 00 07
003+21:12:18.427 N 25: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Settings(0x240), data: 32 00 05 00 02 00 00 00 FF 03 F2 06 FF 00 00 00 00 00 00
```

And this was switching back to weather compensated mode:
```
003+21:12:29.061 N 26: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 03 6A 00 1A (offset 13)
003+21:12:34.636 N 27: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0B 08 1D 1A 09 06 00 10 FF 00
003+21:12:52.969 N 28: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 AF D5
003+21:12:53.154 N 29: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 AF D5
003+21:12:53.338 N 30: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 A5 1F (offset 4)
003+21:12:54.755 N 31: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 C3 FF FF FF FF FF FF FF FF FF FF FF
003+21:12:54.964 N 32: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 03 49 FF FF FF FF FF FF FF FF FF FF FF
003+21:12:55.389 N 33: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 05 83 00 00 00 00 00 00 00 00 00 00 00
003+21:12:55.598 N 34: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 05 8D 00 00 00 00 00 00 00 00 00 00 00
003+21:12:56.001 N 35: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 03 21 00
003+21:12:56.184 N 36: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 A5 7F (offset 3)
003+21:12:56.369 N 37: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 9B 33
003+21:12:56.785 N 38: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2E0), data: 01 28 64 00 01
003+21:12:56.820 N 39: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF6), data: FF 01 A5 02 3F
003+21:12:57.008 N 40: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 04 (offset 26)
003+21:12:58.037 N 41: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: FF FB (offset 27)
003+21:12:58.653 N 42: [emsesp] Thermostat(0x10) -> All(0x00), RC300Curves(0x29B), data: 01
003+21:12:58.837 N 43: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 (offset 4)
003+21:12:59.021 N 44: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 64 (offset 32)
003+21:13:01.635 N 45: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xE7), data: 00 02 00
003+21:13:01.670 N 46: [emsesp] Thermostat(0x10) -> All(0x00), RC300Floordry(0x267), data: 00 00
003+21:13:01.906 N 47: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 D8 21 2B 28 00 2B 25 03 6A 03 03 01 03 6A 00 1A 00 00 11 01 03 08 75 00
003+21:13:02.340 N 48: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 06 04 FF FB 00 00 FF 64 37 00 33 05 FF 01 00 (offset 25)
003+21:13:02.535 N 49: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2E0), data: 01 28 64 00 01
[... cut ...]
003+21:13:35.252 N 56: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0B 08 1D 1B 09 06 00 10 FF 00
003+21:14:02.018 N 57: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xE7), data: 00 02 00
003+21:14:02.044 N 58: [emsesp] Thermostat(0x10) -> All(0x00), RC300Floordry(0x267), data: 00 00
003+21:14:02.236 N 59: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2E0), data: 01 28 64 00 01
003+21:14:02.267 N 60: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2EA), data: 00
003+21:14:02.306 N 61: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 00 00 05 07
```

MichaelDvP, 2020-11-30:
Selecting the program is telegram 0x2B9 Thermostat(0x10) -> All(0x00), RC300Set(0x2B9), data: 01 (offset 11), also listed from Norbert. Do you have only program 1/2 or other settings? (RC35/20 have some predefined settings like "family, single, ..." counting 1..9).
The weather/roomcontrolled modes are in 0x29B, according to Norbert:
```
HC control type (current heating circuit)
- 1=Optimized
- 2=Simplified
- 3=MPC
- 4=Room Flowtemperature
- 5=Room Power
- 6=Constant
```

realthk, 2020-11-30:
Yes, it is only program 1 and 2 in the CW400 (though both can be renamed)

But does it mean the same could be sent as a command from EMS ESP also?
Because now 0x10->0x00 is just a status broadcast from the thermostat to let the others learn about this change, but does it also mean we could send the same message from the ESP addressed as 0x0B->0x10 and the thermostat will listen and execute it? That would be great!

I'm very new to EMS and still trying to understand the exact nature of telegrams.
For example if `Thermostat(0x10) -> All(0x00), RC300Curves(0x29B), data: 01` was the message about switching to "optimized mode", that is referred in Norbert's doc as `01 9B` and if I now do a raw capture `Rx: 90 00 FF 00 01 9B 01 32` is the result.
So it's not clear for me, how it becomes 0x**2**9B in the non-raw capture, and why is the sender 0x90 in raw instead of the 0x10 thermostat?

MichaelDvP, 2020-12-01:
> But does it mean the same could be sent as a command from EMS ESP also?

Mostly, but some paramters are read only. We have to try.
> So it's not clear for me, how it becomes 0x29B in the cooked capture, and why the sender address is 0x90 in raw instead of the 0x10 thermostat?

For the internal use proddy adds 0x0100 to ems+ telegrams to decide if it should send as ems or ems+.  If you use raw telegrams dont add, if you use commands like read, watch you have to use the addition. i.e. a raw `call system send "9B 90 FF 00 20 01 9B"` is the same as `read 10 29B`. Sender address depends on brand: Buderus uses msb: 0, Junkers msb: 1. I think it's only to keep the brands complete incompatible (and do not ran in issues if not fully compatible).

realthk, 2020-12-01:
Thanks, makes sense!

So can I try a send command like that to find out if this mode can be written on CW400?
For start only the control type, that seems to be simpler (the program type change seems to generate 3 01B9 messages, while the control mode only a single 019B with either 01 or 04 as data)

MichaelDvP, 2020-12-01:
Also the program is only a single command. 
Switch weather based: `call system send "9B 10 FF 00 01 9B 01"`
Switch roomcontroled: `call system send "9B 10 FF 00 01 9B 04"`
Select program 1:  `call system send "9B 10 FF 0B 01 B9 01"`
Select program 2:  `call system send "9B 10 FF 0B 01 B9 02"`

realthk, 2020-12-01:
Thanks, it works!

With a slight difference, that my EMS ESP is listed as 0x0B, so the commands are:
Switch to weather based: `call system send "0B 10 FF 00 01 9B 01"`
Switch to roomcontrolled: `call system send "0B 10 FF 00 01 9B 04"`
Select program 1: `call system send "0B 10 FF 0B 01 B9 01"`
Select program 2: `call system send "0B 10 FF 0B 01 B9 02"`

The CW400 reacts to all of these commands, and changes its state accordingly.

So for now, I'll create HA automations to sends the MQTT equivalent of these, and that will solve my problem. If in the future, there might be a native support for selecting these with simple commands, all the better, but that's not important (perhaps an MQTT report of the current thermostat control-mode might be useful)

MichaelDvP, 2020-12-01:
I like to say the `9B` was a test (for Junkers it has to be `8B` and Buderus `0B`), but is was only a typo. Also ems-esp sets the msb by itself if required by the system, so `0B` works always.

I'll add the commands and values, but you can also use the mqtt-system to send raw commands and chaeck the mqtt response.

MichaelDvP, 2020-12-04:
@realthk please check the new dev build and if all works as expected close this issue.
published as `program` and set by `call thermostat program` (or mqtt, api)  with number and optional hc (see wiki).

realthk, 2020-12-05:
Thanks, I tried, but unfortunately, v2.1.1b6 does not work at all on my config: it finds no EMS connection, and if I leave it long enough connected to the system, even breaks the connection between the thermostat and the boiler, as there came a "system failure" message on the CW400 display. Had to reboot the boiler without EMS-ESP.
I've tried it flashed to 2 different Wemos Minis (suspecting an ESP problem first) with the same result.
After putting back the v2.1.1b4, everything is fine again, without any problem (not even TX/RX errors for weeks)

MichaelDvP, 2020-12-06:
I can not reproduce that with the percompiled bin (8266 and esp32). Maybe your file is corrupted on download? Try to download again.

realthk, 2020-12-06:
Sorry for the false alarm, the bin file is fine! Though still do not understand what might have caused it: I did a fresh build from unmodified source (git cloning the repo and checking out dev branch), the same way as with 2.1.1b4 few weeks ago...
No errors during build, only some warnings like `"Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"}"` (and before that, `pio update` found that `platformio/contrib-pysite` is "Incompatible 2.39.201019") and the process finished as success.

The downloaded v2.1.1b6 bin from here works fine, will test it now.

realthk, 2020-12-06:
Well, most of it is perfect, 
controlmode command works fine either on the EMS ESP with 
```
call thermostat controlmode outdoor
call thermostat controlmode room
```
or sending mqtt to` ems-esp/thermostat` with payload `'{cmd:"controlmode", "data": "outdoor"}'`

Also, changing to program 1 with `call thermostat program 1` or mqtt payload` '{cmd:"program", "data": 1}'` changes the program fine, but changing to program 2 with `call thermostat program 2` doesn't seem to send the command.

I started a `watch on 10` and there was no traffic after program change to 2, only after program 1:
```
000+02:33:51.420 I 71: [thermostat] Setting program to 2 for heating circuit 1
000+02:34:00.582 I 72: [thermostat] Setting program to 1 for heating circuit 1
000+02:34:01.466 N 73: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 D7 21 2B 00 00 2B 25 01 44 03 03 01 01 44 02 40 00 00 11 01 03 08 62 00
000+02:34:01.747 N 74: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 06 07 00 00 00 00 FF 00 37 00 2C 02 FF 01 00 (offset 25)
000+02:34:02.038 N 75: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2B 26 25 02 00 FF FF 00 2A 01 E1 20 01 0F 05 00 00 01 25
000+02:34:02.324 N 76: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x2AF), data: 05 28 FC 00 37 32 11 01 19 1E 00
000+02:34:02.600 N 77: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Curves(0x29B), data: 04 01 00 00 FF 01 05 37 37
000+02:34:02.891 N 78: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWmode(0x2F5), data: 01 00 02 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01
000+02:34:03.146 N 79: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300OutdoorTemp(0x23A), data: 00 5C
000+02:34:03.440 N 80: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Settings(0x240), data: 32 00 05 00 02 00 00 00 FF 02 EC 06 FF 00 00 00 00 00 00
000+02:34:03.974 N 81: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 2B 26 25 02 00 FF FF 00 2A 01 E1 20 01 0F 05 00 00 01 25
000+02:34:06.816 N 82: [emsesp] Thermostat(0x10) -> All(0x00), RC300Set(0x2B9), data: 01 (offset 11)
000+02:34:10.080 I 83: [thermostat] Setting program to 2 for heating circuit 1
```

The direct command `call system send "0B 10 FF 0B 01 B9 02"` still works fine of course.

MichaelDvP, 2020-12-06:
Ups, that's a bug, i checked for 0, 1 instead of 1, 2. I'll fix it.

realthk, 2020-12-06:
Ah, [I see](https://github.com/proddy/EMS-ESP/blob/ca27dcb89fb799426b4a1f61261bf2b4943cf0b8/src/devices/thermostat.cpp#L2183) - but with a build that on my machine might result a faulty bin, rather don't compile it again. Never mind, not urgent at all, don't release just because of this.

proddy, 2020-12-17:
@MichaelDvP close this?

closed 2020-12-17 by MichaelDvP