emsesp/EMS-ESP #636: Home Assistant: separate channels current temperature possible?

bartplessers, 2020-11-28:
Hello,

hardware:
- Buderus GB192iT40
- RC310 thermostat
- KM200 gateway
- EMS-ESP  v2.1.0

I receive all data on my MQTT-broker, and in Home Assistant (auto config), I receive "most" entities.
However, in home assistant, I cant' find the "current temperature" from the thermostat as a separte entity.

In Home Assistant, I can see the device"EMS-ESP Thermostat", with a lot of entities, but the "current temperature" is in a combined object "Thermostat hc1":

[image]

Thermostat hc1
- Auto 21 °C
- Currently: 19.7 °C

From the webinterface of the ems module, I can see the current temperature:

```
Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II

Time    12:41:04 27/11/2020
Error code      (0)
Damped outdoor temperature      8.6 °C
Building        medium
Warm water mode own_prog
Warm water high temperature     60
Warm water low temperature      45
(hc 1) Setpoint room temperature        21.0 °C
(hc 1) Current room temperature 19.6 °C
(hc 1) Comfort temperature      21.0 °C
(hc 1) Eco temperature  17.0 °C
(hc 1) Manual temperature       21.0 °C
(hc 1) Target flow temperature  61 °C
(hc 1) Summer temperature       17 °C
(hc 1) Summer mode      auto °C
(hc 1) Mode     auto
(hc 1) Mode type        eco
```
[image]

So my question is::
Is it possible to show the current temperature in a lovelace sensor card?
Right now it's not exposed as separte entity, so I can not use it in a card.

kind regards,
Bart Plessers

glitter-ball, 2020-11-29:
@proddy  has plans to duplicate the values in the climate entity as sensor values in the MQTT auto-discover payload, For now, I have the current and set temperatures in a separate glance card using this in sensor.yaml:
```
- platform: mqtt
  state_topic: 'ems-esp/thermostat_data'
  name: 'Room Temp'
  unit_of_measurement: '°C'
  icon: 'mdi:home-import-outline'
  value_template: "{{ value_json.hc1.currtemp }}"

- platform: mqtt
  state_topic: 'ems-esp/thermostat_data'
  name: 'Set Temp'
  unit_of_measurement: '°C'
  icon: 'mdi:car-cruise-control'
  value_template: "{{ value_json.hc1.seltemp }}" 
```

proddy, 2020-11-29:
it's already done in an earlier build of 2.1.1 on the dev branch - everything is exposed as an entity now. The data is consistent between the console, web and mqtt topics.

glitter-ball, 2020-11-29:
Ok - thanks. I’m still on 2.1.0 and need to update at some stage...

closed 2020-11-30 by proddy