emsesp/EMS-ESP #637: Maintenance Date, Status

TTr0n, 2020-11-28:
In ems data there is a maintenance date (set by date or by hours), wondering that you dont catch it.

Buderus Logamax GB162
1C telegram UBA

Is it possible to get the date in the dashboard?

proddy, 2020-11-28:
do you know where it is in the telegram?

TTr0n, 2020-11-29:
Not exaxctly, but its already in the code !?

I did <watch on 1c> and got:
[emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 00 66 8B 68 00 00 00 00 00 00 00

Then i changed the mainteneance date to today (29.11.2020) and got 
[emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 94 0B 0A 1D 31 08 00 80 00 00 00

The RC35 shows H 8 , but the Dashboard dont show this. I know its no real error, but i expected to see this in the dashboard.

After reset the maintenance, the 80 and the 08 goes to 00   
[emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 94 0B 0A 1D 31 00 00 00 00 00 00

08 means maintenance by date 

I set the date for next maintenance, but no changes in the data. 

i found this:
[image]

MichaelDvP, 2020-11-30:
Do you want to see the settings (telegram 15) as 
```
maintenanceStatus: off/Time/Date
maintenanceTime: 10000 hours (if set)
maintenanceDate: 30.11.2021 (if set)
```
or the message from 1C only if activated? 
And what about changing the settings by extra commands?

TTr0n, 2020-11-30:
Oh sorry, i want to see both.
1C if its ativated and telegram 15 too.

Changing the settings by command is not necessary, but it would be nice. 
I hadn't thought about that yet. 
Maybe changing the date or hours, depending on set.

MichaelDvP, 2020-11-30:
I think from 0x1C we take only the pos 05, the other positions seems to change meaning on other maintenences messages. 
I think pos 05 is the `Hxx` message, H08 for date, H03 for time, and other (H01 .. H18).
 
@proddy is it better (for HA)  always display maintenance message `H00` or skip the message and display it only if there is a message code  > 0?

For the command i'll check the format `call boiler maintenance 20.11.2021`, `call boiler maintenance 6000` or `call boiler maintenance 0` for date/time/off.

proddy, 2020-11-30:
@MichaelDvP don't really mind, it's up to you what makes sense. Just please keep the data between MQTT, Console and Web is consistent.

MichaelDvP, 2020-12-04:
@TTr0n please test the new dev build and check the maintenance. to reset the Hxx messages i use now the command `call boiler maintenance reset` and not an extra reset command.

TTr0n, 2020-12-04:
Where can i find the instructions for the update ?

MichaelDvP, 2020-12-04:
Always the same place: [https://emsesp.github.io/docs/#/Uploading-firmware](https://emsesp.github.io/docs/#/Uploading-firmware)

TTr0n, 2020-12-04:
hmm ... i cant use the webinterface to update ?
EMS-ESP v2.1.0 / system/ upload firmware and then drap&drop ?

TTr0n, 2020-12-04:
"If your Gateway is running on V2.0 or higher, use the web interface to update." <- I did 
After update the MQTT data includes the maintenance informations, but in the webinterface, there is nothing. 
I can see the devices, but there are no details anymore ... 
[image]

MichaelDvP, 2020-12-04:
You have the old webinterface in browser-cache. Reload the page.

TTr0n, 2020-12-04:
After deleting cache, it works!
Tomorrow i`ll test the maintenance things.

TTr0n, 2020-12-05:
Good morning and thx for this update :-)
I tested the reset and it looks like it works, but ...

-the maintenance date is still set for today, just the H08 is gone (we need a command for setting the new date or hours) 
-the maintenanceMessage (H08) disappeared in dashboard and is missing in MQTT telegram. Maybe we need a static maintenanceMessage beeing empty/NULL or zero if there is no active maintenance (H08 or H03) 
-i didnt test a deactivated maintenance. Will the maintenance disappear too or be zero ?

I checked the mode maintenance by hours. 
I can see it in dashboard, but in MQTT telegram its named "maintenance" too, same identifier like by date. Its possible to get 2 different names or maybe we need an info about the mode (no maintenance, by hour, by date) Then we dont need different identifier for hours and date.

MichaelDvP, 2020-12-05:
In first version i used `maintenanceType`, `maintenanceTime`, `maintenanceDate` and `maintenanceMessages`, but this blows the mqtt and we are in critical length. Now i use only the message and `maintenance` with the valid value. The command is also only `maintenance` with date, time, off or reset as value, switching the correct state. This is also in the wiki.
`maintenanceMessage` is only published if there is a code, after clearing the `Hxx` this message is not published. Should we set a value? `0`, or `H00` or `-`, or`""` ?

TTr0n, 2020-12-05:
Ok, thx for updating the wiki.
Maybe building a single mqtt telegram just for maintenance, then the length is no problem ;-)

proddy, 2020-12-05:
In my 2.3 dev branch I split the boiler mqtt into `boiler_data`, `boiler_data_ww` and `boiler_data_info` to keep the payload sizes down. The maintenance values are in the `boiler_data_info`, but it's easy to change (just a single line).

TTr0n, 2020-12-05:
I tried to set the maintenance date with MQTT to Topic ems-esp/boiler and data {"cmd":"maintenance","data":"07.12.2020"} , but it didnt work. 
Setting the date with telnet console works as expected.

How i see incoming MQTT data in console ?

MichaelDvP, 2020-12-06:
I can not reproduce, working for me sending from mqtt, confirmation in console:
```
000+00:04:30.278 I 1: [boiler] Setting maintenance to 23.05.2021
000+00:05:11.529 I 2: [boiler] Setting maintenance to 08.12.2020
000+00:05:47.315 I 3: [boiler] Setting maintenance in 6000 hours
000+00:06:07.713 I 4: [boiler] Setting maintenance off
```

> How i see incoming MQTT data in console ?

only error/confirmation from a incomming command, for seeing mqtt you have to compile with `-D EMSESP_DEBUG`

TTr0n, 2020-12-06:
thx for your answer and patience, after several tests and reboots, now its working as expected.
Found a small failure on the broker -.-

MichaelDvP, 2020-12-07:
@TTr0n what about that:
> -the maintenanceMessage (H08) disappeared in dashboard and is missing in MQTT telegram. Maybe we need a static maintenanceMessage beeing empty/NULL or zero if there is no active maintenance (H08 or H03)

The maintenanceMessage in ems-telegram is a null string and publish a json with null is not to publish this field in mqtt. If you need a static publish we have to set a blank space or something other?  Does the missing field cause any problem or can we leave it as is?

TTr0n, 2020-12-07:
Causing any problems? not really. It was a wish, but not really necessary.
If it turns out that it is needed, i can ask again ;-)

closed 2020-12-17 by proddy