emsesp/EMS-ESP #638: Junkers Cerapur Compact ZWB 24 + "call boiler heatingactivated on/off"

levran, 2020-11-29:
I recently bought an EMS  interface board+esp32, I have a Junkers boiler. The connection is successful.
Previously, I controlled the boiler from HomeAssistant standart HVAC component via the EMS interface by closing / opening the relay.

There is a command "call boiler heatingactivated on/off" I wanted to use it to also turn on / off the heating. 
Especially during the day (when no one is at home) and at night there is no need for the boiler to heat and pump water through the pipes.

With firmware latest stable 2.1, heating is on all the time.
I tried v2.1.1b6, the result is the same (issue 620 should fix it, but in my case not work).

[log.txt](https://github.com/proddy/EMS-ESP/files/5612432/log.txt)

proddy, 2020-11-30:
As Michael mentioned in https://github.com/proddy/EMS-ESP/issues/620#issuecomment-735434684 it doesn't work on all boilers. We may need to disable it by default and on enable the feature for those boilers we have verified they work.

For your Junkers you can help us trace the behaviour. See if the telegram Tx command (via console) is changing the value and whether it switches automatically back. If you're new to telegrams and debugging maybe best to wait until someone else can try this.

proddy, 2020-12-01:
@MichaelDvP any thoughts on this?

MichaelDvP, 2020-12-01:
I think it works for all boilers, but is overwritten by a higher priority ems-device like controller or thermostat. 
I can set heatingactivated off and the boiler confirms, but a second later the controller sets to on again. Same for flowtemp, i can set, but the thermostat will revert the setting. The MC10 controller can also set to fixed flowtemp, than the controller also overwrites thermostat settings. 
It's very difficult to find all combinations and allow/disallow settings. Imho it's better to leave the commands in and add something to the wiki to explain why not all commands work as expected.

proddy, 2020-12-01:
ok. agree. we need a page in the documentation.

MichaelDvP, 2020-12-01:
Maybe a __bold__ remark or a paragraph in [https://emsesp.github.io/docs/#/API?id=commands](https://emsesp.github.io/docs/#/API?id=commands) is sufficient.

levran, 2020-12-01:
> For your Junkers you can help us trace the behaviour. See if the telegram Tx command (via console) is changing the value and whether it switches automatically back. If you're new to telegrams and debugging maybe best to wait until someone else can try this.

Please give me examples, I really want this functionality to work.

MichaelDvP, 2020-12-01:
Example:
[image]

First byte in telegram 0x16 is on (0xFF) or. off (0), after setting heatingactivated off the boiler confirms, but some seconds later in N11 the controller (09) sends a command to activate the boiler.

levran, 2020-12-01:
strange, but I do not have the first byte 0 for the heating off command
[image]

MichaelDvP, 2020-12-02:
It seems the boiler does not allow write to this. 
Have you tried to set the flowtemp to zero, that's what my thermostat do to switch heating off. 
`call boiler flowtemp 0` writes to 0x1A, but i've seen somewhere that some boilers need a write to 0x35 offset 3, i.e. `call system send "0B 08 35 03 00"`

levran, 2020-12-02:
I have not tried to use such command (call boiler flowtemp 0).
I tried "call boiler heatingtemp 0" but below 30 the boiler refused to lower the temperature.

Ok, later i`ll try "call boiler flowtemp 0" and "write to 0x35 offset 3, i.e. call system send "0B 08 35 03 00""

MichaelDvP, 2020-12-02:
On my boiler the heatingtemp is given by the rotary knob on MC10, and trying to change it results also in a overwrite by the controller. Also the heatingtemp is only a upper limit for the flowtemp. Setting the flowtemp is refused if it is above the heatingtemp.

levran, 2020-12-02:
> call boiler flowtemp 0

I am really HAPPY!!!
THANK YOU SO MUCH!!!!

call boiler flowtemp 0
[image]

call boiler flowtemp 33
[image]

it turns out that I don't need the command "heatingactivated " to emulate relay.

levran, 2020-12-02:
The question can be closed, all many thanks for the help !!!

levran, 2020-12-02:
Tried to set the temperature (call boiler flowtemp 0) every 3 minutes. 
In principle, then the heating remains inactive. 
Maybe that's the way to do it?

MichaelDvP, 2020-12-03:
With the `watch on` you should see which device resets the flowtemp. 
Also you can watch the commands the controller sends if you disable the heating or reduce the flowtemp on the controller manually.

levran, 2020-12-03:
in my EMS network there is only a boiler and ems-esp. 
It seems to me that the boiler itself returns the flowtemp = heatingtemp by timeout (in my case = 5 min).

Are you thinking of setting flowtemp = 0 every 4 minutes when no heating is needed is a bad idea?

MichaelDvP, 2020-12-03:
I think without controller (device-id 0x09) a boiler never works, where do set the temperature and what shows the picture you've posted?

levran, 2020-12-03:
Before the appearance of the HA, the boiler was operated very simply. It is enough to set the temperature of the coolant and the boiler constantly maintained it.
True, this is not very convenient, since the temperature in the house or on the street is not taken into account at all.

The temperature is set on the boiler itself and the square, which means the heating is always active.
When the heating needs to be turned off, I set the temperature below 30 in the menu and the square disappears.
[image]

MichaelDvP, 2020-12-03:
please post the output of `show devices` and `show`

levran, 2020-12-03:
ems-esp:/$ **show devices**
These EMS devices are currently active:

(1) Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:43.05)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495

(2) Controller: HT3 (DeviceID:0x09, ProductID:95, Version:43.05)

ems-esp:/$ **show**
EMS-ESP version 2.1.1b6

Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:43.05)
  Heating active: off
  Warm water/DHW active: off
  Service code:
  Service code number: 0
  Selected flow temperature: 0 °C
  Burner selected max power: 100 %
  Burner current power: 0 %
  Pump modulation: 0 %
  Current flow temperature: 29.3 °C
  Gas: off
  Pump: off
  Fan: off
  Ignition: off
  Heating activated: on
  Heating temperature setting: 30 °C
  Circuit pump modulation max power: 100 %
  Circuit pump modulation min power: 10 %
  Circuit pump delay time: 10 min
  Burner min period: 5 min
  Burner min power: 0 %
  Burner max power: 100 %
  Temperature hysteresis on: -5 °C
  Temperature hysteresis off: 0 °C
  Burner starts: 112000
  Burner active time: 386 days 5 hrs 48 mins
  Heating active time: 326 days 22 hrs 3 mins
  Boiler total uptime: 1371 days 7 hrs 59 mins
  Warm water selected temperature: 51 °C
  Warm water set temperature: 51 °C
  Warm water disinfection temperature: 75 °C
  Warm water type: flow
  Warm water charging type: 3-way valve
  Warm water circulation pump available: off
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: off
  Warm water current temperature (intern): 25.6 °C
  Warm water current tap water flow: 0.0 l/min
  Warm water storage temperature (intern): 25.3 °C
  Warm water activated: off
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: off
  Warm water active: off
  Warm water heating: off
  Warm water starts: 39179
  Warm water active time: 59 days 7 hrs 44 mins

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:43.05)

closed 2020-12-23 by levran