emsesp/EMS-ESP #639: Add unit of measurement for valve status for home assistant

hvanoch, 2020-11-30:
I notice there is no unit of measurement posted with the valve status. Could you add the "%" here? 
I think that is the reason why home assistant doesn't tread the values as numbers but rather as enums or something.
[image]
[image]
 

https://github.com/proddy/EMS-ESP/blob/main/src/devices/mixer.cpp#L204

proddy, 2020-11-30:
sure, but you also do a PR yourself? ;)

hvanoch, 2020-11-30:
I wanted to give it a go but I noticed it is already changed in dev.
https://github.com/proddy/EMS-ESP/blob/dev/src/devices/mixer.cpp

Guess I will have to wait for a new release. Any idea when that will be?

proddy, 2020-11-30:
well 2.1.1 was supposed to be a bug patch on 2.1, but we got carried away and added so much more stuff. It's really a 2.2 release now. I need to discuss with @MichaelDvP to when its good to go to stable. In the meantime just take the latest dev firmware from the releases page, or build yourself.

hvanoch, 2020-11-30:
I was looking at the wrong place. It wasn't added yet so I made a PR https://github.com/proddy/EMS-ESP/pull/641

proddy, 2020-11-30:
done, thanks!

closed 2020-11-30 by proddy