emsesp/EMS-ESP #640: Esp reboots when under heavy load

hvanoch, 2020-11-30:
Sometimes when I visit the admin page of the esp it keeps hanging. Looking in the network tab of the browser it seems to be hanging on loading some javascript files. 

In mqtt explorer I see at that point also rebooted, as the uptime was set to 0 again.

Is this a known issue? Any ideas on how I could improve it?

The esp is connected through the service jack. Could it be that it draws to much current of the ems bus. More then it can deliver and that causes it to reboot?

proddy, 2020-11-30:
yes, known problem on the ESP8266. It's memory related. I'm plagued with trying to cram all the services into the ESP8266's limited memory. There's typically only about 9K left in heap memory which also suffers from fragmentation due to the string libraries. The code is highly optimized to tackle this, but it's getting almost reaching its limits.

It depends on a few things
 - how many devices are attached, and how many heating circuits
 - having telnet open and web can cause issues
 - having MQTT with Home Assistant will eat up more memory when booting

You won't see these issues with the ESP32. When bbqkees has his esp32 compatible gateway out I can put the esp8266 into maintenance mode.

The latest dev branch (2.1.1) has some improvements in this area.

hvanoch, 2020-11-30:
Thanks a lot for the clarification. I will keep this feedback in mind.

closed 2020-11-30 by hvanoch