emsesp/EMS-ESP #643: feedback on HA integration

proddy, 2020-11-30:
from florin-laurentiu

> Integration on HOMEASSIST:

-          Each time when is performed an update for EMS-ESP firmware, the
only thermostat is appear most of the time with different number of
firmware in HOMEASSIST: if it was last stable release firmware for example
2.1.0 and We go to next superior one or latest one can remain in HOMEASSIST
2.1.0; need to up date or downgrade more time the firmware and finally will
respect the last firmware writed, as for the devices that is discover in
MQTT.

-          Beginning with 2.1.1 firmware, status of EMS-ESP device in MQTT
of HOMEASSISTANT is continuously/all time UNAVAILABLE.

-          In the overview mode page, looking to thermostat associated to
boiler thermostat, nothing is working beginning with superior update of
revision 2.1.0; more of this even in the last firmware which is running
more of less( 2.1.0) setting temperature in mentioned page of
HOMEASSISTANT, will switch night mode to day mode even you declare that
temperature must have same value day and night. More of this , the
temperature that is set at that time in the thermostat will remain seated
on the thermostat -for example RC35; need to set again in Homeassistant
temperature to put a different value there.

proddy, 2020-11-30:
Laurentiu, thanks for your feedback. I've isolated the HA feedback into a separate issue so we can concentrate on addressing the problems. 

A) Can you give an example of what you mean? Each time a new version of EMS-ESP is uploaded, it will overwrite the HA config topics via MQTT. So you should not be seeing double entries.

B) The status is an attribute of the `sensor.ems_esp_status`. It may be have been renamed to `sensor.ems_esp_status2` if you had older builds. Go into HA and simply delete the old entries (Developer Tools->States)

C) I would suggest updating to 2.1.1. Many Home Assistant bugs were fixed.

florin-laurentiu, 2020-11-30:
hi,
 
please find attached file with example 👍 
first half is with firmware reported for thermostat and second half is firmware reported for EMS-ESP .
[image]

proddy, 2020-11-30:
Ah ok, It may be a mash up from earlier experiments. Go to HA, Configuration->Integration->MQTT and delete all the devices starting with EMS-ESP, and then restart EMS-ESP.

florin-laurentiu, 2020-12-01:
hello,
back in pictures with result of my experiences with EMS-ESP:
- first of all, delete in HA all devices starting with EMS-ESP, after restart HA. Install firmware 2.1.1b6 and after restart EMS interface . In HA detect 4  devices for EMS and 109 entities. Restart HA just for test, and remain 109 entities for EMS, 4 devices.
- again, delete in HA all devices starting with EMS-ESP, restart HA; install firmware 2.1.0 , then restart EMS interface and surprize, after is coming 4 devices as before , but 139 entities, including more 21 in thermostat ( RC 35). restart also HA, remain same 139 entities.
- again, delete in HA all devices starting with EMS-ESP, after restart HA. Install firmware 2.1.1b6 and after restart EMS interface, and now  is coming 4 devices and remain 139 entities. In this case , appear a new thermostat hc4 that for all belongs entities ia appear "" unknown"" . restart also HA, not changing nothing.
- problem that is common to the all firmware beginning with 2.1.0 is that as soon as you use HA to set temperature on thermostat  ( RC35 ) , thermostat that is is previous in auto mode and according with setting of clock in Night mode or Day mode, this one -thermostat  - will go in DAY mode and never come back to AUTO Mode and respect clock to go in Day or Night mode; only chance to make thermostat to go in AUTO Mode and respect clock that is set inside is to push manually the manual switch from RC 35 thermostat in my case .
 - also, trying to set temperature for thermostat in HA is "" with not control"" you never know if is done or not -need to go to the boiler to verify if is done or not.
- all another function on thermostat in HA,  as heating and ON/OFF is not working in the firmware from 2.1.0 till last one; sincerely, for a firmware less than specified, i never check functionality or this commands of thermostat.

florin-laurentiu, 2020-12-15:
just in progress to prepare a report about version V2.1.1.b7, that was released 5 days ago, and for curiosity  introduce/write  this version  of firmware released 7 hours ago; result is that in HA can not control anymore the thermostat of boiler/HC1, which also is not 100 % complete in the previous firmware. 
unfortunately, i was not kept version released 5 days ago , so, need to find it to prepare the report of  integration of EMS-ESP into HA .

proddy, 2020-12-16:
I've been trying to understand your observations and got a little confused. Here are my suggestions
 - forget 2.1.0. There have been hundreds of fixes and changes and switching between 2.1.1 and 2.1 is going to cause issues, especially with HA
 - it looks like you're unable to set the temperature. Let's see if its a core problem first. Use the telnet console and see if it works. See https://emsesp.github.io/docs/#/Troubleshooting?id=commands-via-mqtt-are-not-working. Keep logging on so you can see what is happening, in different modes
 - setting a temperature for a thermostat with "no control", I'm not sure what you mean. Do you have a suggestion of what should be shown?
 - If the Heating on/off for the boiler is shown incorrectly, first check the console with the `show` command. Again we need to know if its bug in the core or something in the mqtt or HA configuration.

thanks for your feedback and helping EMS-ESP be awesome. We will fix these issues!

florin-laurentiu, 2020-12-16:
ok, i will return with feedback ASAP as also with more investigation.

laurentiu

florin-laurentiu, 2020-12-19:
hello,

return for the moment with small question:  disable MQTT in EMS-ESP interface, refresh page on HA, and please see the result.
do you think that is normal?  this is happened even after long time after stop the MQTT service in EMS-ESP.   
i m using last candidate firmware.
[image]
[image]
[image]

proddy, 2020-12-19:
no, that does't look normal. The devices in HA should still be there (because they are flagged as 'retain' messages, and kept under homeassistant/ems-esp'. But the actual values should not be showing since the data in the MQTT broker is gone. Perhaps HA or your browser needs a refresh?

florin-laurentiu, 2020-12-19:
hi,
i was doing many refreshes on my HA browser....no improvement, same
situation.
this it was just for info; need to prepare my experience in HA about
thermostat functionality.

florin-laurentiu, 2020-12-31:
running last beta firmware V2.2.1b.0, ad looks everything OK for me.
that means that this topic can be closed.

Thank you ! for the entire team and wish you HAPPY NEW YEAR !

laurentiu

proddy, 2020-12-31:
great thanks Laurentiu.

closed 2020-12-31 by proddy