emsesp/EMS-ESP #644: choosing with current temperature to use on a thermostat (if there are multiple sensors)

proddy, 2020-11-30:
From https://github.com/proddy/EMS-ESP/issues/582#issuecomment-730197722

Have an option to configure which current room temperature sensor to use
 - "none": which will use a value of 0
 - "thermostat": which will use the thermostat currtemp (if it exists, otherwise 0)
 - "outside": using the thermostat supported outdoor/outside/weather sensor (if it exists, otherwise 0)
 - "other": which points to an MQTT topic which has the value, like your sonoff example

this would be a pull-down select setting in the WebUI for the Master Thermostat.

not sure though how to handle the different heating circuits?

_Originally posted by @proddy in https://github.com/proddy/EMS-ESP/issues/582#issuecomment-730217771_

proddy, 2020-11-30:
and Michael's code:

Why not select automatically by checking the valid value in hierarchical order:
```
           if (Mqtt::mqtt_format() == Mqtt::Format::HA) {
                if (Helpers::hasValue(hc->ha_temp)) {
                    dataThermostat["hatemp"] = Helpers::round2((float)hc->ha_temp / 10);
                } else if (Helpers::hasValue(hc->curr_roomTemp)) {
                    dataThermostat["hatemp"] = Helpers::round2((float)hc->curr_roomTemp / curr_temp_divider);
                } else {
                    dataThermostat["hatemp"] = Helpers::round2((float)hc->setpoint_roomTemp / setpoint_temp_divider);
                }
            }
```
and set the ha_temp by 
```
bool Thermostat::set_roomtemp(const char * value, const int8_t id) {
    float f = 0;
    if (!Helpers::value2float(value, f)) {
        LOG_WARNING(F("Set roomtemperature: Invalid value"));
        return false;
    }
    uint8_t                                     hc_num = (id == -1) ? AUTO_HEATING_CIRCUIT : id;
    std::shared_ptr<Thermostat::HeatingCircuit> hc     = heating_circuit(hc_num);
    if (hc == nullptr) {
        return false;
    }
    if (f > 100 || f < 0) {
        hc->ha_temp = EMS_VALUE_SHORT_NOTSET;
    } else {
        hc->ha_temp = (int16_t)(f * 10);
    }
    return true;
}
```

MichaelDvP, 2020-12-01:
Should i add[1]  my code to the actual thermostat together with the controlmode/program enhancements? Then you only have to edit the fallback-routine to your settings later. You also need a 0-fallback in your selection mode.  

You have to add settings for each heating circuit and most people will select roomtemp and complaining if not working or always displaying zero. I think a foolproof solution is not possible. 

([1] it's in my testbuild, so not remove before pushing is correct)

proddy, 2020-12-09:
@MichaelDvP this is in b7 right? So let's do some testing with the users (specially PhillyGilly who reported the original issue)

MichaelDvP, 2020-12-09:
@proddy, It was in [this commit](https://github.com/proddy/EMS-ESP/commit/2ce90a1dc55920ae859c8edd65cc8601987d7ff4), but does not work because hcs without currtemp were skiped, in b7 they are not skiped. 

Yes i hope the users with HA-thermostat and no roomsensor will test, also with spoofing a external temperature with topic `thermostat = {"cmd":"roomtemp",  "data":21.4}` or something like that (This also overwrites a working sensor if you like to manipulate, setting <0 or >100 disables).

proddy, 2020-12-13:
implemented this in the #632 branch. Still experimental.

closed 2021-01-04 by proddy