emsesp/EMS-ESP #645: MQTT publishing does not work if hostname is set

CheeseE, 2020-12-01:
**Bug description**
*If the hostname is changed under Wifi settings->hostname other than ems-esp, then the MQTT client does not publish information to the broker. (nor on ems-esp topic or on the new hostname topic)
The client says connected, but the MQTT broker says: ems-esp_status Offline*

**Steps to reproduce**
*Set the hostname on the wifi settings something other than ems-esp.*

**Expected behavior**
*MQTT messages can be seen on the broker*

proddy, 2020-12-01:
which version of EMS-ESP is this?

CheeseE, 2020-12-01:
It is version 2.1.0

>   "System": {
    "version": "2.1.0",
    "uptime": "000+00:22:42.907",
    "freemem": 53,
    "fragmem": 2
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 2,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 3,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  }

proddy, 2020-12-01:
I tested on the latest dev 2.1.1 and can't reproduce. Changing wifi hostname reconnects to the wifi and mqtt and all topics are prefixed with the new name. I'm not sure what you mean by "ems-esp_status Offline"

bbqkees, 2020-12-01:
Did you do the obvious restart EMS-ESP and restart the MQTT broker?

CheeseE, 2020-12-01:
I works fine with the default hostname(ems-esp), but stops working when changing the hostname(although not changing the ClientId in the mqtt settings).  
So hostname set to boiler and ClientId left unchanged(ems-esp), and on the broker cannot see any messages under topics ems-esp/# or boiler/#.  
Chaging back hostname to ems-esp solved the issue.

proddy, 2020-12-01:
See https://emsesp.github.io/docs/#/MQTT
`All topics are prefixed with the ESP's hostname, which is defaulted to ems-esp.`

if you change the hostname, so do the parent of the MQTT topics.

closed 2020-12-01 by proddy