emsesp/EMS-ESP #646: thermostat damped outdoor temp can have a 0 value

proddy, 2020-12-01:
from merlin79-de

Hi There! I have a problem with the damped outdoor temp on RC35. For the telegram 0xA3 there is a additonal if clause if the dampedoutdoortemp==0 (thermostat.cpp line 1169). In my setup there is only a RC35 present (no RC20) and if the value is 0 (last night) the MQTT is missing the dampedoutdoortemp because EMS_VALUE_INT_NOTSET! In my case this if clause makes no sense?

MichaelDvP, 2020-12-01:
I'll change that to model-flag-specific and check for RC35. I'm working on thermostat now for #635 commands for CW400  (and also for RC35).

proddy, 2020-12-17:
I believe this has been fixed in b7, @MichaelDvP ?

MichaelDvP, 2020-12-17:
Yes, changed to [model-flag](https://github.com/proddy/EMS-ESP/commit/2ce90a1dc55920ae859c8edd65cc8601987d7ff4), should always shown in RC35 and never in RC20.

closed 2020-12-17 by proddy