emsesp/EMS-ESP #647: Why does tx mode = 0 (off) does not show any devices?

tp1de, 2020-12-02:
Since I do have quite some stabilty problems when attaching EMS-ESP (#628 km200 not working) and errors with mm100 I tried tx-mode 0. There are then are telegrams received but they do not seem to be analysed into device information.

So no devices are shown with tx_mode = 0 ! 
By pressing the refresh-button in the WEB-UI 3-4 times WEB-UI hangs up and blue led is flashing on the gateway. 

Devices are only beeing seen when tx_mode is > 0. Why? Aren't the broadcast telegrams be analyzed as well?
If tx_mode is changed afterwards to 0 (off) then device info is beeing kept updated. 

I believe that a pure listening mode (tx_mode = 0) should be available as well.

MichaelDvP, 2020-12-02:
ems-esp register the devices by product-id to identify the type. The product-id is published in the version information telegram 0x02, but this have to be queried and is not regulary broadcasted. 
As long as no version information is received there is no device registered. 
It's possible to identify the device type, i.a. 0x10 for a thermostat, but not if it is RC20, RC35 or RC300 with very different behaviour.
Normally the boiler scans on start the devices and ask for version, if ems-esp is listenig the devices should also be registered. 
You can also start with tx-mode off, than set tx_mode, wait for devices detected and switch tx off again.

Pressing web-refresh always produces a very large payload, If you hit it in high frequency the 8266 get out of memory and crashes, esp32 have more resources.

tp1de, 2020-12-02:
Understood. Thanks for explanations.

tp1de, 2020-12-02:
Even with tx_mode = 0 and restarting from WEB-UI produces errors in the heating system. 
"Minor Problems" are displayed and MM100 led is blinking. Heating circuits are not working anymore. A hard reset of the heating system is necessary.

I believe that by restarting the gateway some signals are send on the EMS-Bus which produce the errors. Coulkd this be the case?

proddy, 2020-12-03:
isn't this a duplicate of #628 ? We know that EMS-ESP and the KM200 gateway don't play nice, and this is most likely because of the startup sequence when we query all the devices. Let's use the other issue to track progress on resolving the issue.

closed 2020-12-03 by proddy