emsesp/EMS-ESP #648: RC310 no selected temperature

proddy, 2020-12-03:
> @bartplessers could you do a http://ems-esp/api?device=thermostat&cmd=info

[image]

_Originally posted by @bartplessers in https://github.com/proddy/EMS-ESP/issues/628#issuecomment-737751531_

proddy, 2020-12-03:
@bartplessers which version of EMS-ESP are you using? Did this work before? Do you think it's related to the KM200? Does it work after a restart? I wonder if it works if the KM200 gateway is disconnected?

bartplessers, 2020-12-03:
@proddy : 
- I'm currently not at home, but will test some things this evening.
- Did this work before? --> 
- software version: EMS-ESP v2.1.0
- hardware: BBQKees Gateway Premium II
- device directly attached and powered with EMS bus
[image]

- change temp in hass sends mqtt message, and are also reflected in Buderus app:

[image]

[image]

But changes are not reflected in Home Assistant.

kind regards,
Bart

tp1de, 2020-12-03:
@proddy 
@bartplessers 

I do have the same EMS-ESP version and a KB192i floorstanding boiler mit integrated km200 interface, RC310 and MM100.
The result from http://ems-esp/api?device=thermostat&cmd=info
{
  "time": "10:29:25 03/12/2020",
  "dampedtemp": 4.8,
  "building": "1",
  "wwmode": "4",
  "hc1": {
    "seltemp": 22,
    "currtemp": 21.2,
    "comforttemp": 22,
    "ecotemp": 18,
    "manualtemp": 22,
    "targetflowtemp": 55,
    "summertemp": 20,
    "summermode": "1",
    "mode": "auto",
    "modetype": "comfort"
  },
  "hc2": {
    "seltemp": 22,
    "currtemp": 21.2,
    "comforttemp": 22,
    "ecotemp": 20,
    "manualtemp": 21,
    "targetflowtemp": 35,
    "summertemp": 18,
    "summermode": "1",
    "mode": "auto",
    "modetype": "comfort"
  }
}

MichaelDvP, 2020-12-03:
It's not a bug, it's `modetype eco`. In this mode the  thermostat set `seltemp` to zero to switch off the heating. 
 Somewhere in the thermostat settings it can be selected how the eco-mode works. I've only tested on RC35 and have
`reducemode`: `off`, `reduce`, `roomhold` , `outdoorhold`, `nofrost`.  Only in setting `reduce` the seltemp never goes to zero.

bartplessers, 2020-12-03:
Hi @MichaelDvP : but why does the api still reports a seltemp different from zero? This is inconsistent for me...

MichaelDvP, 2020-12-04:
@bartplessers you have only posted one api otuput in [https://github.com/proddy/EMS-ESP/issues/648#issue-755988355](https://github.com/proddy/EMS-ESP/issues/648#issue-755988355) with seltemp 0 and modetype eco, that looks consistent to me.
Please note that it's modetype eco with eco-temperature 17, but seltemp 0! The thermostat sets temporary 0 until the eco-temperature is reached. I dont know what the km200 displays, maybe the setpoint of eco, It's up to you to find out.

bartplessers, 2020-12-09:
@MichaelDvP : my mistake indeed. I was looking at the post of @tp1de, where 
"currtemp": 21."

my apologies
regards,
B

proddy, 2020-12-17:
@bartplessers I'm a bit confused. Is there something wrong that needs fixing here?

bartplessers, 2020-12-17:
@proddy : no, not anymore. Altough it woud be more covenient to have a different temperature displayed in the autodiscovered thermostat in homeassistant. Like @MichaelDvP mentioned: if you set room temp to let's say 21C, I want to have that displayed on my (hass)thermostat object, and not the 0C when the boiler goes to ecomode. Maybe I have to make a new feature for this?

proddy, 2020-12-18:
ok @bartplessers . This feature is already implemented in another feature-branch. From the WebUI you can select which room temperature to use for the HA climate control. We're planning to close 2.1.1, rename it to 2.2 and make it the next stable version and then start a 2.3 with this and other features. What's holding me back is time and the new device value registration model which is killing the ESP8266 heap memory.

MichaelDvP, 2020-12-18:
@proddy The new feature is to select the current roomtemperature, but Bart likes to select the setpoint. 
I think there is (as in boiler) a setpoint and a selected temperature in the rc300, I guess the  selected temperature is in 0x2A5 pos 6, for RC35 i'm searching. The setpoint is the real output of thermostat and goes sometimes to zero, the selected is the value depending on mode.

bartplessers, 2020-12-18:
@MichaelDvP , @proddy 

>but Bart likes to select the setpoint.

Indeed. I thing the most natural experience from an end-user perspective is that what I want to see on my lovelace card is 
- temperature that I have set
- temperature that is measured in the room

Long time ago I had the ambition to create a graphic interface for this:
https://github.com/bartplessers/lovelace-scheduler
[image]
This can be done fairly easy (just create a svg graphic file, with some elements and properties and let hass update the values of these properties), but lack of time...

So no I just use the out-of-the-box thermostat lovelace card in home assitant. Good enough, ... if 
(1) room measured temperature
(2) room set temperature
[image]

grtz
B

proddy, 2021-01-09:
So there are multiple setpoint temperatures in the rc300?

MichaelDvP, 2021-01-09:
> So there are multiple setpoint temperatures in the rc300?

Yes, there was a long discussion and some trys a time ago. See comments in `Thermostat::process_RC300Monitor`, now the setpoint is get from position 3 as the optimized setpoint, which can go to zero if heating is stopped (e.g. summermode). There is another one in pos 6 which shows the programmed temperature (heat/eco) without opimization. 
Thats how i understand it from the earlier discussions. 

Other thermostats like RC35 only shows the optimized setpoint, and for my understanding it's the one we want to know to understand what the thermostat is doing. (why it is getting cold).  But i think there are different opinions.

bartplessers, 2021-01-10:
> But i think there are different opinions.

😇
Indeed. I would like to see what I have instructed. When I take a look at the thermostat and see a "0", it looks as if
- there is something wrong, or
- somebody set accidently to 0

It feels more natural to me to see the temp I have set instead of what the system does with it...

tp1de, 2021-01-11:
@bartplessers @MichaelDvP 
I have to admit, that I am confused. The datapoint manualtemp does not exits in the km200 interface for the rc310 thermostat.
Anyhow if manualtemp is changed by http post on ems-esp it changes only the hc1 value (radiators) but only in ems-esp not on the heating system and/or km200.

Anyhow I still have difficulties to aline ems-esp rest-api datapoints to the respective km200 rest-api datapoints.
Within km200 there is a datapoint heatingCircuits.hc1.temporaryRoomSetpoint existing (... and also for hc2).
The value is normally -1 (automatic mode). The room setpoint is then set by automatic time program. 
A value 0 will switch off hc temporarily until next switch pint
Any value between 5-30 will change room setpoint termporarily.

The km200 rest-api datastructure for a hc looks like:

[image]