emsesp/EMS-ESP #649: ESP8266 vs ESP32 with BBQKees Gateway - 1st experiences of differences

tp1de, 2020-12-04:
I am using / testing BBQKees Gateway Premium II since 2 weeks. 
My Heating System: Buderus KB192i Boiler with integrated IP, RC310 thermostat and MM100 mixer.
I tried different versions of EMS-ESP: 2.1 and 2.1.1b5 and b6.

I had quite some stability problems with the gateway attached to my EMS-Bus. I tried different things and finally I bought a new D1 mini board with an ESP32. I want to comment on my first findings.

**Original Wemos D1 Mini with ESP8266:**
- I attached to service jack and/or by cable to EMS-Bus
- Operating by cable connected to ems-out on my mm100 mixer was periodically bringing the mm100 mixer into error state (led blinking). External USB power supply on the Wemos D1 was needed.
- My heating system went regularly into error state due to ems errors (minor defect are shown on the display)
- The ems-esp gateway regularly lost the connection to EMS-Bus and therefore no device info was available anymore
- The mqtt-interface stopped after a while 
- every device scan initiated by web-ui stoped the ems-bus -> Errors shown in my heating system and mm100 led was blinking
- refreshing device info 2-3 times by web-ui crashes the ems-esp software. The system restarted and no ems-bus connection was possible.
- when restarting the heating system them km200 web interface was not working correct anymore #628

**Newly bougth AZDelivery ESP32 D1 Mini NodeMCU board** 
- Pin compatible with old Wemos board but larger and led at different location /mazbe there are better fitting boards then the one I bought on Amazon ...
- I flashed with platform io (usb)
- works on BBQKees Premium II board but board is slightly larger - could not close the plastic case for the moment
- all stability problems seems to be solved but external usb powering needed. Device scan without any problem and no web-ui crashes anymore. 
- I am not using mqtt for the moment, since I use web api out of node-red.
- km200 problem still exists #628 

**The ESP8266 seems to be too limited. I would recommend to go for ESP32.**

MichaelDvP, 2020-12-04:
>but external usb powering needed.

Current consumption is nearly the same as esp8266, Bus-powering works only if there are not too much bus clients, service jack or usb powering is than more stable.

> board is slightly larger - could not close the plastic case for the moment

Two small cutouts from the board and the case closes. Only for the endshield there is a piece from 3d-printer needed.
[image]

tp1de, 2020-12-04:
Thanks for the picture. Yes I was thinking about making cutouts as well .. Not having an endshield is fine for me.
I am happy about the switch to ESP32. I think Kees from BBQKees should consider to make the ESP32 standard for the gateway.

bbqkees, 2020-12-05:
Thanks for the feedback.
You can also remove a bit of plastic from the screw bushings on the inside of the enclosure instead of cutting into that particular ESP board.

For most use cases the ESP8266 is still fine. There are actually very few customers who experience problems.
But of course EMS-ESP is now so much more than it was at the start two years ago and to have enough room for future improvements and features the ESP32 is inevitable.
So yes at some point there will be an ESP32 Gateway but that is still a few board iterations away.
If you bring a product on the market that needs to work reliably everywhere for everyone it's not that simple as buying a vague no-name clone chip from Aliexpress and plug it onto an existing board.

MichaelDvP, 2020-12-06:
@bbqkees: the esp32 D1mini is not from lolin/wemos, it's orginal from chinese [MH-ET Live](https://mhetlive.nodebb.com/topic/8/mh-et-live-minikit-for-esp32), but i don't know if this a sign for constant quality.

tp1de, 2020-12-06:
I baught the gateway some 3-4 weeks before. I do have a Lolin D1 mini V3.1.0. But for stability and speed the ESP32 is the right choice.

tp1de, 2020-12-09:
@bbqkees 
> You can also remove a bit of plastic from the screw bushings on the inside of the enclosure instead of cutting into that particular ESP board.

Yes it works. enclosure can now be closed but blue led is at different position and cannot been seen easyly anymore.
The endshield still needs a change for accessing usb power and maybe seeing the blue led status as well.. 

Looking at the development of the ems-esp gateway software, I believe that the esp32 is a must in the future to secure stable operations. Not using the service jack, usb powering is in most cases a must as well. Better be prepared for both.

proddy, 2021-01-17:
we've incorporated some of this feedback into the ESP32 build.

closed 2021-01-17 by proddy