emsesp/EMS-ESP #650: Include Heating System Return Temperature in Monitoring

tp1de, 2020-12-06:
I miss the folowing information which is available in EMS-messages:

Heating System Return Temperature out of UBAMonitorFast Telegram:
`
08 --> 00 id 0x18:  Position byte 18-19 type analog with divisor 10 unit °C.
`
Since I started some coding-tests and firmware buildup: Is there any chance to implement such changes by myself?
If yes, is there a short documentation how to do it?

MichaelDvP, 2020-12-06:
The sensor is read and published if there is a valid value. Maybe you dont have a sensor? 
I've added a dallas to the return flow, since my boiler also have no sensor. 
Check:
`ems:/# read 8 18
000+00:18:34.660 N 1: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFast(0x18), data: 39 01 D7 64 64 09 01 15 01 01 BC 7D 00 80 00 02 0A FF 2D 48 00 00 FF 00 10 83 00`
pos 18 in telegram is pos 13 in data (counting from 0)  -> `8000`  = no sensor

tp1de, 2020-12-06:
I used return temperatur for 15 months out of the km200 gateway. September Buderus changed gw sw version and return temp disapeared. Sensor should be available.

I made a new watch and I am astonished that message id changed from UBAMonitorFast(0x18) to UBAMonitorFastPlus(0xE4).
I have no idea why, and if data content is still the same:
```
[emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 48 00 C8 38 02 37 3C 27 03 00 00 00 00 00 01 7B 01 8F 11 00 02 37 80 00
[emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 1B 80 00 7F FF 80 00 (offset 27)

```
I took pos 18 out of this document: https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#ubaerrormessages1

MichaelDvP, 2020-12-06:
Yes, in E4 the return temp is unknown. Candidates are 017B = 37,9°C (pos17) and 021B = 53,9°C (pos27). Can you check? Flowtemp is 56,7°C (0237).

tp1de, 2020-12-07:
pos 17 is the return temp (you start counting from 0 - I started from 1)
pos 27 is always approx 1.5 - 2.5 °C lower than current boiler flow

tp1de, 2020-12-07:
Is there any easy way to implement this values into the code by myself?
I do have some other candidates I would like to test.

Or is there any way to access the telegrams like watch on console but from a remote computer? E.g. by web-api?
If yes it would be easy for me to debug by testing from node-red or similar iot environment.

MichaelDvP, 2020-12-07:
> Is there any easy way to implement this values into the code by myself?

add in boiler.cpp around line1040 in `process_UBAMonitorFastPlus`:
```
    telegram->read_value(retTemp_, 17); // can be 0 if no sensor
    if (retTemp_ == 0) {
        retTemp_ = EMS_VALUE_USHORT_NOTSET;
    }
```
(for your privat use the first line is sufficient)
> Or is there any way to access the telegrams like watch on console but from a remote computer? E.g. by web-api?

With web-api no, it can not push the later received value. With mqtt you can use the system-topic with raw commands like `{"cmd":"send","data":"0B 88 E4 11 02"}` and check the topic: `response` for the received data. 

>  (you start counting from 0 - I started from 1)

In ems-esp we are counting the data-offset. In [https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme](https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme) he counts the telegram bytes, our databyte 0 if then 5. Also positions in telegram 0x18 and 0xE4 are completly different.

tp1de, 2020-12-07:
> add in boiler.cpp around line1040 in `process_UBAMonitorFastPlus`:
> 
> ```
>     telegram->read_value(retTemp_, 17); // can be 0 if no sensor
>     if (retTemp_ == 0) {
>         retTemp_ = EMS_VALUE_USHORT_NOTSET;
>     }
> ```

This was easy. Took me 5 minutes to implement. Thanks a lot.

MichaelDvP, 2020-12-08:
@tp1de can you please also check for syspressure in the same telegram?  I think i can be in pos 21 (1,7 bar): If you have a pressure sensor try with:
` changed_ |= telegram->read_value(sysPress_, 21);`

tp1de, 2020-12-08:
@MichaelDvP system preasure works with pos 21

proddy, 2020-12-17:
@MichaelDvP shall we add this and finalize 2.1.1 this week?

MichaelDvP, 2020-12-17:
It's added in [this commit](https://github.com/proddy/EMS-ESP/commit/8c164a5d856afcae33a72aa215ee4fa7bab3f4b7),
and i think you can release. The few beta users have not found new issues last time, now it's time for a larger user-base.

proddy, 2020-12-17:
ok closing this issue then. I'm just trying to tidy things up for the release.

closed 2020-12-17 by proddy

tp1de, 2020-12-17:
@proddy @MichaelDvP 

- For me the esp8266 version does not work stable and after 24-48 hours I get errors on the EMS-Bus and heating displays minor errors on the thermostat display. My mm100 mixer (where I connected the ems-esp to) ist in error mode (led blinking) and ems-bus disconnected. 
- The esp32 seems to work stable for at least 2 weeks until now.

Could this be a memory allocation problem and/or a timing problem on the ems-bus?
Any idea how to debug and find out what's the reason?

proddy, 2020-12-17:
@tp1de create a new GitHub issue for this and I'll add my thoughts on how to find the root cause of the ESP8266 bus errors.

This issue is about monitoring the return temperature.