emsesp/EMS-ESP #651: CRC error an missing ES72

alexvr6, 2020-12-06:
Hello,
my telegrams contain CRC errors. Where could the mistake be?
See example:
ems-esp:/$ [K000+00:28:52.916 E 0: [telegram] Rx: C4 08 00 18 00 23 01 40 63 00 01 01 20 60 80 00 80 00 01 3E 00 00 0E 30 59 00 CC 00 00 00 18 (CRC 18 != 13)                                                                                   
ems-esp:/$ [K000+00:29:32.676 E 1: [telegram] Rx: C4 08 00 19 00 00 12 01 3A 80 00 00 00 00 37 00 9C C4 13 52 E8 00 00 00 13 52 E8 00 9C C4 62 (CRC 62 != 69)                                                                                   
ems-esp:/$ [K000+00:30:32.932 E 2: [telegram] Rx: 19 08 00 19 00 00 12 01 37 80 00 00 00 00 37 00 9C C4 13 52 E8 00 00 00 13 52 E8 00 9C C4 71 (CRC 71 != 02)                                                                                   
ems-esp:/$ [K000+00:31:00.890 E 3: [telegram] Rx: C8 08 00 07 00 0B 03 02 00 00 00 00 00 00 00 00 00 00 B7 (CRC B7 != BA)

My equipment:
[angelroda-wetter.de/tmp/ems-esp.JPG](url)     
Whereby an ES72 is not displayed.
I would be very grateful for any help

MichaelDvP, 2020-12-06:
You have a lot of devices on the bus, that could be critical for the crc-errors. Where do you have ems-esp connected and how is it powered? Try another bus position and powering from usb and see if crc errors are reduced. 

This ES72 thermostat is not in the database for now. It is connected as roomcontroller for hc2 in your system, i think you get mostly 0xAF telegrams from it? Can you do a `watch on 19` in terminal to see what it is sending, for more than a minute.

Can you please also post the output of `call system info` and `show`.

alexvr6, 2020-12-06:
Yes, there are a lot of devices on the bus. Everything works too, as far as I can tell. The ems-esp is connected directly to the ES72 and is supplied with voltage via USB with a power supply unit.
Here is the report from watch on 19:

ems-esp:/$ watch on 19
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x19
000+00:26:27.492 N 53: [emsesp] Unknown(0x19) -> Boiler(0x08), UBAMonitorWW(0x34), data: 07
000+00:26:27.510 N 54: [emsesp] Boiler(0x08) -> Unknown(0x19), UBAMonitorWW(0x34), data: 0A 80 00 80 00 00 00
000+00:26:55.935 N 55: [emsesp] Unknown(0x19) -> All(0x00), ?(0xAF), data: 00 D6 00
000+00:27:01.650 N 56: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 18 02 01 80 00 00 00 00 00 00 9C C6 13 53 0D 00 00 00 13 53 0D 00 9C C6
000+00:27:13.843 N 57: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 00 18 02 01 80 00 00 00 00 00 00 9C C6 13 53 0D 00 00 00 13 53 0D 00 9C C6
000+00:27:56.752 N 58: [emsesp] Unknown(0x19) -> All(0x00), ?(0xAF), data: 00 D6 00
000+00:28:01.434 N 59: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 18 02 00 80 00 00 00 00 00 00 9C C6 13 53 0D 00 00 00 13 53 0D 00 9C C6
000+00:28:13.603 N 60: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 00 17 02 00 80 00 00 00 00 00 00 9C C6 13 53 0D 00 00 00 13 53 0D 00 9C C6

I am not so fit in telnet, call system info doesn't work. The result of the show is here:
ems-esp:/$ watch on
Watching incoming telegrams, displayed in decoded format
000+00:20:13.957 N 0: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 07 02 0F 00 00 00 00 00 60 80 00 80 00 01 45 00 00 12 30 48 00 CB 00 00 00
000+00:20:17.020 N 1: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00
000+00:20:17.673 N 2: [emsesp] Thermostat(0x10) -> All(0x00), RC35Monitor(0x3E), data: 04 00 00 00 D8 00 00 05 05 05 00 00 00 00 05
000+00:20:17.923 N 3: [emsesp] Thermostat(0x10) -> All(0x00), RC35Monitor(0x48), data: 04 00 00 00 D6 64 3C 05 05 05 00 00 00 00 05
000+00:20:18.644 N 4: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBATotalUptime(0x14), data: 03
000+00:20:18.656 N 5: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBATotalUptime(0x14), data: 72 8A CA
000+00:20:21.851 N 6: [emsesp] Switch(0x11) -> Boiler(0x08), ?(0x1E), data: 01 BC
000+00:20:23.556 N 7: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 07 02 0E 00 00 00 00 00 60 80 00 80 00 01 45 00 00 12 30 48 00 CB 00 00 00
000+00:20:25.994 N 8: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 01 02 02 00 00 00 00 00 00 00 00 00
000+00:20:26.218 N 9: [emsesp] Unknown(0x19) -> Boiler(0x08), UBAMonitorWW(0x34), data: 07
000+00:20:26.233 N 10: [emsesp] Boiler(0x08) -> Unknown(0x19), UBAMonitorWW(0x34), data: 0A 80 00 80 00 00 00
000+00:20:28.914 N 11: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 03 02 02 00 00 00 00 00 00 00 00 00
000+00:20:31.913 N 12: [emsesp] Switch(0x11) -> Boiler(0x08), ?(0x1E), data: 01 BC
000+00:20:33.551 N 13: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 07 02 0E 00 00 00 00 00 60 80 00 80 00 01 45 00 00 12 30 48 00 CB 00 00 00
000+00:20:39.661 N 14: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0C 17 06 2B 17 06 00
000+00:20:41.980 N 15: [emsesp] Switch(0x11) -> All(0x00), ?(0x9C), data: 01 BC 00 01 00
000+00:20:42.273 N 16: [emsesp] Switch(0x11) -> Boiler(0x08), ?(0x1E), data: 01 BC
000+00:20:43.624 N 17: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 07 02 0E 00 00 00 00 00 60 80 00 80 00 01 44 00 00 12 30 48 00 CB 00 00 00
000+00:20:50.771 N 18: [emsesp] Thermostat(0x10) -> Mixer(0x21), MMSetMessage(0xAC), data: 05 00 04
000+00:20:51.036 N 19: [emsesp] Unknown(0x19) -> All(0x00), ?(0xAF), data: 00 D6 00
000+00:20:51.836 N 20: [emsesp] Switch(0x11) -> Boiler(0x08), ?(0x1E), data: 01 BB
000+00:20:53.601 N 21: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 07 02 0E 00 00 00 00 00 60 80 00 80 00 01 44 00 00 12 30 48 00 CB 00 00 00
000+00:20:56.152 N 22: [emsesp] Thermostat(0x10) -> All(0x00), RCOutdoorTemp(0xA3), data: 02 00 00
000+00:21:00.525 N 23: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 72 8A CA
000+00:21:01.387 N 24: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 5A 63 00 06 FA 0A 01 05 64 37 02
000+00:21:01.796 N 25: [emsesp] Switch(0x11) -> Boiler(0x08), ?(0x1E), data: 01 BB
000+00:21:02.201 N 26: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 17 02 0B 80 00 00 00 00 00 00 9C C6 13 53 0D 00 00 00 13 53 0D 00 9C C6
000+00:21:02.830 N 27: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 00 00 3C FB 00 0F FF 02 46 00 FF
000+00:21:03.302 N 28: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Monitor(0x3E), data: 04 00 00 00 D8 00 00 05 05 05 00 00 00 00 05
000+00:21:03.619 N 29: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 07 02 0D 00 00 00 00 00 60 80 00 80 00 01 44 00 00 12 30 48 00 CB 00 00 00
000+00:21:04.214 N 30: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Set(0x3D), data: 04 1C 2A 21 05 28 FA 02 00 05 05 2D 01 01 04 3C 05 4B 01 00 3C FF 0E F6 05 00 02
000+00:21:04.792 N 31: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Set(0x3D), data: 00 02 00 01 0F 32 0A (offset 25)
000+00:21:05.251 N 32: [emsesp] Mixer(0x21) -> All(0x00), MMStatusMessage(0xAB), data: 05 00 F0 00 9C 01 10
[... cut ...]
000+00:21:33.689 E 49: [telegram] Rx: C8 08 00 18 00 07 02 0D 00 00 00 00 00 60 80 00 80 00 01 44 00 00 12 30 48 00 CB 00 00 00 2E (CRC 2E != 9D)
000+00:21:40.039 N 50: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 14 0C 17 06 2C 18 06 00
000+00:21:41.551 N 51: [emsesp] Switch(0x11) -> All(0x00), ?(0x9C), data: 01 BA 00 01 00
000+00:21:41.843 N 52: [emsesp] Switch(0x11) -> Boiler(0x08), ?(0x1E), data: 01 BA
000+00:21:43.607 N 53: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 07 02 0D 00 00 00 00 00 60 80 00 80 00 01 44 00 00 12 30 48 00 CB 00 00 00

Thanks for the help.

MichaelDvP, 2020-12-07:
> I am not so fit in telnet, call system info doesn't work. The result of the show is here:

I wanted to know which software version you are using, but it seems to be v2.1.0. 
For call commands you have use `su`, read here: [https://emsesp.github.io/docs/#/Console](https://emsesp.github.io/docs/#/Console)
> The ems-esp is connected directly to the ES72 and is supplied with voltage via USB with a power supply unit.

I think ES72 as roomcontroller have a very long cable to the boiler? Maybe better to connect ems-esp more near the boiler to reduce crc errors. But the few crc-errors do not realy hurt, ems-telegrams are repeated often, you will miss nothing.  Noticeable that the crc-errors are mainly on boiler messages which have not poll in front.

I'll add the ES72 to database, but as remote-roomcontroller you will not get any new information except the name. The ES72 only checks the WW-status(0x34)  from boiler (you have this in boiler parameters) and sends the roomtemperature (0xAF) to the masterthermostat. The master publishes the roomtemperature with the heatingcircuit and you get it from there in thermostat_data.
Also the switch sends his temperature by 0x1E and 0x9C, i can add it to switch parameters, but it is also republished as switchtemp from boiler in the 0x18 telegram. So you have the value already.

TTr0n, 2020-12-07:
Lots of CRC Errors are here too:

000+08:06:57.097 E 0: [telegram] Rx: 98 08 00 34 00 3C 02 6B 02 6B 21 00 01 00 00 02 76 A4 00 1F 7F 00 5F (CRC 5F != AB)
000+08:07:00.999 E 1: [telegram] Rx: 10 00 A3 00 06 11 01 01 2B 01 31 01 89 (CRC 89 != 2D)
000+08:07:20.909 E 2: [telegram] Rx: 80 FF E0 FF E0 80 FF 06 DC (CRC DC != 1D)
000+08:08:55.902 E 3: [telegram] Rx: 10 00 3E 00 04 02 32 7D 00 00 00 (CRC 00 != C1)
000+08:08:56.124 E 4: [telegram] Rx: C0 C0 E0 F0 8C (CRC 8C != 54)
000+08:09:03.805 E 5: [telegram] Rx: 10 0F 3F 00 01 18 00 30 01 4E 00 8A 21 18 20 30 21 4E 20 8A 41 18 8D (CRC 8D != 16)
000+08:09:05.821 E 6: [telegram] Rx: 10 0F A5 00 00 00 00 00 FF F6 01 06 00 01 00 01 00 FF FF 00 00 05 1F 05 28 FF 00 00 (CRC 00 != 7F)
000+08:09:56.990 E 7: [telegram] Rx: 10 00 3E 00 04 02 32 7D 00 00 00 00 00 00 00 00 60 89 (CRC 89 != FD)
000+08:09:57.529 E 8: [telegram] Rx: 10 00 3E 00 04 02 32 7D 00 00 00 00 00 00 00 (CRC 00 != BC)
000+08:09:58.491 E 9: [telegram] Rx: 10 00 3E 00 04 02 32 7D 00 00 00 00 00 00 00 00 44 89 (CRC 89 != D9)
000+08:09:59.403 E 10: [telegram] Rx: 10 00 3E 00 04 02 32 7D 00 00 00 00 00 00 00 (CRC 00 != BC)
000+08:10:00.141 E 11: [telegram] Rx: 10 00 3E 00 04 02 32 7D 00 00 00 00 00 00 00 00 64 11 2C 01 1A 89 (CRC 89 != 37)
000+08:10:02.294 E 12: [telegram] Rx: 10 0F 3E 00 04 02 32 7D 00 00 00 00 00 00 00 00 64 11 2C 01 F8 89 (CRC 89 != E6)
000+08:10:17.827 E 13: [telegram] Rx: 90 08 00 34 00 3C 02 6B 02 6B 21 00 01 00 00 02 76 A4 00 1F 7F 00 5F (CRC 5F != FC)
000+08:10:23.006 E 14: [telegram] Rx: F0 C0 F0 FC 89 (CRC 89 != E1)
000+08:10:47.371 E 15: [telegram] Rx: 10 00 08 00 34 00 3C 02 6B 02 6A 21 00 01 00 00 02 76 A4 00 1F 7F 00 D6 (CRC D6 != 93)
000+08:10:57.353 E 16: [telegram] Rx: 98 08 00 34 00 3C 02 6A 02 6A 21 00 01 00 00 02 76 A4 00 1F 7F 00 C0 (CRC C0 != 34)
000+08:11:00.776 E 17: [telegram] Rx: 10 00 3E 00 04 02 32 7D 00 00 00 00 (CRC 00 != 9B)
000+08:11:01.456 E 18: [telegram] Rx: 10 00 BE 89 (CRC 89 != FE)
000+08:11:01.778 E 19: [telegram] Rx: 10 00 3E 00 04 02 32 7D 00 00 00 00 00 00 00 (CRC 00 != BC)
000+08:11:05.034 E 20: [telegram] Rx: 10 0F 3F 00 01 18 00 30 01 4E 00 8A 21 18 20 30 21 4E 20 0A 89 (CRC 89 != A3)

ems-esp:/$ show devices
These EMS devices are currently active:

(1) Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:03.06)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495

(3) Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.06) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0x3E 0x3D 0x3F 0x48 0x47 0x49 0x52 0x51 0x53 0x5C 0x5B 0x5D 0xA5 0x37

(2) Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)

Firmware: 2.1.1.b6

bbqkees, 2020-12-08:
Are you using a BBQKees EMS interface board or your own circuit?

alexvr6, 2020-12-08:
Thank you very much for your detailed answers. I use the ems level shifter 0.9: https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/blob/master/PCB-files/V0.9/README.md. A problem with me should also be the cables, there are 2 single lines. I will exchange it for data lines.

bbqkees, 2020-12-08:
In general the Level shifter 0.9 gives a clean signal. So I expect indeed a noise issue due to the cabling.
Try to mount the board near the boiler.

MichaelDvP, 2020-12-08:
@alexvr6 the ES72 should now displayed in 2.1.1b7. 
@TTr0n you have not so much devices, is the RC35 as roomcontroller far away from boiler or boiler-mounted? Where is ems-esp connected and how is it powered? 
In general crc errors are caused by
a) bad powering of the esp. => try powering by service-jack or usb
b) overload on the bus if ems-esp is buspowered and there are a lot of other devices
c) low/disturbed signal on the bus lines. The ems is plumber compatible: line/star/mixed topographie, no termination, no polarity, simple cable quality. But this can cause issues:  EMC because of no shielding, reflections because of no termination, etc. This causes lost telegrams (crc-errors) and for that all telegrams are sended repeated and redundant.  To improve bus quality it can help to vary connection point, cable length, cable positions, etc.

TTr0n, 2020-12-08:
Its boiler mounted, maybe 50cm cable!? 
Powered with ems connected, jumper on upper position. service-jack not possible, because my girlfriend cleaned up the livingroom, now missing the cable -.-
I didnt see the usb connector !? How to connect with usb ??
Maybe i`ll try to use a shielded cable tomorrow

MichaelDvP, 2020-12-09:
> After using the USB, there were no more errors

Good.
> but with plugged in USB cable, i cant close the case :(

[scroll down this page](https://bbqkees-electronics.nl/wiki/gateway/connecting-to-the-boiler.html)

TTr0n, 2020-12-12:
I know, but the case ist still on tension with pugged in USB.
Maybe i'm going to sand something down.

proddy, 2020-12-17:
closing this. There are known issues with the bus, power and line interference. There's not much we can do with the software.

closed 2020-12-17 by proddy