emsesp/EMS-ESP #652: How to control my installation

mtvw, 2020-12-08:
**Question**
I want to use the temperatures from my Loxone system to regulate my heating. But I can't find a command that can do that... Looking for a solution or a workaround.

**Screenshots**
[image]

My setup:
- Boiler Junkers CerapurSmart A with heatronic3
- MM200 mixer with 2 heating circuits
- 2 thermostats CW100 (one for each circuit of the mixer)
- an outdoor temperature sensor that is not installed outdoor

**Additional context**
- The 2 thermostats are not in the room they should be heating. I can't get a cable to those rooms to hang the thermostats there. But I have a temperature sensor in those rooms that is managed in Loxone. I can send the temperatures through REST or MQTT.
- The outdoor sensor is not placed outside. same story: can't get a cable there. Loxone fetches the outdoor temperature from some weather API

I don't really need my thermostats or outdoor sensor. If I can bypass them somehow and regulate the mixer directly, that works for me. Or if I can override the temperature sensor of the thermostats, that would be even better.

**I have tried**
- Setting variables on my thermostats through REST or MQTT. I'm able to set stuff like target temps, operating mode, ... But the actual temperatures seem to be read-only
- Setting variables on my mixer. But I can't set anything on this device...

MichaelDvP, 2020-12-09:
Interresting configuration. Does ems-esp work with two CW100, it was a theroretical option and noone has ever checked. Can you please post the full system configuration (`show` in terminal).

Manipulating the sensors was discussed in #558, maybe for the roomtemperatures you can try with changing the calibration of internal sensor, but the command for CW100 is unknow, you have to figure out. 
For outdoor temperature this is not possible. 
Another idea, the sensors are 10k NTC, you can try to replace the sensor by a digipot like [this](https://www.analog.com/en/products/ad5272.html#product-overview) and set it to the desired value. (check electrical parameters). Temperature resolution is not very good because on nonlinearity of NTCs. This may work also on the thermostats, because cw100 can use a external sensor.

The mixer is not writeable because normally the thermostat writes to mixer and will overwite any command. If there is no thermostat i think a write is possible. ems-esp ony reads the mixer status, the set-command have to be figured out.

Replacing the thermostat was discussed in #524, but for now there is no solution.

mtvw, 2020-12-09:
full system configuration:
```
EMS-ESP version 2.1.0

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:23.12)
  Heating active: on
  Warm water/DHW active: off
  Service code: 
  Service code number: 0
  Warm water selected temperature: 60 °C
  Warm water set temperature: 60 °C
  Warm water disinfection temperature: 75 °C
  Selected flow temperature: 44 °C
  Burner selected max power: 76 %
  Burner current power: 51 %
  Pump modulation: 89 %
  Warm water type: flow
  Warm water charging type: 3-way valve
  Warm water circulation pump available: off
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: off
  Outside temperature: 17.2 °C
  Warm water current temperature (intern): 24.3 °C
  Warm water current tap water flow: 0.0 l/min
  Current flow temperature: 46.0 °C
  Mixer switch temperature: 43.2 °C
  Warm water storage temperature (intern): 24.3 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: off
  Warm water active: off
  Gas: on
  Pump: on
  Fan: on
  Ignition: off
  Warm water heating: off
  Heating activated: on
  Heating temperature setting: 50 °C
  Circuit pump modulation max power: 100 %
  Circuit pump modulation min power: 10 %
  Circuit pump delay time: 3 min
  Burner min period: 3 min
  Burner min power: 0 %
  Burner max power: 76 %
  Temperature hysteresis on: -5 °C
  Temperature hysteresis off: 0 °C
  Warm water starts: 20759
  Warm water active time: 29460
  Burner starts: 34343
  Warm Water active time: 20 days 11 hours 0 minutes
  Total burner operating time: 415 days 12 hours 41 minutes
  Total heat operating time: 395 days 1 hours 41 minutes
  Total UBA working time: 1444 days 2 hours 51 minutes

Thermostat: RC200/CW100 (DeviceID:0x19, ProductID:157, Version:41.08)
  Damped outdoor temperature: 17.0 °C
  Warm water mode: auto
  Heating circuit 2:
    Setpoint room temperature: 23.0 °C
    Current room temperature: 21.5 °C
    Target flow temperature: 49 °C
    Mode: manual
    Mode type: day

Thermostat: RC200/CW100 (DeviceID:0x18, ProductID:157, Version:41.05)
  Time: 13:16:26 08/11/2020
  Error code: (0)
  Damped outdoor temperature: 17.0 °C
  Warm water mode: auto
  Heating circuit 1:
    Setpoint room temperature: 24.0 °C
    Current room temperature: 22.2 °C
    Comfort temperature: 21.0 °C
    Eco temperature: 16.0 °C
    Manual temperature: 24.0 °C
    Target flow temperature: 39 °C
    Summer temperature: 17 °C
    Summer mode: auto °C
    Mode: manual
    Mode type: day

Mixer: MM200 (DeviceID:0x20, ProductID:161, Version:29.04)
  Heating circuit 1:
    Current flow temperature: 40.7 °C
    Setpoint flow temperature: 39 °C
    Current pump status: off
    Valve status: 62 %

Mixer: MM200 (DeviceID:0x21, ProductID:161, Version:29.04)
  Heating circuit 2:
    Current flow temperature: 28.0 °C
    Setpoint flow temperature: 30 °C
    Current pump status: off
    Valve status: 46 %
```

The setup with the 2 heating circuits (ground floor/1st floor) is in place because the Junkers approver said it was required...

Manipulating the sensors or soldering some extra components on the thermostats doesn't really feel like a stable/maintainable solution...

I'll try to reverse engineer the commands that the thermostat is sending to the mixer. Ideally I would be able to send those commands straight from Loxone to the mixer.

bbqkees, 2020-12-10:
In my opinion it would still be better if you could somehow drill some holes and add wiring to put the thermostats where they belong, so the system regulates itself fault-free instead of relying on external input from a home automation system that runs on a SD card.

tp1de, 2020-12-10:
Is this a house with just floor heating or do you have radiators as well?
How many seperate rooms / heating circuits are installed and do they do not have seperate servo controllers?
Why 2 mixers - I do not really understand this concept - or is one mixer for the radiator heating circuit ? 

My floorheating is just managed by outside temparture sensor and respective heating curve parameters. 
Not contolled by RC310 temp sensor within living room. This works as well.

PhillyGilly, 2020-12-10:
> In my opinion it would still be better if you could somehow drill some holes and add wiring to put the thermostats where they belong, so the system regulates itself fault-free instead of relying on external input from a home automation system that runs on a SD card.

@bbqkees Absolutely. I've lost a couple of SD cards on my RPi3b. I'm not planning to make the temperature in my house (and by extension my fellow inhabitants' happiness!) dependent on equipment that is not manufactured or tested to a reasonably industrial standard.

mtvw, 2020-12-11:
Thanks for all the feedback!

For the outdoor sensor, there might be a way to get the sensor outdoors. I cannot (for the sake of the happiness of my fellow inhabitants) grind new cables in the wall to put the indoor themostats in the perfect place, although I can improve their location a little bit. I'll look into the wireless thermostats, maybe that's the solution for me!

I was hoping I could bypass this because I already have these temperatures in Loxone, I just wanted a way to push these values to my heating system... Now I need a whole bunch of costly, redundant, vendor-specific devices to execute logic which I can also do on the hardware I already have...

@tp1de It's actually 1 device (MM200) with 2 heating circuits (visualized as 2 misers on EMS). I have just floor heating, no radiators. The installer split it into 2 "large" circuits for some reason. There are 14 "lowest level" circuits, which I don't control individually (other then balancing the flow rates).

Just for the record, some more info about my installation (because I have the feeling there are some misconceptions about Loxone):
The SD-card is only used to store the configuration (and load it when Loxone boots). The moment it fails, I just install my config on a new card and it's up again. When a device dies, I just replace it and Loxone still has the config for that device. I've had a lot more issues when one of my KNX sensors died, or when the controller of my boiler died... But maybe I'm biased because I'm a software guy :)

My install is running for 5 years on the same SD-card without issues.
I'm not using a rpi anywhere btw... I do have a home server which is gathering data out of Loxone, but it's nothing crucial and more of a hobby thing.

tp1de, 2020-12-11:
@mtvw 
Difficult to understand the concept of two independent heating circuts with very different setpoint temperatures (9°C different).
There must a reason. Ask your installer why, if you don't know.

Nevertheless for floorheating you will have one or more heat distributers for the different heating circuits (1-2 per room).
The outbound flow temp per buderus hc (mixer) is set by the respective heating curve parameters - normally depending on outside temp.

On the heat distributers there should be servo motors for each circuit which control the volume flow per room. Each of this servo motors is then linked via a master controller to a temperature sensor in the room (by cable or radio frequency). 
These temperature controls are then completely independent from the rc310 or cw100 thermostats. You will use the cw100 just for parameters and/or time controls of the central boiler.

What do you want to do? Control of temp per room (in Germany this has to be done for all new installations by law) - or setting the buderus flowtemp for all rooms on the heating circuit  per mixer just by the temp of one room measured by cw100?
(by the way there is a "cw100 rf set" existing including rf outside temp sensor)

MichaelDvP, 2020-12-11:
I thought about how Bosch manage it to use a zigbee outdoor sensor (T1RF) instead of the boiler mounted outdoor sensor. The boiler sensor is only published on ems and not writable (btw: for ems+ we do not know the the outdoor temperature, i guess it's pos 4/5 in telegram 0xE5, but  that's not confirmed, maybe someone with E5-telegram and sensor can check). 
I think the RFM200 is a complete thermostat and reads the outdoor-sensor T1RF, ignoring the boiler publish. Since the outdoor temperature is only used by the thermostat, it can decide to use it's own RF or the published boiler-value. But it's not possible to tell a normal cable-thermostat to use another temperature.

@mtvw If you want to build your own thermostat you mainly have to calculate the circuit flow temp by outdoor temperature (heating curve), by room temperature or by both. Than you have to send a command to the mixers and the boiler. The boiler flow temp is usualy ~5 degree more than the mixer-flow temps and there is a boiler command in ems-esp. For the mixers the commands are unknown for now. You have to watch which commands are send from thermostat to mixers (e.g. watch on 20). I saw somewhere in summer the telegrams: `Thermostat(0x10) -> Mixing Module(0x20), (0x2E1), data: 01 00 00 00 01` and `Thermostat(0x10) -> Mixing Module(0x20), (0x2EB), data: 00`. But in summer the heating is off and we have to identify the position of set-flow-temp and pump on/off. Both values are republished by the mixer in telegram 0x2D7 (for hc1), data0 for pump, data5 for set-flowtemp.
If you sent a log from the mixer-watch we can add commands for setting the mixer-flowtemps (they will be overwritten if there is a thermostat in the system). For the first test you can also use the system send command to control the mixers.
Your simple own thermostat have to do:
- read loxone temperatures outdoor and room
- calculate flowtemp for heatingcircuits
- send desired flowtemp to mixers and boiler

tp1de, 2020-12-11:
@mtvw First decide if you would like to control your heating system by outside or room temp.
@MichaelDvP I would not recommend to build own components like sensors. The whole rf set (thermostat / outside sensor / and ems-bus connector) cost around 150€ on the net. Keep it standard for maitenance and stability reasons.

MichaelDvP, 2020-12-11:
@tp1de : I don't recommend it. I only give technical information how the thermostat in general works. I also don't recommend to make hardware modifications to spoof temperatures, but it's possible to do so. It's up to @mtvw to decide what he want to do if he know all possibilities.

tp1de, 2020-12-11:
I do have floor heating and radiators. In some rooms both. The floorheating is reacting very slow on temp changes. The radiators are equipped with homematic thermostats. What I do by my ioBroker home automization system is changing temporarely the "temporary room setpoint" to off (-1°C) to avoid overheating by strong sun radiation. I still do this by rest-api requests to the km200 gateway. This is complicated because has to be encrypted.  Maybe in future by ems-esp. There are pro and cons.

But I do not know always know which km200 datapoint corresponds to the respective ems-esp data point and vice versa.
Is there already an overview existing?  If not I will try to prepare one.

mtvw, 2020-12-11:
Thanks for all the information!

I'll start by moving my outdoor sensor to a suitable place outdoors, and evaluating the impact on my temperatures. Maybe it will be enough to stabilize everything.

I don't need to control the temperature per room on demand. As it's floor heating, it's too slow anyway to have granular control. 

The custom thermostat approach by @MichaelDvP seems like an interesting project for when I have more time :) But I'll start by storing the telegram details for future reference. Thanks for the kickstart!

The ideal system should not only keep track of the current outdoor temperature, but also keep in mind the temperature forecast + cloud coverage. But that's a project for when I have a lot more time...

PhillyGilly, 2020-12-13:
I will be very interested in your thoughts. My primary heating for hot water is solar with my gas boiler there to heat it if there isn't any sunshine. In the summer that's fine, and in the winter I heat water in the early morning for the whole day. However on at least 50% of days there is some sunshine so I don't need to use as much gas to make hot water to meet our needs. So I'm interested in day ahead solar radiation forecasts which is clearly related to day ahead cloud cover forecasts. At the moment I haven't found much discussion about this subject, so keep us posted on your progress.