emsesp/EMS-ESP #653: Source code download for dev tree not working

tp1de, 2020-12-09:
@proddy @MichaelDvP 

The download of source code for dev tree does not work. Instead the download contains latest main version.
I think you both are aware. Is ist difficult to "repair" this issue?

proddy, 2020-12-09:
yes I heard from Michael. Do you mean forking/downloading or the artifacts (zip file)?

tp1de, 2020-12-09:
My issues #628 and #650 are solved in latetest dev version. Since I could not download from proddy's github, I did it from MichaelDvP's githiub directory to review the source code changes we have done together and then compiling a new binary for updating. Are boths directories in sync?

When should I close my issues? Or are you doing it?

tp1de, 2020-12-09:
@proddy 
>yes I heard from Michael. Do you mean forking/downloading or the artifacts (zip file)?

I select dev tree from repository and then the green Code Buttom and dowwload zip.
... I wanted to check if all the changes I had done together with Michael are included in b7.

proddy, 2020-12-09:
@tp1de GitHub doesn't work like that. You can only checkout/download the repository. Then you need to switch to the dev branch using a github CLI call or via the IDE.

tp1de, 2020-12-09:
@proddy OK I learned something. But I have no glue what CLI call and IDE stands for ....

But downloading from the assets on https://github.com/proddy/EMS-ESP/releases/tag/dev should give the code which corresponds to the bin-files. Is this right?

proddy, 2020-12-09:
CLI is Command Line Interface. So if you've unzipped the code, you would do a `git checkout dev`. IDE is the GUI tool you're using to code in, like Visual Studio Code. In VSC you use the icon on the bottom left to switch branches.

The assets should be correct, too. Need to verify that.

proddy, 2020-12-09:
oops, you're right. The assets have the old code from the master/main branch. I'll fix that. Thanks for pointing out.

realthk, 2020-12-10:
BTW, do you have the same config settings for build as in the dev branch?
Because [I had a strange experience with 2.1.1b6](https://github.com/proddy/EMS-ESP/issues/635#issuecomment-739512161): although compiling it from source resulted successful build, it did not work on my ESP, could not connect, and the CW400 thermostat even displayed a system failure msg after a while, which could be cleared only by rebooting the boiler without the ESP connected.
Downloading the precompiled bin worked fine.
I've noticed there was some 50 bytes size difference of the downloaded bin file against mine, so perhaps there were other config settings used for creating it (I did not change anything, used the defaults from the dev branch)

proddy, 2020-12-10:
@realthk the build process uses the same process. You can see the script [here](https://github.com/proddy/EMS-ESP/blob/main/.github/workflows/build_firmware.yml). It just does
```
pip install -U platformio
platformio upgrade
platformio update
platformio run -e esp8266
```
on your local system make sure you have the latest pio and library updates, and also not overriding any settings in a `pio_local.ini` file. I will also do the same and compare binary sizes

realthk, 2020-12-10:
@proddy yes, that's how I did. Now tried again, no change to the dev branch:
```
$ git status
On branch dev
Your branch is up to date with 'origin/dev'.

nothing to commit, working tree clean
```
yet still, my result is some 50 bytes larger:

`-rw-rw-r-- 1 thk thk 972272 dec   10 18:26 build/firmware/EMS-ESP-2_1_1b6-esp8266.bin
`
as opposed to the downloaded one:
`-rw-rw-r-- 1 thk thk 972224 dec    6 14:53 /home/thk/Downloads/EMS-ESP-2_1_1b6-esp8266.bin`

I do not try it on the ESP now, but guess it wouldn't operate correctly, just like it wasn't a few days ago.
Had no such thing with 2.1.1b4 before (did not compare its size back then, but it worked from my build)

There were only these warnings about some Mac platform during build, probably not important:
```
npm install
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.2 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/jest-haste-map/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/watchpack-chokidar2/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/webpack-dev-server/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
```

and these messages after `platformio update` Googling shows incompatible new version of contrib-pysite is quite common now, but perhaps that "detached" expressif8266 platform is not healthy?

```
Updating platformio/contrib-piohome           3.3.1 @ ~3.3.1                     [Up-to-date]
Updating platformio/contrib-pysite            2.38.191020 @ ~2.38.0              [Incompatible 2.39.201019]
Updating platformio/tool-unity                1.20500.200612 @ ~1.20500.0        [Up-to-date]
Updating platformio/tool-scons                4.40001.0 @ ~4.40001.0             [Up-to-date]

Platform Manager
================
Platform espressif8266
--------
Updating espressif8266                        2.6.2                              [Up-to-date]
Updating platformio/toolchain-xtensa          2.40802.200502 @ ~2.40802.191122   [Up-to-date]
Updating framework-arduinoespressif8266       3.20704.0 @ ~3.20704.0             [Up-to-date]
Updating platformio/tool-esptool              1.413.0 @ <2                       [Up-to-date]
Updating platformio/tool-esptoolpy            1.20800.0 @ ~1.20800.0             [Up-to-date]
Updating platformio/tool-mkspiffs             1.200.0 @ ~1.200.0                 [Up-to-date]
Updating platformio/tool-mklittlefs           1.203.200522 @ ~1.203.0            [Up-to-date]

Platform espressif8266
--------
Updating espressif8266                        1.8.0                              [Detached]
Updating platformio/toolchain-xtensa          1.40802.0 @ ~1.40802.0             [Up-to-date]
Updating framework-arduinoespressif8266       2.20402.4 @ ~2.20402.0             [Up-to-date]
Updating platformio/tool-esptool              1.413.0 @ <2                       [Up-to-date]
Updating platformio/tool-mkspiffs             1.200.0 @ ~1.200.0                 [Up-to-date]

Platform espressif8266
--------
Updating espressif8266                        2.2.3                              [Detached]
Updating platformio/toolchain-xtensa          2.100100.200706 @ <3               [Up-to-date]
Updating framework-arduinoespressif8266       2.20502.0 @ ~2.20502.0             [Up-to-date]
Updating platformio/tool-esptool              1.413.0 @ <2                       [Up-to-date]
Updating platformio/tool-esptoolpy            1.20600.0 @ ~1.20600.0             [Up-to-date]
Updating platformio/tool-mkspiffs             1.200.0 @ ~1.200.0                 [Up-to-date]
```

proddy, 2020-12-10:
okay, I'll do the same test tonight and compare. On Ubuntu and OSX. If you're interested the contents of the build can be found in https://github.com/proddy/EMS-ESP/actions?query=workflow%3A%22Build+Firmware%22 which is useful for comparing the output.

proddy, 2020-12-11:
I did a build using the same script on OSX vs Ubuntu and the firmware binaries differ by a few bytes. Not sure why but still investigating.

realthk, 2020-12-11:
THX! I use only Ubuntu 20.04. The size different in itself could be OK, but my wrong bin result (in spite of the successful build) last week makes me worry

MichaelDvP, 2020-12-29:
Still there is something wrong in the github-dev-branch. `EMS-ESP-main.zip` download works fine with v2.2, also the 2.3-refactor, but `EMS-ESP-dev.zip` download still is 2.1.0 from 31.10.2020. Maybe it's a solution and good time to delete the dev-branch on github completly and create new based on 2.2.0.

proddy, 2020-12-29:
The problem is with the github actions that automatically deletes and rebuilds the `dev` release using the script at https://github.com/proddy/EMS-ESP/blob/main/.github/workflows/build_firmware.yml. It's based on some 3rd party plugins and I'm not sure why it's using the old source code tree for the assets.  I think the problem is in https://github.com/softprops/action-gh-release

This problem is not high on my to-do list at the moment (I'm only interested in the binaries), so anyone who wants to take a look I'd appreciate it.

proddy, 2021-01-04:
I think this is fixed. I made the dev the default branch

closed 2021-01-05 by proddy