emsesp/EMS-ESP #655: Adding Filter (Device ID / Telegram ID) for Syslog

tp1de, 2020-12-09:
It would help debugging telegram messages and content if there would be a filter function available in syslog settings (similar to watch on console). 

It should be possible to filter by device id and by telegram id. (both together or just one).

MichaelDvP, 2020-12-09:
I dont think this is very usefull in syslog. In therminal it is, terminal is interactive, reading a few lines, change something, etc. But syslog you set once and look in much later. Then it's better to `grep 0x10 /var/log/debug`  or something like that.

tp1de, 2020-12-09:
Depends on what you want to do. Since I haven't found any other way yet to watch telegrams from remote computers, I installed a syslog-server on my raspberryPi and changed syslog settings to all. I then read syslog with tail in node-red and screen the telegrams. Works fine so far, but syslogfile is extremly large within short notice. To avoid this, a filter would be nice to have since I have'nt found a size based rotation logic in syslog-ng yet. Actually I have 26 unknown telegram types and 39 known ones.

I would like to analyse for example all messages to and from km200 gateway. This works by filtering within node-red as well, but a filter in the settings would be very helpfull.

MichaelDvP, 2020-12-09:
You are free to set notice/info in syslog and set in terminal a watch, this also effects the syslog. With `set timeout 0` you can also disable automatic closing of terminal window (which stops watching).

proddy, 2020-12-17:
this is the method. We should perhaps also document this.

closed 2020-12-17 by proddy