emsesp/EMS-ESP #658: Auwch, what happened! Lost my boiler devices in home assistant....

bartplessers, 2020-12-11:
Hello,

I was playing with my hass instance:

- rpi4
- docker with
  - homeassistant
  - mqtt broker
  - zigbee2mqtt
Don’t know what I messed up but I can’t see my (mqtt)boiler anymore. Device + entities seem “unavailable”.

However
- mqtt-broker up & running
- boiler is sending mqtt messages.
- other instances of hass on other devices, connected to same mqtt broker still see my boiler.

Is it possible to delete the device, but how to add it again? Boiler devices were added by autodiscovery…

No my dashboard looks sadly like…
[image]

remark: the working temp display is from another mqtt sensor, proving that mqtt + hass works...

proddy, 2020-12-11:
when EMS-ESP reboots it will re-create the HA discovery topics. They are also flagged as 'retained' messages so surprised they are gone. Look at the integrations in the HA system screen, under MQTT you'll see the EMS-ESP devices (or not)

bartplessers, 2020-12-11:
yup, a reboot did the trick.
Devices were not available (before reboot), now they are.

cheers
B

proddy, 2021-01-05:
@Neonox31 does this happen after you've upgraded EMS-ESP? It may be because the Device information changes due to the version. I've seen it happen once on my setup.

Neonox31, 2021-01-05:
@proddy unfortunately I don't know because before I didn't have a home-assistant, I just installed it.
On the other hand I had several "cuts" but I haven't yet found out to which event they are related.
I will investigate.

proddy, 2021-01-05:
in the meantime I'll make a change in the code so when any MQTT parameter is set and the Home Assistant format is being used it'll re-publish all the HA device details again.

Neonox31, 2021-01-05:
> in the meantime I'll make a change in the code so when any MQTT parameter is set and the Home Assistant format is being used it'll re-publish all the HA device details again.

Good idea, I think it will solve the problem 👍

proddy, 2021-01-09:
> > in the meantime I'll make a change in the code so when any MQTT parameter is set and the Home Assistant format is being used it'll re-publish all the HA device details again.
> 
> Good idea, I think it will solve the problem 👍

implemented in 2.3. Not sure if I'll back-port to 2.2 just yet

closed 2021-01-09 by proddy