emsesp/EMS-ESP #659: Q: pump modulation control

SpaceTeddy, 2020-12-12:
**Question**
Is it possible to control the value of the heating pump modulation (boiler_data_pumpMod)?

Reason:
If the outside temperature is lower then the set "frost" temperature, the boiler set the pump speed to 100% to secure the heating system. This may generate a lot of water flowing noise. It would be cool to reduce the pump speed to around 50% .

thank you,
Chris

MichaelDvP, 2020-12-13:
Are you sure that it is the frost-temperature? There is another temperature setting in RC35 (and i think also in your ES73) to prevent reducing if boiler/radiators are to small for the low outdoor temperature. We do not have it in ems-esp for now, it's in pos 38 of telegrams 0x3D/0x47,.(hc1/hc2,..).  

For pump modulation there is a setting in telegram 0x16, pos 9 for max and 10 for min. Check with `read 8 16` and try if it writable with `call system send 0B 08 16 09 32` for 50% max. If your boiler uses 0xE6 for parameters i'm not sure about position, maybe pos 13(dez) for max modulation, check value before changing, should be 64 for 100%, and try `call system send 0B 08 E6 0D 32`.

SpaceTeddy, 2020-12-15:
well, if I log:
``
ems-esp:/# read 8 16
015+23:05:27.443 N 24: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 5A 64 00 06 FA 0A 01 05 64 37 02
``
``
call system send 0B 08 16 09 32
``

show me an error:
``
ems-esp:/# call system send 0B 08 16 09 32
Too many arguments for command
ems-esp:/# call system sendConnection closed by foreign host.
``

i'm on 2.1.0

MichaelDvP, 2020-12-15:
Yes there are quotation marks missuing. it should be `call system send "0B 08 16 09 32"`, sorry.

SpaceTeddy, 2020-12-15:
alright,
seems to be writable :)

```
ems-esp:/$ read 8 16
Command not found
ems-esp:/$ su
Password: 
000+02:08:05.139 N 0: [shell] su session opened on console
ems-esp:/# read 8 16
000+02:08:11.284 N 1: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 5A 64 00 06 FA 0A 01 05 64 37 02
000+02:09:00.712 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 5A 64 00 06 FA 0A 01 05 64 37 02
ems-esp:/# call system send "0B 08 16 09 32"
000+02:09:05.983 N 3: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 5A 64 00 06 FA 0A 01 05 37 37 02
000+02:09:06.457 N 4: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 5A 64 00 06 FA 0A 01 05 37 37 02
ems-esp:/# show
EMS-ESP version 2.1.0

Boiler: BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:03.06)
  Heating active: off
  Warm water/DHW active: off
  Service code: 0H
  Service code number: 203
  Warm water selected temperature: 55 °C
  Warm water set temperature: 10 °C
  Warm water disinfection temperature: 65 °C
  Selected flow temperature: 7 °C
  Burner selected max power: 0 %
  Burner current power: 0 %
  Pump modulation: 0 %
  Warm water type: buffer
  Warm water charging type: 3-way valve
  Warm water circulation pump available: off
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: off
  Outside temperature: 7.0 °C
  Warm water current temperature (intern): 49.7 °C
  Warm water current temperature (extern): 49.7 °C
  Warm water current tap water flow: 0.0 l/min
  Current flow temperature: 18.7 °C
  Return temperature: 16.6 °C
  Mixer switch temperature: 0.0 °C
  System pressure: 0.9
  Max temperature: 23.7 °C
  Warm water storage temperature (extern): 49.7 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: off
  Warm water active: off
  Gas: off
  Flame current: 0.0 uA
  Pump: off
  Fan: off
  Ignition: off
  Warm water heating: off
  Heating activated: on
  Heating temperature setting: 90 °C
  **Circuit pump modulation max power: 55 %**
  Circuit pump modulation min power: 55 %
```

thank you.

MichaelDvP, 2020-12-15:
I've added commands for `pumpmodmin` and `pumpmodmax` to the v2.1.1b7.

SpaceTeddy, 2020-12-16:
Hi Michael,

as I mentioned i'm still on master 2.1.0, and compiled the current dev build with your changes.
Unfortunatly, for any reasyon, the 2.1.0 does not accept an update.

_What I did:_

1. compiled the dev build for ESP32 and  board = wemos_d1_mini32
2. OTA update which was successful. After reboot, the Web interface still shows me 2.1.0
3. HW reset -> same behaviour
4. update firmware.bin via web interface (no upload with Big Sur and Chrome & Safari); Win10 + Chrome works.
5. web shows "activating new FW"
6. -> still 2.1.0

any idea?

kr,
Chris

MichaelDvP, 2020-12-16:
There is a issue downloading the dev-source, you always get the 2.1.0 main-source, but i think git fetch works. Or you can download from my [github-fork](https://github.com/MichaelDvP/EMS-ESP), it's in sync with 2.1.1b7. 
Or try the precompiled esp32 build, it's compatible to the D1-mini32 board.

SpaceTeddy, 2020-12-16:
wow, are you always on? Thanks for the information. I try the pre-build first and then probably your fork.
If it works we can close this issue.

thanks again.

SpaceTeddy, 2020-12-16:
pre-build works!

closed 2020-12-16 by SpaceTeddy