emsesp/EMS-ESP #664: thermostat_data_lastcode 806 incorrect

SpaceTeddy, 2020-12-16:
Dear freaks :),

with the latests builds the last error code of the thermostst or boiler is transmitted via mqtt.
I noticed that the error code for i.E. 806 is not 100% correct, becasue the code does not include the heating circuit.
Example:
MQTT Error code:  ems_esp/thermostat_data_lastcode | A2(806) 03.12.2020 07:53
-> 806 is correct
->for A2 you has to add the number of the heating circuit straigt behind the "2" -> A2[hc]

source: https://productslu.buderus.com/buderus/productslu.buderus.com/user/de/buderus_servicehandbuch_ems_2019.pdf
```
806 A Raumtemperaturfühler für
Heizkreis defekt
(A21=Heizkreis 1;
A22=Heizkreis 2;
A23=Heizkreis 3;
A24=Heizkreis 4;
A25=Heizkreis 5) 
```
kr,
Chris

MichaelDvP, 2020-12-16:
Can you please check the telegram 0x12: `read 10 12`. My RC35 gives only 2 characters and then the code:
`Thermostat(0x10) -> Me(0x0B), RCErrorMessage(0x12), data: 32 32 03 30 94 0C 12 0F 00 00 01 19 `
This is error A22 (816) and the 'A' is not shown in telegram ' 2' (32) '2'(32) 816 (03 30), date, duration (00 01), device (19), there is no space for a third char.

SpaceTeddy, 2020-12-16:
sure,

```
read 10 12                             
[Kems-esp:/# [K000+02:17:54.494 N 1: [emsesp] Thermostat(0x10) -> Me(0x0B), RCErrorMessage(0x12), data: 41 32 03 26 94 0C 07 03 35 00 01 18 41 32 03 26 94 0A 10 16 2C 00 35 18 41 32 03                                           
ems-esp:/# [K000+02:17:54.868 N 2: [emsesp] Thermostat(0x10) -> Me(0x0B), RCErrorMessage(0x12), data: 32 03 26 94 09 11 0C 1A 00 04 18 41 32 03 26 94 09 11 0C 06 00 01 18 (offset 25)                       
```

MichaelDvP, 2020-12-16:
The oldest is from 12.09.2020, also A2(806) from device 0x18, but there are also only two characters. Ems-esp can only report what is in this telegram.

SpaceTeddy, 2020-12-17:
Understood. But if possible you can thing about, if it would be possible to add the [hc] number. usually it should be possible, becasue the used hc for that thermostat or boiler is known. Not really needed, but would be nice.

thanks,
chris

MichaelDvP, 2020-12-17:
Difficult. As you see my last error shows `22(816)` for `A22(816)`, i dont know how bosch handles it. Does it help to add the device id? i.g. `A2(806) id:0x18 03.12.2020 07:53`

SpaceTeddy, 2020-12-17:
Guess that this is not an option. My wish would be to get the same error message like my thermostat or Boiler shows me, but if it is difficult to implement this, then it is also fine for me. No worries. 
Lg,
Chris

closed 2021-10-04 by proddy